theyosh/TerrariumPI #88: Update?

Barbara1984, 2018-02-05:
Hi, my Rasp finds no update under cdTerrariumPI> git pull, do I have to do something else?

theyosh, 2018-02-05:
You probably downloaded the zip file. That cannot be updated with git.

Stop the software. Move the existing folder to a backup name. Make sure you do this all as user pi :

`mv TerrariumPI TerrariumPI.bak`

Then follow the installation steps for a git installation: https://github.com/theyosh/TerrariumPI#installation

Then before you reboot, copy the old settings.cfg and history.db file from the backup folder to the new folder.

And restart the Pi. You will then have a new installation with your existing config and history data. Be aware that the first restart will take about 5 minutes. This is due to the database update with the new version.

Barbara1984, 2018-02-06:
Hi, thanks for your info but which folder I have to secure exactly because I do that on Filezilla :-)

Barbara1984, 2018-02-06:
Only the settings.cfg or?

theyosh, 2018-02-06:
and history.db

You should download those with filezilla. And upload back after the installation.

Barbara1984, 2018-02-06:
[image]

What is the problem, it can not start anymore?

Barbara1984, 2018-02-06:
[image]

theyosh, 2018-02-06:
Ok, sounds like configuration issues... Try a reboot.

If that does not help, remove the file settings.cfg and reboot again. Then you have to setup again your setup.

Else it is hard to explain what to do with only two pictures.

So reboot pi, and see what happens.

Barbara1984, 2018-02-06:
[image]

Can it be due to the permission? without Settings.cfg and history it starts.

theyosh, 2018-02-06:
No, it is a code issue. But the quick fix now is that you leave the history.db file on the pi and remove the settings.cfg

And reboot. This will restart an empty system with old data. When you then add your sensors again it should pickup the old history.

You will have to wait about 5 minutes when you reboot the Pi, because the DB is being updated. That takes some time.

Barbara1984, 2018-02-06:
Ok, thanks for your help, everything works again.

closed 2018-02-06 by Barbara1984

theyosh, 2018-02-06:
Cool! Nice! Upgrades of configuration is a bit hard to test in code...