theyosh/TerrariumPI #90: UV sensor Support

Jelly-Flinger, 2018-02-07:
Hi

Would it be possible to add support for a UV sensor? I was thinking of adding VEML6075.
https://www.vishay.com/docs/84304/veml6075.pdf I believe its an i2c device. I have read around a little and it's one of the few inexpensive devices that truly measure UVA and UVB? (RS electronics sell 5 for approx $10 australian.)

I have read various stories about cheaper UV bulbs failing to emit the appropriate spectrum within weeks. I think it would be interesting to have the capability to monitor the bulbs output. It would be quite informative if different folk track and report the performance of different brands of bulbs with time. I guess in the longer term, it might even be possible to modulate the bulbs output to hit an appropriate output?

Cheers

theyosh, 2018-02-09:
Euh, I see the reason. I also do have that question about my lights. And I know that some lights only work for a couple of months. So I already use more expensive lights that work for one year.

But I would also to know how good they are. And this sensor could help. The only issue I see is that you need to solder it with some resistors and other parts. For me is that not a problem. But I cannot support that soldering. So I think I will give it a try. But as there are more hardware wishes, I cannot give an ETA on when this will be done.

theyosh, 2018-02-11:
Pi Zero should work. I only have not tested it myself.

Jelly-Flinger, 2018-02-13:
Great - will feed back on how i get on. 

FYI VEML6075 is available on various breakout boards ($6 australian). When I researched it I had not realised how tiny it was! Possibly I will get the board and have a play. I am not sure my soldering skills are adequate for such work yet.

theyosh, 2018-02-13:
Can you post a link? Just for reference

Jelly-Flinger, 2018-02-22:
Sure.

Here is a link on amazon (out of stock) [ https://www.amazon.com/Interface-Board-VEML6075-Sensor-Module/dp/B076KQYVBV ](url)

However, there are numerous options on Ali express for 4 or 5 dollars US. 
[https://www.aliexpress.com/w/wholesale-veml6075.html?initiative_id=SB_20180222025426&site=glo&groupsort=1&SortType=price_asc&g=y&SearchText=veml6075](url)

Jelly-Flinger, 2018-02-22:
Interestingly the image also appears to show 6040 on the same board. This is a visible light spectrum measuring  chip from the same manufacturer. No reference made to this in the tech specs so maybe an image for a slightly different product?

theyosh, 2018-06-27:
Just order some from China.... lets see how fast they get here...

theyosh, 2018-07-04:
Hardware arrived at the local postal office. Waiting on custom clearing...

theyosh, 2018-07-07:
Hardware is being tested.....

theyosh, 2018-07-13:
Running in production now

Jelly-Flinger, 2018-07-17:
Thats awesome! I have a couple of the sensors and will let you know how I get on. I was just thinking I need to buy a new bulb so I will try and get it set up asap for a before/after comparison. 

As always appreciate all the hard work!

theyosh, 2018-07-17:
The sensors I have, are not able to change the I2C address. Therefore, only one sensor can then be used... :(
The code does already contains the code, so you could already start with it.

theyosh, 2018-07-18:
I am closing this issue as hardware is supported. It could be that some graphs need some changes regarding UV. But that is a different issue. So for now, it is implemented and working in my production terrarium :)

closed 2018-07-18 by theyosh