theyosh/TerrariumPI #91: webcam settings

nopalpite, 2018-02-07:
Hi,
How can i change camera settings like resolution, aperture, brightness ? Is it possible ?

theyosh, 2018-02-08:
Euh... not yet. Resolution will be added. Other options not. I work with auto balances and wait for 2 seconds to let the camera auto adjust. Then I take an image of the webcam.

A rpi-cam has always the same resolution. And external does not need to know it, it can read it out from the source.

theyosh, 2018-02-08:
Update with git pull. Restart and clear browser cache. That should give you the new resolution options

theyosh, 2018-02-09:
So, i will close this issue now. I will not make other webcam settings for now. To much variation with different types of webcams.

closed 2018-02-09 by theyosh

nopalpite, 2018-02-26:
Ok,
Sorry for my late answer, thanks for adding resolution settings. I tried to change settings directly with v42l -ctl but it change nothing. Images are over-exposed.
[image]
Thanks again for your work

theyosh, 2018-02-26:
I can do this: https://unix.stackexchange.com/questions/412214/how-to-manage-the-white-balance-in-webcam

And it will change my USB webcam color settings and white balance

theyosh, 2018-02-26:
Or other trick. Use it on your laptop and see if the camera will work correctly. Use tools to adjust the white balance. When happy, disconnect from laptop and use it on the PI.

It could be that this will somehow the webcam to work better with auto white balance or other settings

nopalpite, 2018-03-03:
Hi, so i tried 2 differents solutions.It's not working for me. So i use another webcam, but when i try to add it don't save changes. I decided to reinstall the last one, same result changes not saved .... Bizarre... Is there a bug since the last  webcam resolution update ?