theyosh/TerrariumPI #94: Mobile interface, and system environment

Barbara1984, 2018-02-09:
Hello, why are the system heating and cooling at the same time? That makes no sense or I understand something wrong :-)
[image]

And why is my phone always Invalid Date and s.PC everything fits, is the interface optimized phone?
[image]

Thx for your Help

theyosh, 2018-02-09:
Hmm, because of the invalid date message, I expect some javascript errors. Could you try to flush your browser cache on your phone?

Else it is a problem with locale settings and formatting. I have used some external code which I cannot fully test on all devices. In order to figure that out, I need to debug on a mobile device...

The interface is a existing design that should work on a mobile. So that is not an issue. In this case, there are some javascript problems...

theyosh, 2018-02-09:
Also if you make changes, it can take about 1 minute before your browser will pick it up due to caching.

Barbara1984, 2018-02-09:
Ok thanks that helped with the cache on the phone, but why is the heating and cooling running at the same time?

theyosh, 2018-02-09:
euh, not clear what you mean. Because the image shows temperatures and no times.

But this is something that has been setup in the environment page. Could you make a screenshot of the environment page at the heater and cooler settings

Barbara1984, 2018-02-09:
I have :-) At the top is a small picture :-)

theyosh, 2018-02-09:
No, that is dashboard. Need environment page.

Barbara1984, 2018-02-09:
Sorry now you can not see it because both are out but earlier both were on at the same time.
[image]

theyosh, 2018-02-09:
But this is all correct. You have selected for both settings the sensor modus with the same sensor.

So that will show the same values. So that is correct.

What are you expecting?

theyosh, 2018-02-09:
Ah, now I see.

Barbara1984, 2018-02-09:
But it should not happen that the heating and cooling are turned on at the same time, right? It would have to be so if the heating is on, the cooling must be off.

theyosh, 2018-02-09:
Both systems do not exclude each other. So at some point they are switched on, and will stay on when the temperature does not reach either the max or min value.

Your idea:

> But it should not happen that the heating and cooling are turned on at the same time, right? It would have to be so if the heating is on, the cooling must be off.

Is implemented through the light system. That is my main heater base, and the heater and cooler are helpers. So the cooling and heating are both looking at the lights system.

But what I would expect in this setup is that both switches / devices are OFF. And when temperature drops below 21, the heater should go on. The cooler should stay off. Until 29 degrees. Then the heater should go out, and the cooler should go on till it is 21 degrees again. That is what I would expect. And that looks like both are excluding each other. But that is based on the sensor logic, not what we logical would expect with heating and cooling.

So, what you could try is to manually switch either the heater or cooler switch off. And see if the system does switch it on again. If so, then we have a bug. Else, if it stays off, it could be that there was some initial state issue. And if that does not repeat later in time, I see it as a glitch.

theyosh, 2018-02-09:
With switching off I mean that you go the to the power switch status page and click the big power icon to toggle the switch

Barbara1984, 2018-02-09:
Ok, if I click on the green switch icon it turns blue and then turns green again

theyosh, 2018-02-09:
Hmm, I did find a bug. But that is still not the final solution. Could you do a git pull and restart the software? That will load the latest version of the code.

then we need to look into the logging. In the folder log there is a terrariumpi.log file. Look for 'terrariumEnvironment' log messages

Aapjes94, 2018-02-09:
I had the same errors as Barbara1984 seems to be having after updating to 3.3.0. Switches being stuck and some error with the time stamps. Everything was resolved when resetting the pi and clearing the cache. Have you also reset the system from the screen session?

Barbara1984, 2018-02-09:
Okay, but unfortunately, it must be tested again tomorrow afternoon now has to work in bed tomorrow again :-(

theyosh, 2018-02-09:
Yeah, software is under heavily development... :P So after each update restarts and cache clearing is needed..

theyosh, 2018-02-09:
But for now, I can't find anything in the code that could cause this. So a restart of the software could help, or a full pi reboot.

Else we have to see if we can find in the logging

Barbara1984, 2018-02-11:
Hi ich habe gerade das Update gemacht. Aber ich sage bescheid wenn etwas ist ...

closed 2018-02-11 by Barbara1984