theyosh/TerrariumPI #95: USB power switch

bartch, 2018-02-11:
Hi,
as I read on the start page it is possible to use usb power switches, but either I am to stupid to find it in the web config or the one I use is not supported. 

I found this cool energenie EG-PM2 power switch which can be used via USB at the raspberry PI. I tried it on the terminal with sys call sispmctl and it works really nice. This would help to control everything which has a normal 230V plug. 

Do you support this type and I am to stupid or is there a possibility for you to add it?

BTW: I am really impressed how much time you spend for this project. I know a lot of GitHub projects which are not even half that vivid than yours ;)

theyosh, 2018-02-11:
Hmm, I have seen those in the past. Did not wanted to use them at first due to the amount of space they take.

But I do like to support it, and so I have ordered a USB and Lan version. If I do not use the powerswitches for my terrarium, there is always other use for them :)

theyosh, 2018-02-13:
Alpha/Beta code is added. 

1. Do a `git pull`, 
2. run installer `sudo ./install.sh` for updating packages
3. restart according to https://github.com/theyosh/TerrariumPI/wiki/FAQ#how-to-start-terrariumpi
4. clear browser cache.
5. add a new 'Energenie USB' power switch

You are now able to add 'Energenie USB' switches. The are numbered from 1 - X. It should be possible to use multiple devices. Port 7 for example would then be port 3 on second device. (This is not tested)

theyosh, 2018-02-13:
Now also for Lan devices! YEAH

bartch, 2018-02-14:
Works like a dream! Thank you!

closed 2018-02-14 by theyosh