theyosh/TerrariumPI #96: Change to install script

Natrixz, 2018-02-11:
Hey Yosh,

It might be good to make the install (and start.sh) script more dynamic.
Eg. Change the user "pi" to be a dynamic field, mainly to be the user invoking the install script.
Shouldn't be a big change but it allows for a bit more easy install when you don't run "pi" as your main user (or have it deleted)

With kind regards,
Natrixz

theyosh, 2018-02-11:
There is more into it. Changing files during installation will give issues with new updates through git pull. And that will give more questions by people not familiar with git.

Also the 'other' user needs to be added to certain system groups in order to read out hardware.

So what I am doing now, is that the installer script will do a user check. If correct the installer continues and force the rights of the software folder. Those rights will then be used by the start script to read out the user that should run the software.

Natrixz, 2018-02-12:
Yes that is true,

While writing this I noticed you already changed it to include the current user running the installer.
I was typing up;
import os
os.getlogin()

If you need any help testing/tinkering, let me know :)

theyosh, 2018-02-13:
So, for you it is all working? Then I think this issue can be closed. It is all working fine for me still

Natrixz, 2018-02-14:
Yes, both my actual pi (different user) and my test pi running "pi" work fine updating/git pull and re-running setup :)
Nice work!
Saves the people like me from changing the scripts every git pull haha

closed 2018-02-14 by theyosh

kaju666, 2019-01-06:
HI i do not know why your install script removes oracle-java8-jdk. It's not ussles like others apps e.g. libreoffice, wolfram, sonic.

In my situation i also on the same rpi use OpenHAB which uses java so every Terrarium Pi update i need to manually change install.sh