theyosh/TerrariumPI #97: Unable to add new switch

MichalDuban, 2018-02-11:
In the latest version of your app., I am not able to add new switch (source GPIO). I fill all required fields, but add button does not respond.

theyosh, 2018-02-11:
Clear browser cache? The code has been updated. I guess you have old HTL code in your browser cache

Barbara1984, 2018-02-11:
In the system environment, the plus symbol is not synonymous top right :-)

theyosh, 2018-02-11:
Found the bug. Do a git pull and clear browser cache. You should now be able to add switches again.

theyosh, 2018-02-11:
@Barbara1984 the plus symbol should not be there :) and is removed

MichalDuban, 2018-02-12:
Hi, it seems add button is working now, but submit button not.

theyosh, 2018-02-13:
Sorry for that. Do another `git pull` and clear browser cache. Should work now

theyosh, 2018-02-13:
I have added already new switches, and that is working. So I think it is all ok now?

MichalDuban, 2018-02-14:
Great! Now it is working properly. Thank you.

closed 2018-02-14 by theyosh