theyosh/TerrariumPI #101: Alerts by Email ?

nke69, 2018-02-15:
Good morning,
Would it be possible to integrate in a future release the possibility to receiving alerts by Email, in addition to that present in the interface. "Relevant" alerts like temperature, humidity, etc.?

Thanks in advance ;)
nke69

theyosh, 2018-02-15:
Hmm, I think this will get a very low priority. Because this is a lot of work.

- Every alarm will send an email? That can be very much mails
- How to configure mail settings. Need to connect to other mail server to be able to sent mail
- This can become a spammer if you get a lot of mails.
- What kind of message would you have? Custom messages editor?

So I think that this will not be implemented that fast. Due to the fact that I do not need it, and that it will be more code then you first will expect

theyosh, 2018-02-15:
If somebody else want to make this and send a pull request that is fine

nke69, 2018-02-16:
I will try to see what I can do;)
At first alerts that the user can choose (temperature for example), not systematic alerts on all points :)

kahuwi14, 2018-03-02:
Hey guys!

Two weeks ago, I implemented pushover. 
I'm parsing the logfile with a python script, once a minute. If I find some keywords (warning, temperatureSensor), I send a pushover message. 
If somebody is interested, I would upload the codes.

nke69, 2018-03-02:
I'm very interested 👍 

thx

theyosh, 2018-03-02:
Ah, yes pushover. I was looking into that, but my phone does not allow Google Play stuf (Jolla / SailfishOS) and therefore I cannot test it :(.

But maybe upload the code here, and I add it as a code contrib for other user to use if they want to

Natrixz, 2018-03-02:
I could test it if you want, i have a pushover account anyway :)

kahuwi14, 2018-03-02:
I'm just writing a little README and after finishing that, i'll upload a container with all the used files!

kahuwi14, 2018-03-02:
[logcheck.tar.gz](https://github.com/theyosh/TerrariumPI/files/1776060/logcheck.tar.gz)

If you have any questions, just leave a comment.

kahuwi14, 2018-03-02:
I improved the script a bit.
Now it's possible to use a different config-file for ntfy, if you'd like to define a specific config while using ntfy for TerrariumPi.

[logcheck.tar.gz](https://github.com/theyosh/TerrariumPI/files/1776298/logcheck.tar.gz)

Natrixz, 2018-03-03:
You also need to add "parse" as a dependency.
Secondly, When the temperature reaches the "min" or "max" and go's into warning, it doesn't generate an alert in the logs, hence it doesn't get parsed.

kahuwi14, 2018-03-03:
> You also need to add "parse" as a dependency.
Ohh yes, of course.. ^^

> Secondly, When the temperature reaches the "min" or "max" and go's into warning, it doesn't generate an alert in the logs
That's right and okay for me, because that are the triggers for the heating and if you set them very thight, you'll get a lot of messages.

> Would it be possible to get a warning in the log as well as in the GUI? :)
is that a question to me? ;)

kahuwi14, 2018-03-03:
I just left the new files under the commit.

theyosh, 2018-03-04:
New files updated.

To get the same notifications as in the GUI, I have to create new code for it. That is now not possible. The logfile is more a logfiles for checking the hardware. And not if there are alarms based on settings.

So this can be added, but still at slow pace....

kahuwi14, 2018-03-04:
> The logfile is more a logfiles for checking the hardware.

and the script is a little workaround until you have found the time for something else, but I think it's enough at the moment.

> So this can be added, but still at slow pace....

Imho, there are not that time critical events in a terrarium. ;)

kahuwi14, 2018-03-12:
Hey there!
I renamed the script and implemented some features, mainly debugging and logging.
@theyosh would it be easier to publish the files over a pull-request? In that case, I'm not sure how.. ;)

[TePiNot.tar.gz](https://github.com/theyosh/TerrariumPI/files/1803996/TePiNot.tar.gz)

kahuwi14, 2018-03-12:
Hey guys!
I improved the script and renamed it too.
Now there is rudimentary debugging and logging.
For further information, have a look at the README. ;)
@theyosh should I create a pull-request? If so, I do not exactly now how.. ^^

[TePiNot.tar.gz](https://github.com/theyosh/TerrariumPI/files/1804607/TePiNot.tar.gz)

theyosh, 2018-03-15:
No, this will work also... just added to the development branch

theyosh, 2018-06-05:
And here it is.... the first Alpha version for alerts. Please be prepared to be bombarded with messages :)

- Support for email, twitter, pushover en telegram bot
- For now, you can have about 20 different messages from different parts of the system
- Use of variables like temperatures or other values in your messages

There is not a lot of help information yet. So it is at your own to try this. In order to get it all working, new python modules are needed. Run as root:

`pip install python-twitter python-pushover python-telegram-bot`

or re-run the install.sh script

Natrixz, 2018-06-08:
Can you add (different) SSL options for the smtp part?
My SMTP server requires a certain SSL and it doesn't "default" work.

theyosh, 2018-06-10:
@Natrixz what does it need?

Natrixz, 2018-06-10:
Hi Yosh,

It needs either SSL or StartTLS (on a different port)

theyosh, 2018-06-10:
Just enter the right port number. Code will always try to do a StartTLS connection. So that should be working already.

If that fails, the code will drop back to normal mail.

Natrixz, 2018-06-10:
Hmmm strange, mine doesnt seem to do anything. No outbound Connections what so ever (according to firewall)
How did you implement the code? Any place i can check?

theyosh, 2018-06-10:
I add some new code for error messaging. Look at the screen sessions for errors when trying to send mail.

I also added an authentication on requesting the notifications credentials. Because also the tokens should not be shown for normal users.

try a new git pull on the development branch. At this time, I am just making the basic functionality based on my setup. And for me mail is working. Also make sure you are allowed to reach to an external mail server at your ISP. I am not allowed to connect to external mail servers :(

Natrixz, 2018-06-11:
Right now I do see some errors however they might just be related to sensors momentarily not registering.
Could you maybe add a test notification button? :)
(that way it's easier to trigger a message)

theyosh, 2018-06-11:
This looks like you are using SSL for mail. That was not implemented yet. Just did. Latest commit should now try smtp with SSL when normal fails. This could introduce a slight delay in sending mails.

But I think the amount of persons using SSL with mail is low compared to normal or TLS

Natrixz, 2018-06-11:
I will test them both :)
My mail server supports both

Natrixz, 2018-06-11:
Both SSL and TLS work at my 2 pi's :)

theyosh, 2018-06-11:
Nice! Now it is fine tuning this. Twitter, pushover and telegram are disabled in code for now

Natrixz, 2018-06-11:
Yes I noticed haha
Was expecting a christmas to appear first time a alert popped up lol
I don't mind testing pushover(and probably have a telegram bot somewhere)

Natrixz, 2018-06-11:
Ps. I did notice that even tho I turned off "System warning messages" it still mails me with;
Measured value 127.66% from humidity sensor 'Links-Koud' is outside valid range 50.00% - 105.00% in 0.55139 second

theyosh, 2018-06-11:
Hmm, not sure about your latest issue. But the code has been changed once again :). Now you can select witch service you want to use for each notification. And therefore twitter, pushover and telegram bot are now also enabled.

And I have been switching on and off notification messages and works like expected. So try the latest development branch. Make sure you clear your browser cache to get the new notifications page working correctly.

Natrixz, 2018-06-12:
For some reasons the mails are not beeing send anymore...
The new notification screen looks better tho

theyosh, 2018-06-12:
Euh, sounds like mixing up code.... I can not reproduce. Make sure you have enabled/disabled the services per message and press the save button below. Toggling the services behind the messages will not update the notification settings. So always press save button below.

[image]

theyosh, 2018-06-12:
And you could delete the file notification.cfg and restart the software. This will start the setup of notification system all over again. And should not send anything

Natrixz, 2018-06-12:
Woops, apparently i need to restart the software for the changes to take effect :)

theyosh, 2018-06-15:
Ok, I think that it is very close for a new release. I am currently testing it on a live terrarium.

If there are still issues or wishes, please tell now.... or silence for ever ;)

dasolan, 2018-06-16:
Still looking forward to the new release, hurray!

Natrixz, 2018-06-18:
As soon as I fill in the notifications and restart terrariumpi it won't start and ends with an error.
Happens since the rewrite (I think)

theyosh, 2018-06-18:
What is in the screen session? Should be an error I guess

Natrixz, 2018-06-18:
2018-06-18 19:43:13,000 - INFO    - terrariumWrapper     - Starting TerrariumPI server at location: http://192.168.69.214:8090 ...
Traceback (most recent call last):
  File "/home/natrixz/TerrariumPI/terrariumPI.py", line 17, in <module>
    terrariumEngine = terrariumEngine()
  File "/home/natrixz/TerrariumPI/terrariumEngine.py", line 113, in __init__
    self.__load_power_switches()
  File "/home/natrixz/TerrariumPI/terrariumEngine.py", line 213, in __load_power_switches
    callback=self.toggle_switch)
  File "/home/natrixz/TerrariumPI/terrariumSwitch.py", line 97, in __init__
    self.set_state(terrariumSwitch.OFF,True)
  File "/home/natrixz/TerrariumPI/terrariumSwitch.py", line 285, in set_state
    self.callback(data)
  File "/home/natrixz/TerrariumPI/terrariumEngine.py", line 618, in toggle_switch
    self.notification.message('switch_toggle_' + ('off' if data['state'] == 0 else 'on'),data)
  File "/home/natrixz/TerrariumPI/terrariumNotification.py", line 578, in message
    self.send_email(title,message)
  File "/home/natrixz/TerrariumPI/terrariumNotification.py", line 448, in send_email
    msgImage = MIMEImage(fp.read(), filename=os.path.basename(self.__profile_image))
  File "/usr/lib/python2.7/email/mime/image.py", line 43, in __init__
    raise TypeError('Could not guess image MIME subtype')
TypeError: Could not guess image MIME subtype
2018-06-18 19:43:29,000 - INFO    - terrariumWrapper     - Restart counter: 1/5

theyosh, 2018-06-18:
What do you use as profile image? Because it is complaining not a valid image :(

Natrixz, 2018-06-18:
Its a picture of my Brazilian rainbow boa ^,^’

theyosh, 2018-06-18:
duh... could you upload here?

Natrixz, 2018-06-18:
Your too fast! xD I was booting my laptop lol

theyosh, 2018-06-18:
A new git pull should fix it. Restart after the pull

Those colors of your boa are real? No filter on the image? Those are freaky colors.... 
And a tip, scale your image down for my software. Because your image is now 1MB. Will produce some big emails :)

Natrixz, 2018-06-18:
Yep it's real, no filter at all =)
That's him about 3 months ago, he grew a bit bigger in the meanwhile tho haha
The scales reflect the light in such a way into our eyes (and the camera) that makes him appear like a blue-ish rainbow :)
(his main color, outside the rings, have gotten more red with the latest shed)

Thanks for the tip btw! I will scale the image down a bit =)

Natrixz, 2018-06-18:
Fix works and survives a restart :)

theyosh, 2018-06-20:
This was a long way.... but is has been released! I will leave this issue open for a couple of days if there are some bugs or questions

Natrixz, 2018-06-24:
So far I haven't found any bugs, just as a heads up :)

theyosh, 2018-06-24:
@Natrixz thaks for the update. I did found a small bug in reloading the messages. But it seems to work well.

theyosh, 2018-07-02:
Closing.. it is working

closed 2018-07-02 by theyosh