theyosh/TerrariumPI #102: Hedging if interface is also accessible via DNS?

Barbara1984, 2018-02-15:
Hi,

is it possible if i'm on the go, for example, and need my interface to install a security. For example: a password before the interface is displayed?

Thank you for your help and your time

Barbara1984, 2018-02-15:
https://de.wikihow.com/Eine-Webseite-mit-Passwort-sch%C3%BCtzen

Maybe something like that?

theyosh, 2018-02-15:
Your request is clear. Your provided example as a very bad example. :) Because passwords in javascript is no security. I can just read out the file and know the passwords.

So what I have done is that I added an extra option in the system settings to enable 'full authentication' so that when you open the webinterface, you will have to enter a username and password.

theyosh, 2018-02-15:
Git pull, restart, clear the browser cache, go to the settings page and test it out :D

Barbara1984, 2018-02-15:
Hi, no that's not what I mean. I try to explain it differently:
If you open the interface comes a password query and only then you can see the entire interface, as an example of a photo :-)
[image]

theyosh, 2018-02-15:
That is way more work... can't do that. The functionality is the same. Only I will not make that kind of login window.

So, I add the option to completely lock the web interface. And after a login, you can see the interface. That is how far I can go with this.

Barbara1984, 2018-02-15:
Ok, thanks anyway, but can you give me a tip on how I could protect my interface from strange views in the www?

Barbara1984, 2018-02-15:
Funny thing is that I have now made git pull and because cache emptied and pi reboot but I'm still not prompted to enter a password I come immediately to the interface

Barbara1984, 2018-02-16:
Ok, now it works, thank you so much for your help.

closed 2018-02-16 by Barbara1984

theyosh, 2018-02-16:
If you have authenticate one time it will stay in the browser as long as the browser is running. So after enabling it, to test it, you have to restart your browser. But I guess you have done that, and you see the authentication

Barbara1984, 2018-02-16:
Yes, I saw it today, which makes me very happy, thank you