theyosh/TerrariumPI #103: Denkovi USB v2

Natrixz, 2018-02-17:
Hello Yosh,

I have recently orderd a new denkovi 4 relay USB v2 and it doesn't appear to work.
The serial commandline is a little different.

Would you happen to be able to support this version as well? :)

Natrixz, 2018-02-17:
If you need any info from me, or testing...please let me know :)
(we could talk in Dutch if needed haha)

theyosh, 2018-02-18:
Using the Jar is the only way with the V2 version... that sucks... because either existing V1 code is removed in favor or the Jar file. Or add multiple denkovi's types.. 

I am not sure what I am going to do... I have to think a bit more about how to approch this.

Natrixz, 2018-02-18:
Yeh indeed, I needed it for my home automation so figured I'd give it a try on the terrarium as well.
Multiple denkovi's would be a pain in the ass as well, because then you really need to add alot of extra code for it to be compatible with all the versions. (because of the crappy commandline usage)

theyosh, 2018-02-19:
Bad news... I will not support V2. I have looked in the JAR file they provide. It is all closed binary driver stuff. For every OS different DLL and SO files. Also a lot of native code for Windows, Mac, Linux, ARM etc....

So it is totally unclear what happens in the closed code. So we have to wait till somebody does a reverse engineering. But I guess that will also give problems with denkovi.

Natrixz, 2018-02-19:
Yeh i kinda figured, been spending Some time in and with the jar file for something else as well
Dont think denkovi Will like it much if we start reversing their application.
(Plus there are other choices to use for switching relais)

Ps. i might be bugging you with some other feature request if I cant make it work with the current set of options :P

closed 2018-02-19 by Natrixz