theyosh/TerrariumPI #106: Dimmer usable with temp-sensor

Natrixz, 2018-02-21:
Hey Yosh,

Would it be possible to Add a new option to the dimmer section?
or maybe a new section to the Heating section?

Namely;
If the temp exceeds the min/max temperature either raise or lower the dim-percentage.
And if possible, add a night-time section too?

That way the temperature is always able to be controlled, for example by a ceramic heat emitter that is dimmable.
Night time; lower temps
Day time; higher temps

With kind regards,

theyosh, 2018-02-22:
I see what you want, but I think this is just possible with switching on and off. Specially with ceramic, because that will not lose it heat at once.

So specify a high and low temperature that are close to each other, and the heater will keep it at a steady temperature. Enable the option 'during the day' and the heater will run for 24/7 at the same value. Or set the heater in timer mode with a sensor. That way the heater can also run 24/7 and keep the temperature between min and max of the sensor.

So I will not add this option. Because you are already able to keep the temperature in a window of 1 degree with the current system.

Natrixz, 2018-02-22:
Would it atleast be possible to add a nighttime min/max temp?

I do not want to have the same temp at all times of the day, currently im trying to make it happen by setting the dimmer-off value in such a way that its around a certain temp but it appears to change from day to day (depending on room temp etc)

Would be Nice if its possible to have a different night/daytime max (Maybe based on the lights/sundown)

theyosh, 2018-02-24:
Euh, this is exactly what this software does.

When you select a temperature sensor, and configure its min and max value, you can use that for the heater configuration. That way, the heater will only go on when it is to cold, staying on when it is to hot. Stay off when it is to cold again, and will toggle on.

As you specify a min and max value at the temperature sensor that is about 1 degree difference, you should get a pretty steady temperature.

The heater is designed to run when the lights are off. That is default

Natrixz, 2018-02-25:
Yes i know but I dont run any under tank heaters, just an Arcadia deep heat projector. This lamp is my source of heating and currently runs on a separate dimmer with thermostat so i can make the temp drop and be constant at night and higher during the day.

Otherwise I would need to add a second source of heating (low power CHE) for during the night.
Im trying to prevent having to do that

theyosh, 2018-02-25:
Hmm, ok, maybe this will work.

Put the Arcadia at a dimmer device. Put the **off** percentage on **100%**. Put the **on** percentage at **50%**.
This will make the power switch produce 100% power at off state. Yes, sounds strange, but this how it is made :).

So use that switch in the heater environment part. Use weather mode. Do not use sensors.

With this combination you get a situation that when the lights are on, the heater is turned off. But that turns the dimmer at 100% power. Which is effectively powering up the Arcadia for full heat.

When the night falls, the lights go off, and the heater goes 'on'. Which is lowering the dimmer to 50%. 

This way, the Arcadia will run at 100% during the day, and at 50% during the night. Which should result in different temperatures.

Watch the temperatures and play with the dimming values of the switch to change the temperature.

Natrixz, 2018-02-25:
Hmmm, that might actually work!
Will have to try it on a test setup first tho, the deep heat projector produces quite a bit of heat lol
Let's see if I can come up with a percentage that keeps a fairly constant temperature.

I will let you know how it go's :)
Shouldn't take me much more than a day or 2-3 (hopefully)

Natrixz, 2018-02-25:
Indeed, im curious to see what it does :)
Not sure how the commercial products dimm the lights....and what is needed with the deep heat projector.
Let's hope it just needs a constant power input to keep the temperature and doesn't warm up more and more when the power is constant.

Natrixz, 2018-02-26:
Hmm, for some reason the temperature is a bit erratic....
If I change the dimmer value's so it reaches about 32degree's, it slowly starts climbing more and more.
This does stop at some point, but higher than I set it at first. So after tinkering around with it a bit more, it also seems to go down again to a bit lower then where it started if I just leave it for a few hours O.o

Not sure what's causing this...multimeter shows no change in voltage supplied.

Natrixz, 2018-02-27:
Have tested by hooking up my multimeter to my reptile dimming thermostat and I can read from the multimeter that it actually actively dimms it.
It keeps fluctuating to maintain a steady temp

kahuwi14, 2018-02-27:
Hey yosh!
 
At first: thank you very much for your great work!!! I've so much fun in playing around with all the opportunities!!
Sadly, with so much possibilities, you'll get more and more ideas... ;)
At the moment, I'm with Natrixz: I'd just like to have a second heating in the environment section. In that way, it would be possible to set a temp for daytime and a temp for the hours at night and you are free in the choice of the heating system.

Would it be possible to implement this feature?

With best regards,
Karsten

theyosh, 2018-02-28:
Ok ok, two against one :).

I have thought about this, and I want to break it up in small parts. So first part is that I will add an extra option at the dimmer, which is the amount of % to go up or down when the temperature is out of range. The heater and cooler part will use that value to change the temperatures. So that should at least make it possible to 'play' with the power % to get the right temperature. But this will also need probably a value in which the new temperature settles before lowering it again. Not all devices will change in temperature that fast.

The other part will then be that you can change the working temperature during day and night. I still think this can be done already, but not with one temperature sensor...

As you want to specify the temperature, you NEED a sensor. So to get this to work, the min and max values of the used sensor should be changed to the new values. In order to do this, I was thinking of that you can specify a value in which the temperature will go up or down (negative value) based on the configured temperatures during the DAY.

So 1 sensor has to operate between 20 and 22 degrees during the day, and will operate between 25 and 27 during the night. So the difference is here add 5 degrees during the night.

Adding another heater part is to much. Also I am afraid that it will get to complex to setup. The features has already tripled last year :)

Natrixz, 2018-02-28:
Cool :)

I think that the addition of both features would be really nice and super usefull in many setups!

Offers a way to have a day temp of 32-33 and then drop it at night to 22 just by setting a value of 10 in the difference field.

And yeh the dimmer part Will probably need a steady/hold time, altho i reckon that when it works as it should; dropping the dimmer a few percentage with a right set delay and temp set 1 or 2 degrees below the max should be able to hold it fairly steady :)

kahuwi14, 2018-02-28:
The idea with the difference-value sounds great for me and would perfectly solve my problem!!! The second heater was just my first thought..  

Tripled??  That's crazy!! And really great success!!

theyosh, 2018-03-01:
(lightbulb)! :)

Just a thought, with a dimmer, I will try the following logic:

Calculate the average between min and max. Take about 10-15% margin and when the temperature is between the margin values around the average, take NO action. In other words leave as is..

In other words, I will try to get the dimmer to run on an average value in stead of min and max...

kahuwi14, 2018-03-01:
The new dimmer-feature seems to be really good for a heater that is able to dimm!
Now, the differences between ON and OFF are more smooth.

But you're still going to add the feature with the "Day&Night"-function?
... in some days/weeks of course.. ;)

Natrixz, 2018-03-01:
Sounds really good! My test bench is ready to roll with it :)
Good and fast progress! Nice work :)

theyosh, 2018-03-01:
tada... do a git pull and switch to branch development. Restart terrariumpi according to https://github.com/theyosh/TerrariumPI/wiki/FAQ#how-to-start-terrariumpi

Clear browser cache, and update your dimmer by setting the dim step.
Secondly go to environment page and update the heater settings.

Day & Night temperature will be next, but I have at the moment two solutions with cons and pros.. so that needs some more thinking how I will add that.

Natrixz, 2018-03-01:
Pulled and set-up, I will let it play out over night and tomorrow and get back to you how it's working on my test bench :)

theyosh, 2018-03-01:
fyi: heater settle time is in seconds in then environment settings page

Natrixz, 2018-03-01:
Yeh i figured :) 120minutes default would be long lol and setting it to 2 made it jumpy :P

Natrixz, 2018-03-01:
Quick observation; the 10% seems to be to big of a step, altho that might be with my lamp/setup. However can you make the step configurable?
(Yes i know, its another damn box...)

kahuwi14, 2018-03-01:
If a may help you to solve the problems, let the me know! ;)

theyosh, 2018-03-01:
the 10% value is at the power switch / dimmer setting

Natrixz, 2018-03-01:
Ah, dumbass me filled in 10 percent while reading the post on my 2nd monitor -,-
Set to 5% for the night, Lets see how it go’s :)

Ps using temp with 2 degrees between min/max at the moment

Natrixz, 2018-03-02:
So far it appears to work really well, even if I had a separate heatsource that adds a few degrees the dimmer Will lower the output within parameters :)

theyosh, 2018-03-02:
Way cool!

kahuwi14, 2018-03-02:
[logcheck.tar.gz](https://github.com/theyosh/TerrariumPI/files/1776045/logcheck.tar.gz)

If you have a question, just leave a comment.

theyosh, 2018-03-03:
Git pull, restart and clear browser cache. There is a new option in the heater section.

kahuwi14, 2018-03-03:
After hours of "fighting" with git, I finally downloaded and testet the other branch! ;)
It Looks great!! Thank you very much!!
An extensive test is going to follow in the next week!

While testing, i got an other idea: Should I use the language-files from the dev-branch?
It seems to be an older version and I could translate the new functions..

Natrixz, 2018-03-04:
I Will test it this week as well :)
Will get back to you when i have Some more results

theyosh, 2018-03-04:
There is one catch with the current solution: Do not change you temperature settings during the night. Because the code will temporary increase/decrease the used temperature sensors. When you than edit the settings and save them, they will store the temporary night temperature in config and will be used as day temperature.

@kahuwi14: Translations is a bit hard now. On the development branch I do change field names and other text during testing. That would result in translating multiple times the new features. I think that is not workable. So I think the best is just to translate the master branch.
What I can do is creating a new translation issue where you can subscribe. When the master branch is ready for translation, I give an update in the issue which will trigger a message for you that you can translate the missing text. Sounds like a plan?

kahuwi14, 2018-03-04:
Copy that! ;)

Natrixz, 2018-03-08:
Sorry for the lack of response, bit busy at the moment.
Test setup IS running, just haven't had much time to check on it yet.

Wanted to test a few things with it :)
(altho I did notice my temp-sensors beeing slow in dropping the temps)

theyosh, 2018-03-08:
That also means that it is working :) Else you would spent more time on it ;)

Natrixz, 2018-03-09:
Yes it does work! :)
And with it respecting the min and max it now doesn't overshoot or undershoot anymore.
Altho I do notice that the temp sensors are very slow in actually giving the correct temperature...but that's not something you can change (I think)
Or well, typing it; it does it fast(er) when the temperature go's up, not when it go's down. Down is slow :P
Wich causes the night time not to drop as quickly and the dimmer to clock down alot before going up again. But that's not that bad imho :)

theyosh, 2018-03-15:
Ok, then I close this issue as resolved

closed 2018-03-15 by theyosh