theyosh/TerrariumPI #108: create a new link to a page

nke69, 2018-02-24:
Hello,

I just placed my PH probe SKU SEN0161 and it works perfectly.
My python script tells me the good ph and everything is OK.

But how to call the python script in a page (tpl) so that it returns the value of the PH.

I created a new link in the menu on the left with the sensors and a new page called ph.tpl tpl but I'm looking for a moment, the code is not interpreted as if it was php.

[image]

I do not know this framework at all, would there be a trick, in the immediate future I would need to display the value of the ph on demand no need db storage or other ??

in advance thank you for helping me;)

theyosh, 2018-02-24:
It is a bit more work. But I suggest that I create a new branch and test this between us.

But did you add an extra ADC converter to it? Because the analog output cannot be read directly. It so, it that device capable of 8 analog devices? If so, I will also add an extra section to configure the pins for the ADC board

theyosh, 2018-02-24:
The template engine is described here: http://bottlepy.org/docs/dev/stpl.html#simpletemplate-syntax

You can run python code in it, but that will be cashed with the data first time requested.. :( So that is not really working I am afraid

nke69, 2018-02-24:
No worries and thanks in advance;)

yes of course I placed an adc MCP3008 8-channel 10-bit ADC which is the drawing : 
[image]
the probe is connected to the channel 0 on which the ph value is read

nke69, 2018-02-24:
i will try anyway

theyosh, 2018-02-24:
I have tried this at home: http://rasp.io/wp-content/uploads/2016/06/RasPiO-Analog-Zero.pdf

But it takes a bit of space... And I need a lot of GPIO pins to test everything :P

nke69, 2018-02-25:
Aie,  
after restart the server crash with this error :  

```
2018-02-25 00:56:48,000 - INFO - terrariumWrapper - Starting TerrariumPI server at location: http://192.168.1.61:8090 ...
_Traceback (most recent call last):
  File "/home/pi/TerrariumPI/terrariumPI.py", line 11, in <module>
    from terrariumEngine import terrariumEngine
  File "/usr/local/lib/python2.7/dist-packages/gevent/builtins.py", line 93, in __import__
    result = _import(*args, **kwargs)
  File "/home/pi/TerrariumPI/terrariumEngine.py", line 22, in <module>
    from terrariumSensor import terrariumSensor
  File "/usr/local/lib/python2.7/dist-packages/gevent/builtins.py", line 93, in __import__
    result = _import(*args, **kwargs)
  File "/home/pi/TerrariumPI/terrariumSensor.py", line 16, in <module>
    from gpiozero import MCP3008
  File "/usr/local/lib/python2.7/dist-packages/gevent/builtins.py", line 93, in __import__
    result = _import(*args, **kwargs)_
**ImportError: No module named gpiozero**
```
i run `pip install gpiozero`

restart

and have this like error : 

```
2018-02-25 01:00:55,000 - INFO - terrariumWrapper - Starting TerrariumPI server at location: http://192.168.1.61:8090 ...
The GPIO channel has not been set up as an OUTPUT
2018-02-25 01:01:14,549 - ERROR   - terrariumSensor    - Error updating hc-sr04 distance sensor 'Hauteur Eau' with error:
Traceback (most recent call last):
  File "/home/pi/TerrariumPI/terrariumSensor.py", line 157, in update
    GPIO.output(terrariumUtils.to_BCM_port_number(self.sensor_address['TRIG']), False)
RuntimeError: The GPIO channel has not been set up as an OUTPUT
2018-02-25 01:01:18,000 - INFO - terrariumWebserver - Running webserver at :::8090
The GPIO channel has not been set up as an OUTPUT
2018-02-25 01:01:19,260 - ERROR   - terrariumSensor    - Error updating hc-sr04 distance sensor 'Hauteur Eau' with error:
Traceback (most recent call last):
  File "/home/pi/TerrariumPI/terrariumSensor.py", line 157, in update
    GPIO.output(terrariumUtils.to_BCM_port_number(self.sensor_address['TRIG']), False)
RuntimeError: The GPIO channel has not been set up as an OUTPUT
```

nke69, 2018-02-25:
i've entered 0 as adress but nothing appear ..

theyosh, 2018-02-25:
Ok, try a new git pull and restart... Not sure if the error above will stop getting data from other sensors, but I will think so

nke69, 2018-02-25:
yes the first time I did the install but the package was missing, I did it manually and the start was able to start at that time ..

I have just restarted but it seems that there are still some errors .. I will check later and tell you what I see ..

Anyway thank you very much for your time and your involvement in this feature :)

theyosh, 2018-02-25:
Just did another commit... this should make it work :D

nke69, 2018-02-25:
here are the last mistakes I have:

I'm going to sleep but thank you in any case :)

2018-02-25 01: 51: 51,000 - INFO - terrariumWrapper - Starting TerrariumPI server at location: http://192.168.1.61:8090 ...
variable local 'pulse_start' referenced before assignment
2018-02-25 01: 52: 09,827 - ERROR - terrariumSensor - Error update hc-sr04 distance sensor 'Water Height' with error:
Traceback (most recent call last):
  File "/home/pi/TerrariumPI/terrariumSensor.py", line 167, in update
    pulse_duration = pulse_end - pulse_start
UnboundLocalError: local variable 'pulse_start' referenced before assignment
2018-02-25 01: 52: 14,000 - INFO - terrariumWebserver - Running webserver at ::: 8090
variable local 'pulse_start' referenced before assignment
2018-02-25 01: 52: 14,926 - ERROR - terrariumSensor - Error editing hc-sr04 distance sensor 'Water Height' with error:
Traceback (most recent call last):
  File "/home/pi/TerrariumPI/terrariumSensor.py", line 167, in update
    pulse_duration = pulse_end - pulse_start
UnboundLocalError: local variable 'pulse_start' referenced before assignment
variable local 'pulse_start' referenced before assignment
2018-02-25 01: 53: 16,402 - ERROR - terrariumSensor - Error updating hc-sr04 distance sensor 'Water Height' with error:
Traceback (most recent call last):
  File "/home/pi/TerrariumPI/terrariumSensor.py", line 167, in update
    pulse_duration = pulse_end - pulse_start
UnboundLocalError: local variable 'pulse_start' referenced before assignment
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/bottle.py", line 862, in _handle
    return route.call (** args)
  File "/usr/local/lib/python2.7/dist-packages/bottle.py", line 1740, in wrapper
    rv = callback (* a, ** ka)
  File "/home/pi/TerrariumPI/terrariumWebserver.py", line 43, in webserver_headers
    return fn (* args, ** kwargs)
  File "/home/pi/TerrariumPI/terrariumWebserver.py", line 81, in wrapper
    return func (* a, ** ka)
  File "/home/pi/TerrariumPI/terrariumWebserver.py", line 255, in __update_api_call
    result ['ok'] = self .__ terrariumEngine.set_config (path, postdata, request.files)
  File "/home/pi/TerrariumPI/terrariumEngine.py", line 915, in set_config
    update_ok = self.set_sensors_config (data)
  File "/home/pi/TerrariumPI/terrariumEngine.py", line 517, in set_sensors_config
    self .__ load_sensors (data)
  File "/home/pi/TerrariumPI/terrariumEngine.py", line 140, in __load_sensors
    self .__ UNIT_TYPE)
  File "/home/pi/TerrariumPI/terrariumSensor.py", line 73, in __init__
    self.update ()
  File "/home/pi/TerrariumPI/terrariumSensor.py", line 129, in update
    old_current = self.get_current ()
  File "/home/pi/TerrariumPI/terrariumSensor.py", line 342, in get_current
    indicator = self.get_indicator (). lower ()
AttributeError: 'NoneType' object has no attribute 'lower'

theyosh, 2018-02-25:
Do a git pull and try again? We are getting close to the working code

nke69, 2018-02-25:
HEllo Yosh,

this is the alert i have in screen : 

```
pi@PIFish:~/TerrariumPI $ screen -r
  File "/home/pi/TerrariumPI/terrariumSensor.py", line 183, in update
    current = round(sum(values[1,-1]) / 3.0,2)
TypeError: list indices must be integers, not tuple
2018-02-25 16:26:33,000 - INFO - terrariumWebserver - Running webserver at :::8090
list indices must be integers, not tuple
2018-02-25 16:26:35,183 - ERROR   - terrariumSensor    - Error updating sku-sen0161 ph sensor 'ph' with error:
Traceback (most recent call last):
  File "/home/pi/TerrariumPI/terrariumSensor.py", line 183, in update
    current = round(sum(values[1,-1]) / 3.0,2)
TypeError: list indices must be integers, not tuple
list indices must be integers, not tuple
2018-02-25 16:27:06,286 - ERROR   - terrariumSensor    - Error updating sku-sen0161 ph sensor 'ph' with error:
Traceback (most recent call last):
  File "/home/pi/TerrariumPI/terrariumSensor.py", line 183, in update
    current = round(sum(values[1,-1]) / 3.0,2)
TypeError: list indices must be integers, not tuple
The GPIO channel has not been set up as an OUTPUT
2018-02-25 16:27:07,047 - ERROR   - terrariumSensor    - Error updating hc-sr04 distance sensor 'Hauteur Eau' with error:
Traceback (most recent call last):
  File "/home/pi/TerrariumPI/terrariumSensor.py", line 157, in update
    GPIO.output(terrariumUtils.to_BCM_port_number(self.sensor_address['TRIG']), False)
RuntimeError: The GPIO channel has not been set up as an OUTPUT
list indices must be integers, not tuple
2018-02-25 16:27:37,887 - ERROR   - terrariumSensor    - Error updating sku-sen0161 ph sensor 'ph' with error:
Traceback (most recent call last):
  File "/home/pi/TerrariumPI/terrariumSensor.py", line 183, in update
    current = round(sum(values[1,-1]) / 3.0,2)
TypeError: list indices must be integers, not tuple
The GPIO channel has not been set up as an OUTPUT
2018-02-25 16:27:41,168 - ERROR   - terrariumSensor    - Error updating hc-sr04 distance sensor 'Hauteur Eau' with error:
Traceback (most recent call last):
  File "/home/pi/TerrariumPI/terrariumSensor.py", line 157, in update
    GPIO.output(terrariumUtils.to_BCM_port_number(self.sensor_address['TRIG']), False)
RuntimeError: The GPIO channel has not been set up as an OUTPUT
```

nke69, 2018-02-25:
New log after restart : 

```
2018-02-25 16:35:12,000 - INFO - terrariumWrapper - Starting TerrariumPI server at location: http://192.168.1.61:8090 ...
/home/pi/.local/lib/python2.7/site-packages/gpiozero/pins/pi.py:132: SPISoftwareFallback: failed to initialize hardware SPI, falling back to software (error was: failed to import spidev)
  'software (error was: %s)' % str(e)))
list indices must be integers, not tuple
2018-02-25 16:35:27,457 - ERROR   - terrariumSensor    - Error updating sku-sen0161 ph sensor 'ph' with error:
Traceback (most recent call last):
  File "/home/pi/TerrariumPI/terrariumSensor.py", line 183, in update
    current = round(sum(values[1,-1]) / 3.0,2)
TypeError: list indices must be integers, not tuple
2018-02-25 16:35:35,000 - INFO - terrariumWebserver - Running webserver at :::8090
list indices must be integers, not tuple
2018-02-25 16:35:36,753 - ERROR   - terrariumSensor    - Error updating sku-sen0161 ph sensor 'ph' with error:
Traceback (most recent call last):
  File "/home/pi/TerrariumPI/terrariumSensor.py", line 183, in update
    current = round(sum(values[1,-1]) / 3.0,2)
TypeError: list indices must be integers, not tuple

```

theyosh, 2018-02-25:
Ok, it looks like you have some mixed up code.  All errors about hc-sr04 should be fixed, due to the fix in the master branch. So stop the program and remove all the *.pyc files and make sure you have the latest code.

Because I did a small change yesterday late. So please update to latest version on the feature/ph branch. Remove the *.pyc files to make sure new .py files are used.

And see what the output will be

theyosh, 2018-02-25:
Just added some debug. Please do another git pull

nke69, 2018-02-25:
hum i have this : 

```
   GPIO.output(terrariumUtils.to_BCM_port_number(self.sensor_address['TRIG']), False)
RuntimeError: The GPIO channel has not been set up as an OUTPUT
Start measurement sku-sen0161
[]
Got voltage: 0.420615534929
[0.16806729321873473]
Got voltage: 0.418661455789
[0.16806729321873473, 0.16838215948644356]
Got voltage: 0.409868099658
[0.16806729321873473, 0.16838215948644356, 0.1683191862329018]
Got voltage: 0.42843185149
[0.16806729321873473, 0.16838215948644356, 0.1683191862329018, 0.1683191862329018]
Got voltage: 0.42550073278
[0.16806729321873473, 0.16838215948644356, 0.1683191862329018, 0.1683191862329018, 0.16817749641243282]
Sorted values
[0.16806729321873473, 0.16817749641243282, 0.1683191862329018, 0.1683191862329018, 0.16838215948644356]
list indices must be integers, not tuple
2018-02-25 16:48:40,061 - ERROR   - terrariumSensor    - Error updating sku-sen0161 ph sensor 'valeur P  H' with error:
Traceback (most recent call last):
  File "/home/pi/TerrariumPI/terrariumSensor.py", line 190, in update
    current = round((sum(values[1,-1]) / 3.0),2)
TypeError: list indices must be integers, not tuple
The GPIO channel has not been set up as an OUTPUT
2018-02-25 16:48:40,282 - ERROR   - terrariumSensor    - Error updating hc-sr04 distance sensor 'Hauteur Eau' with error:
Traceback (most recent call last):
  File "/home/pi/TerrariumPI/terrariumSensor.py", line 157, in update
    GPIO.output(terrariumUtils.to_BCM_port_number(self.sensor_address['TRIG']), False)
RuntimeError: The GPIO channel has not been set up as an OUTPUT
```

theyosh, 2018-02-25:
Another try? Updated the code again. Now I hope the errors will be gone

nke69, 2018-02-25:
git pull and git ph/mesure ./install and reboot  . i'm waiting ;)

theyosh, 2018-02-25:
oh... the install script is not needed now.... that was only once... 
we will wait

nke69, 2018-02-25:
i've got a Current PH to 0.17 ..
with my ither python script i've got 7.12 ?

what's the difference ?

theyosh, 2018-02-25:
I think the way the voltage is measured... so can you test with your working script what the output is of the variable `trim_pot1 = readadc( adcnum, SPICLK, SPIMOSI, SPIMISO, SPICS )`

I use a different library to read out the voltage. Could be that we have to invert the value

nke69, 2018-02-25:
ok I will look a little on my side what it is;) I come back to you for the rest .. thank you for your time for me

nke69, 2018-02-25:
Absolutely PERFECT ! 🥇 
Many thanks for your time and patience for this hack 
value is effectively correct now ;)

nke69, 2018-02-25:
I'm really happy that it works and offers you virtually a good bottle of old Scottish wisky :) health! :)

[image]

theyosh, 2018-02-25:
Thanks! And cool that it works. I will update my own installation as well to test if it is still working for me. I guess it will. Then it will be merged to the master branch.

The debug code will then also be removed. And maybe in the future I can do something with this on the environment part. For now it is just measurement.

Also big thanks to you for testing this and giving feedback. Together we fixed it in 24 hours with sleeping in between :) :+1:

theyosh, 2018-02-25:
Merged into master

closed 2018-02-25 by theyosh