theyosh/TerrariumPI #112: German update 2.0

Barbara1984, 2018-02-26:
Hi, sorry for the problems but if you have time you can explain to me with the pull requests again exactly?

Here is the german file
[German.zip](https://github.com/theyosh/TerrariumPI/files/1760378/German.zip)

Barbara1984, 2018-02-28:
So the update did not work?

theyosh, 2018-03-01:
Hi, yes, but I did not had time to publish it again. I am changing a bit the development path. And I have to think what I do with language updates.

But I will add it to the current version. I have it already on my new development version, but that is a different branch.

And maybe this can help for future pull requests. https://reflectoring.io/github-fork-and-pull/

The last one did not work, probably your version was out of date. So try to see if the section 'Updating your Fork' will work for you. I know that this is all developer talk. Because github is designed for coding stuff.

theyosh, 2018-03-01:
So your new update is now updated in my master branch. Thanks again.

Barbara1984, 2018-03-01:
I'll read through Link for me. If you know something about the progress of future language updates, please let me know, I will do more updates in the near future because there is a lot to fix :-)

closed 2018-03-01 by Barbara1984