theyosh/TerrariumPI #113: changing the default port 8090 by 80 ?

nke69, 2018-02-28:
Hi Yosh,
I wanted to change the port 8090 (default) of the web server by port 80.
So I was in the interface settings put port 80 and then restarted the server, clean the cache of the internet browser and erase the * .pyc file on the remote server.
But at the restart of the server the site was not accessible either on port 80 and also on the 8090.

I had to re-use my old settings.cfg and restart the raspberry again .. the config is distributed on port 8090.

Do I have to do anything else for the web server to answer on port 80?
I also specify that after putting the port 80 the relay no longer responds ..

theyosh, 2018-02-28:
My software will not run on ports lower then 1024. I use a normal user to run the software, and normal users are not allowed to open ports below 1024. And mostly 8080 is used for proxy servers so I use 8090.

Easy solution would be install NGINX and use this vhost config: https://github.com/theyosh/TerrariumPI/blob/development/contrib/nginx.conf

theyosh, 2018-03-01:
So, for now running on port 80, nginx is needed. The other problems you described, I guess that is a result of a not working webserver, and therefore a crippled system.

The NGINX vhost config and explanation will be in the next release

closed 2018-03-01 by theyosh

nke69, 2018-03-01:
Hi Yosh,

No problem for me and it's definitely not a todo;)
just a question for my personal culture :)
I usually work with "Apache", but I just made a port forwarding on the router .. easier for the lazy I am :)

kaju666, 2018-11-16:
> So, for now running on port 80, nginx is needed. The other problems you described, I guess that is a result of a not working webserver, and therefore a crippled system.
> 
> The NGINX vhost config and explanation will be in the next release

Cant find this explanation, after nginx instalation what to do with your .conf file?

theyosh, 2018-11-16:
@kaju666 you are right... documentation was lacking. Now the needed steps are here: https://github.com/theyosh/TerrariumPI/wiki/FAQ#use-and-enable-nginx-server-on-port-80 . And the link is also in the nginx.conf vHost file. So I hope that will be enough to get it running.

@nke69 I missed your remark. But have you it running with Apache? If so, could you share your used config for Apache vHost? Then I can add that to the contrib folder as well. Because I have not used it for more then 5 years now and 2.4 is a bit new to me :)