theyosh/TerrariumPI #115: Translation updates

theyosh, 2018-03-04:
Subscribe to this issue to get updates when the master branch is ready for translation.

theyosh, 2018-03-16:
Master branch is ready for translations.

Barbara1984, 2018-04-11:
Hello Theyosh, here is another german translation.
[German update.zip](https://github.com/theyosh/TerrariumPI/files/1900042/German.update.zip)

theyosh, 2020-01-27:
@nke69 @Barbara1984 @kebs19 @bfonn @kahuwi14 @penzoiders You all made some translations for me in the past. If you are still using my software, would it be possible to update the translations?

There is a new system for translation through a webbrowser at: https://weblate.theyosh.nl/engage/terrariumpi/ . So you do not need any external programs anymore!

I would really appreciate if you could all update your own language. Thanks in advanced

kahuwi14, 2020-01-27:
Hey there!
I just had a quick look at the tool and made some sentences for testing. 
It's a really handy way and I'm going to do some more translations in the next few days.

bfonn, 2020-01-31:
will fix the norwegian translation over the weekend

penzoiders, 2020-02-09:
Hi, sorry I'm quite near burnout in this period but I'll do my best to sort this out. meanwhile I'll subscribe to this thread.

fujexo, 2021-11-25:
[image]

Found a little issue with the german translation but I'm not sure how to fix it.
The watertank volume in english is `volume`, which can be translated to `Lautstärke` oder `Volumen` in German. While translating, I've checked the files which the word appears on and tranlated it to `Lautstärke` as it was shown for the audio sources.

I cannot find a way in weblate to give the same word a different translation based on the source file. Could you assist here?

theyosh, 2021-11-25:
Hmm, I am afraid that I have to change the original language first. I have used on different place the word volume, and yes, I do see that it can have multiple definitions and therefore different translations.

I think I will change the volume here, to 'water volume'. And maybe volume for audio to 'audio volume'. Then it should be clear.

theyosh, 2021-11-26:
Updated the translations and already translated to German ;)