theyosh/TerrariumPI #117: Memory-Usage in System-Status-View

kahuwi14, 2018-03-04:
Hello there!
I discovered the other day , that the memoryusage-warning appeared in the systemstatus-view.
After a little research it turned out, that there is not a big memory-usage of one or more applications.
The warning depends on the usage of linux itself: the os uses the free memory for buffering/caching.
@theyosh If you got freetime ;) you could perhaps change the warning-trigger into a little difference between total and used memory. 

[image]

theyosh, 2018-03-04:
Actually, there is a memory leak in my code. I have seen this earlier. When you drop the OS caches, a lot of memory is freed. But after a minute it is all taken again. And I do have graphs that does show some memory increment over time with my Python code. But I have no idea how to tackle memory leaks in Python yet :(

In the meantime I can see if this memory calculation can be done better / more real.

kahuwi14, 2018-03-04:
> Actually, there is a memory leak in my code.

I don't agree with that.. If you have a look on my screenshots, you'll see in the first line of the table behind the "free"-command, that I have 875 MB memory (plus 100 MB Swap), 105 MB are used and 194 MB free available. That meens, there is a "leak" of 576 MB and that's the size of the buff/cache.

> When you drop the OS caches, a lot of memory is freed. But after a minute it is all taken again.

And that fits perfectly to that theory.

I've found a good descripton in German and a less detailed one in English as well:
[https://www.raspberrypi.org/forums/viewtopic.php?f=31&t=26405#p239066](url)

So I think, the one and only fault is the warning-trigger.. ;)

theyosh, 2018-03-05:
Turns out that there was a better way of doing this: https://psutil.readthedocs.io/en/latest/#memory

closed 2018-03-05 by theyosh

kahuwi14, 2018-03-05:
Thumbs-up!

commit 4d5d61a0 (2018-03-05): Better memory usage reporting in status view. Close #117