theyosh/TerrariumPI #118: Log file

Barbara1984, 2018-03-04:
Can you tell me what that warning is?

2018-03-04 19:48:07,494 - WARNING - terrariumSensor    - Measured value NoneC from temperature sensor 'Mitte' is outside valid range 0.00C - 100.00C in 12.64397 seconds.

2018-03-04 19:49:45,721 - WARNING - terrariumSensor    - Measured value None% from humidity sensor 'Mitte' is outside valid range 0.00% - 100.00% in 12.63838 seconds.

theyosh, 2018-03-04:
I suspect that this is a DHTXX sensor. Those do have once in a while a problem with reading out the data. This happens after 12 seconds of waiting for information. 
I have it also. As long this happens not more than once a hour, it should not be a problem.

If you see them all the time, it could indicate on a faulty sensor or a bad connection.

Could it be that this is caused by adding a new switch on the same GPIO port? My software does not check that and could cause unknown problems.

theyosh, 2018-03-04:
Ok, but still this is a hardware issue. And I cannot fix that. So could it be that by adding the latest switch this is happening. And by removing that switch this is going to work again. Maybe disconnected something when changing the hardware?

I do not know how much devices on GPIO can be used with a Pi. It will also depends on the amount of power that is available through the power adaptor.

Barbara1984, 2018-03-04:
Unfortunately, the buttons Save does not work

theyosh, 2018-03-04:
Please give more information. Do you now mean the save button at the sensors page? I am willing to help, but please provide more information then just 'it does not work'.

If saving your SENSORS does not work, this is caused by the faulty DHTXX sensors. Either make sure it is working correctly of remove it from the SENSORS settings page and save

Barbara1984, 2018-03-04:
No, it's about the switch, I'll do that later, I'll keep writing to you in # 116.

kahuwi14, 2018-03-04:
Hello!
I got the same issue every 3-6 Minutes and I also think, that it depends on the DHT11. I'm using this sensor, until the DHT22 sensors arrive from China... ;)

offtopic: That's the reason for the little if-statement..

Barbara1984, 2018-03-04:
I use a DHT22 sensor with integrated resistance. Connected via a Cat. 6 line

kahuwi14, 2018-03-04:
Oh no... 
Do you have another sensor to try?

Barbara1984, 2018-03-04:
No, unfortunately not, why?

kahuwi14, 2018-03-04:
To be sure, that it's not a problem with the sensor itself. 

I hope mine will arrive soon, than I could test these and perhaps it's the sensor indeed.

Barbara1984, 2018-03-04:
I do not think it's the sensor, but I have to admit that I have not tested any other, but if you have a DHT11 that falls away, then I do not need to buy it

kahuwi14, 2018-03-05:
Finally, the Sensors arrived today!!! 
I'll be at home at the weekend and then, I'm going to test them!

Barbara1984, 2018-03-05:
Cool, please let me know.

theyosh, 2018-03-05:
@Barbara1984 it is always wise to have multiple sensors at home. I buy everything in a couple of 2 of 4 devices. Saves a lot of time when you can just swap a sensor and it works again.

Barbara1984, 2018-03-05:
Already ordered :-)

Natrixz, 2018-03-05:
My DHT22’s die alot...their crap if they get hit with a drop of water...altho coating them helped a bit.

Jelly-Flinger, 2018-03-06:
hi  I have this problem but I am able to get good data  from the command line. I am just in the middle of swapping all the sensors to 5v ti see if it helps.....

Jelly-Flinger, 2018-03-06:
I should mention I have a dht11 and a dht22 both have the same problem

theyosh, 2018-03-06:
I do have them also. But I am not sure why. And I do not have that much hardware knowledge.

For my own operational terrarium I do not use DHT sensors. I have everything through 1-wire. Which is more stable in my opinion.

But I do know that I do not close pins, and keep them always open/active. Maybe that could cause this because the hardware gets to less power or confused... To change this in code will take some time and a lot of testing... so at this time I am not sure that changing a lot of code will fix this issue.

theyosh, 2018-03-06:
So, other code using the same DHT library: https://github.com/adafruit/Adafruit_Python_DHT/blob/master/examples/google_spreadsheet.py#L98

It just does a read every second and with an error it just wait. The comment suggest that it is not a stable sensor... so I am afraid that this could be very hard to fix

Barbara1984, 2018-03-06:
Do you have a tip how (with which) I can measure with 1 wire sensors humidity and temperature. Or something more stable than DHT11 and DHT22.

theyosh, 2018-03-06:
Ok, found an alternative way of reading out DHT sensors. If you are known with git and branches, try the current development branch. It just got an update for reading out DHT sensors

Barbara1984, 2018-03-06:
hmm, unfortunately I do not know how I am now on the update, so I can not test it right now.

theyosh, 2018-03-06:
oh crap... that is true...

Barbara1984, 2018-03-06:
No matter it will certainly test another user, if I am more knowledgeable, I could also test it but well :-)

kahuwi14, 2018-03-07:
I tried the fix but with a DHT11, the values are absurd: The temperature was over 300C and the humidity over 1900%... 😅

theyosh, 2018-03-08:
@kahuwi14 if possible, try the latest development branch. There is now different code to read out DHT sensors

kahuwi14, 2018-03-10:
@theyosh I'm finally on the last commit of the dev-Branch!! :+1: 
Had some trouble with the Sensor (I'm now using a DHT22).
The restart doesn't work, I had to reboot the Pi to bring the sensor-values into the right area.
Before the reboot, it only shows -999% and -999C.. I don't know why, I tried every commit from #116 until now: in every commit the same fault.. :-/

theyosh, 2018-03-11:
Euh not clear what you are trying to tell. Both master and development are currently the same :) I have merged them yesterday. So with the latest pull from either the master or development branch you still have problems?

But now you should have a different way of reading out DHT22. If there is still issues, I can only say it has to be hardware. Because I do not have all those problems with my DHT22. But then, I only use one and with a short cable.

kahuwi14, 2018-03-12:
No no, I haven't problems any more... 
At first I had, but now it works fine!!
My problem was, that the sensor gave false values, but with the Adafruit-Examples, I had plausible values. But after a reboot of the Pi, it worked! :+1:

closed 2018-03-14 by Barbara1984