theyosh/TerrariumPI #120: Hummidity sensor issues

Jelly-Flinger, 2018-03-06:
Hi,

Thanks for all the great work so far!

Full disclosure, I am new to this so may very well have messed up somewhere!
I am having some issues with humidity sensors and one of my temperature probes.

**errors**
in the screen I see 
 four Warning lines for sensor (temp and H for both my Am2302 and Dht11) saying value none was returned. Prior to seitching from 3v to 5v I also saw **Warning Terrariumengine update took too long (50 seconds)**
One of my 3 dsb1820 is also sometimes erratic in the values returned (maybe fixed since I switched to 5v?).
I am sometimes successfully reading humidity from both sensors and temperature from the Am2302. The trends are running on the web interface and show periodic updates, maybe every hour or so.
I have tested from the shell and I can always successfully read from the humidity probes using command line. It takes maybe 5 seconds to return a value.

**Hardware**
I have
1) pi zero with 1A power supply
2) SparkFun Logic Level Converter (LLC) - Bi-Directional (I thought maybe with the cables 3v was not enough to power the sensors properly)
3) 3 dsb1820 (one of which is erratic) all on. Single wire via LLC on GPIO 4 with pull up resistor
4) one Am2302 (purchased on board with included 2.5k pull up resistor) via LLC on GPIO 3
5) one DHT11 (purchased on board with included pull up resistor) via LLC on GPIO 2
6) 4 channel relay (currently disconnected while I figure this out)
I have soldered up a prototype board with all this and checked for continuity.

theyosh, 2018-03-06:
Thirst thing I would try is a heavier power supply. I always run at 2A or higher adapters

theyosh, 2018-03-06:
And I have not tested with a pi zero. I have heard from others that it did work at their place. So I cannot say if a Pi zero could be the cause of this... I think it should not be.

theyosh, 2018-03-06:
Ok, found an alternative way of reading out DHT sensors. If you are known with git and branches, try the current development branch. It just got an update for reading out DHT sensors

Jelly-Flinger, 2018-03-07:
Thanks! 

My 3 x DS1820 are single wire. I did wonder about the size of the power supply so I will swap it out. 

Just reading up on git to learn how to get the development branch! I will report back soon.

theyosh, 2018-03-07:
If you used git to download this repository, just enter the folder TerrariumPI and do the following commands:

1. git pull
2. git checkout development
3. restart TerrariumPI

Jelly-Flinger, 2018-03-14:
Many thanks! Sorry for the delay in responding I got distracted making up an enclosure for all the electronics (did not want 240v cables flapping about!). 

Short version is DHT11 is not working and I suspect the software is trying to read it as a DHT22. 

OK so I have an updated version. I think from your other comments you have merged the humidity sensor updates into the main branch?

My DHT 22 is good. Looks to be regularly updating the trend with reasonable values. 

The DHT11 is no longer responding at all (it used to send correct data very intermittently). When I look in the log file I see crazy out of range values. 700C. 

However, both instruments are working from the command line.

Finally when I set the wrong instrument type for DHT11 in command line I get the crazy out of range values that appear to be an exact for the log file errors. 
 
**Command line**
pi@raspberrypi-Zero:~/Adafruit_Python_DHT/examples $ sudo ./AdafruitDHT.py 11 2
Temp=28.0*  Humidity=78.0%
pi@raspberrypi-Zero:~/Adafruit_Python_DHT/examples $ sudo ./AdafruitDHT.py 22 2
Temp=717.7*  Humidity=1996.8%

**Log file**
2018-03-14 12:16:20,344 - WARNING - terrariumSensor    - Measured value 717.7C from temperature sensor 'Ambient H Temp' is outside valid range 0.00C - 100.00C in 0.08347 seconds.

Jelly-Flinger, 2018-03-14:
Also is there a debugger you would recommend? I was trying to learn more about your code (an python in general). I tried to install webide with no success :-(

theyosh, 2018-03-15:
Ok, I was hoping that all DHT would be same regarding reading them out. Guess not... so I will revert some code back to fix that. 

As a debugger, I have no clue. In all my programming years I never used one :P. I debug just with echo commands and logging. So you could look in the logging.conf file to enable debug logging. But that is not the same as code debugging with an IDE

theyosh, 2018-03-15:
I do see now error message about voltage issues:

[Thu Mar 15 20:54:42 2018] Voltage normalised (0x00000000)
[Thu Mar 15 20:55:15 2018] Under-voltage detected! (0x00050005)
[Thu Mar 15 20:55:19 2018] Voltage normalised (0x00000000)

And it looks like this is happening when the DHT22 is being read out. I still think DHT sensors are fun for small robot projects.... but not that stable for 24/7 projects....

Natrixz, 2018-03-16:
Yeh the DHT's suck for this kinda thing...
Im currently trying a one-wire humidity/temp sensor wich at the moment works really well.
Have to see how it'll hold up

theyosh, 2018-04-01:
I have made a change in code that every DHTXX sensor will have a sleep time 2.1 seconds before being used. So it should not happen that the sensor will be requested to much.

When you have 1 sensors, you can add 2 times it in the software. Once for temperature, and once for humidity. But that means two read outs. And between read outs there should be 2 seconds or else the sensor can get in trouble. So it will take more time to read out the DHTXX sensor, but this is the last thing I can come up with....

theyosh, 2018-04-13:
@Jelly-Flinger do you have any updates? I think my 2 seconds delay for a DHTXX sensor did give a bit more stability and less warnings. But still I have sometimes read timeouts with DHTXX sensors.

For me, it is good enough

BashSer, 2018-04-13:
[image]
Uptime 60 hours + and warning less x3. Nice result !

theyosh, 2018-04-20:
Closing due to no response in 7 days

closed 2018-04-20 by theyosh