theyosh/TerrariumPI #121: Webcam still not working

Aapjes94, 2018-03-07:
So I know there have been some camera issues before. I did the `git pull` and update last night, but still no luck. I also cleared the cache.

`2018-03-07 15:28:15,972 - WARNING - terrariumWebcam    - Attempt 3 of 3 for getting raw for rpicam type 'RPIcam' did not succeed at location rpicam. Will retry in 1 second.
PiCameraError: Camera is not enabled. Try running 'sudo raspi-config' and ensure that the camera has been enabled.
"Camera is not enabled. Try running 'sudo raspi-config' "
File "/usr/lib/python2.7/dist-packages/picamera/camera.py", line 460, in _init_camera
self._init_camera(camera_num, stereo_mode, stereo_decimate)
File "/usr/lib/python2.7/dist-packages/picamera/camera.py", line 431, in __init__
with PiCamera(resolution=(self.resolution['width'], self.resolution['height'])) as camera:
File "/home/pi/TerrariumPI/terrariumWebcam.py", line 125, in __get_raw_image_rpicam
Traceback (most recent call last):`

I've already made sure the camera is enabled in the interface options within the `raspi-config` menu

theyosh, 2018-03-07:
Reboot pi? Make sure that the camera is well connected.

Aapjes94, 2018-03-08:
I tried rebooting the pi but that didn't help either. I did, however, manage to solve the issue. After some more time tinkering, it appears to be a hardware issue with the PiCam (or the ribbon cable). The status LED on the PiCam was flickering as I wiggled the connection. I can't really confirm as I don't have any spares on hand at the moment, but it seems obvious. 

Closed

closed 2018-03-08 by Aapjes94