theyosh/TerrariumPI #122: dehumidifiers

Barbara1984, 2018-03-07:
Theyosh, I think we've written about this topic before, is such an option in the system environment possible e.g. like the heating / cooling function, if 60% is set as the standard value, but 45%, a humidifier will switch on and above 60%, the dehumidifier will switch on and the humidifier will switch off. If you understand what a principle I mean.

Aapjes94, 2018-03-08:
You already asked this two months ago #67

Natrixz, 2018-03-08:
And if it's getting too humid, you mist/spray too much...would be better to fix that than to correct the mistake using other hardware

Barbara1984, 2018-03-08:
Hi, thanks for your reply I will try to get this problem in handle.

closed 2018-03-08 by Barbara1984

theyosh, 2018-03-08:
You can also set the spray timer to a lower value. So that it will also spray a shorter time.
That could result in spraying a few more times, but the humidity should not rise that much.

But, I do know that you have asked before, but I think I will implement a feature like this. But in a more general way.

I am thinking of adding an extra option to select a second part of power switches. And that way, you can have a sprayer when to dry, and dehumidifier when to high. Not sure when I will change it, because the heater and cooler part will also be changed then. But I have to think about it what this will do with current setups.

So for now, I think that you are helped by shorting the spray period. That way it will climb slower after a spray.

Barbara1984, 2018-03-15:
Thank you Theyosh for your answer, I am glad when you are ready with your idea