theyosh/TerrariumPI #123: Update crash

Aapjes94, 2018-03-15:
When I woke up today the website was no longer responding. SSH'ing into the pi gave me [this error](https://imgur.com/rLWrzIW). I shut down the software and rebooted the pi. It didn't start up the server again and gave me [this error](https://imgur.com/VRtjGoq)

After an hour of trying to solve this, I ended up reinstalling the whole thing. But do you perhaps know what could have caused this to happen in the first place?

theyosh, 2018-03-15:
Damm... that sucks...

I have seen this before and the cause was a faulting SD card. The long update timeouts could indicate that writing the sensor data to sqlite was not fast enough. And that could be caused by the SD card.

The errors in the second image are not related to my code. Those are system errors, which I think is also SD card related.

It could also a faulty file system. A fsck could maybe fixed it. Another option could be to just to delete the file history.db. Then a new one will be created. As the last line of the first image says that the db is malformed.

But I get the idea that there was a problem with the SD card itself or the filesystem. I do know that writing a lot to the SD is not good for the SD. So I use class10 or better cards. It will make it all faster, more stable and less prone to errors.

theyosh, 2018-03-15:
The second image shows that the file terrariumEngine.py could not be red from disk

Aapjes94, 2018-03-16:
I just double checked and it's a Class 10 SD card. I'll just hope it's a one-time thing and close the issue. If it happens again I'll reopen this.

closed 2018-03-16 by Aapjes94