theyosh/TerrariumPI #124: Adding Sensor Results from Serial (Arduino)

rrgrow, 2018-03-18:
<p><span style="color: #ff0000;"><em><strong>Note... </strong></em><em><strong>Not an Issue just a question..</strong></em></span></p><p>I've managed put this py script together from another project&nbsp;(probably porly) to receive data from the arduino mega through serial with multiple sensors readings split into pieces on one line then received by the Raspberry PI 3 on usb port x and insert into a sql database. I would run this script once a minute through crobtab.</p><h2>Hardware used for 2 water tanks:</h2><ol><li><strong>Arduino Mega&nbsp; x1</strong></li><li><strong>HC-SR04&nbsp; &nbsp;(DISTANCE) x2</strong></li><li><strong>DHT22&nbsp; &nbsp; &nbsp; &nbsp;(TEMP,HUM FOR HELPING DISTANCE CALILATIONS) x2</strong></li><li><strong>DS18B20&nbsp; &nbsp;(Water Temperature Sensor) x2</strong></li></ol><h2>Results from running Script in terminal:</h2><h2>1&nbsp; &nbsp; 46&nbsp; &nbsp;40.70&nbsp; &nbsp;24.50&nbsp; &nbsp;17.56&nbsp; &nbsp; &nbsp;2&nbsp; &nbsp; &nbsp;47&nbsp; &nbsp; 41.60&nbsp; &nbsp;24.00&nbsp; &nbsp;19.50</h2><h2>Tank 1</h2><table width="559"><tbody><tr><td>0</td><td>1</td><td>2</td><td>3</td><td>4</td></tr><tr><td>1</td><td>46</td><td><p>40.70</p></td><td>24.50</td><td>17.56</td></tr><tr><td>Sensor_1</td><td>Distance_1</td><td><p>&nbsp; Humidity_Distance_1</p></td><td>Temp_Distance_1</td><td>Water_Temp_1</td></tr></tbody></table><p>&nbsp;</p><h2>Tank 2</h2><table width="524"><tbody><tr><td>5</td><td>6</td><td>7</td><td>8</td><td>9</td></tr><tr><td>2</td><td>47</td><td><p>41.60</p></td><td>24.00</td><td>19.50</td></tr><tr><td>Sensor_2</td><td>Distance_2</td><td><p>&nbsp; Humidity_Distance_2</p></td><td>Temp_Distance_2</td><td>Water_Temp_2</td></tr></tbody></table><p>&nbsp;</p><p><strong>&nbsp;</strong></p><h2>Python Script:</h2><p>#!/usr/bin/env python<br />import datetime<br />import serial<br />import time<br />import MySQLdb as mdb</p><p>arduino= serial.Serial("/dev/ttyUSB0")<br />arduino.baudrate=9600</p><p>data = arduino.readline()<br />pieces = data.split("\t")<br />now = datetime.datetime.now()<br />con= mdb.connect('localhost','root','XXXXXX','water');</p><p>with con:<br /> cursor=con.cursor()<br /> cursor.execute("INSERT INTO level (date_time,sensor,level,hum,temp) VALUES (%s,%s,%s,%s,%s)",(now,pieces[0],pieces[1],pieces[2],pieces[3])) <br /> cursor.execute("INSERT INTO level (date_time,sensor,level,hum,temp) VALUES (%s,%s,%s,%s,%s)",(now,pieces[5],pieces[6],pieces[7],pieces[8])) <br /> cursor.execute("INSERT INTO temp (date_time,sensor,temp) VALUES (%s,%s,%s)",(now,pieces[0],pieces[4])) <br /> cursor.execute("INSERT INTO temp (date_time,sensor,temp) VALUES (%s,%s,%s)",(now,pieces[5],pieces[9])) <br /> con.commit()<br /> cursor.close()<br />print pieces[0],pieces[1],pieces[2],pieces[3],pieces[4],pieces[5],pieces[6],pieces[7],pieces[8],pieces[9]</p><p>&nbsp;</p><p>&nbsp;</p><h2>Arduino Sketch:</h2><p>// Include DHT Libraries from Adafruit<br />// Dependant upon Adafruit_Sensors Library<br />#include "DHT.h"</p><p>// Include NewPing Library<br />#include "NewPing.h"</p><p>// DS Library<br />#include &lt;OneWire.h&gt;<br />#include &lt;DallasTemperature.h&gt;</p><p><br />// Define Constants<br />#define DHTPIN_1 12 // DHT-22 Output Pin connection<br />#define DHTPIN_2 13 // DHT-22 Output Pin connection<br />#define DHTTYPE DHT22 // DHT Type is DHT 22 (AM2302)<br />#define TRIGGER_PIN_1 2<br />#define ECHO_PIN_1 3<br />#define TRIGGER_PIN_2 4<br />#define ECHO_PIN_2 5<br />#define MAX_DISTANCE 400<br />#define WATER_TEMP_1 6<br />#define WATER_TEMP_2 7</p><p><br />NewPing sonar1(TRIGGER_PIN_1, ECHO_PIN_1, MAX_DISTANCE);<br />NewPing sonar2(TRIGGER_PIN_2, ECHO_PIN_2, MAX_DISTANCE);<br />OneWire oneWire1(WATER_TEMP_1);<br />DallasTemperature watertemp1(&amp;oneWire1);<br />OneWire oneWire2(WATER_TEMP_2);<br />DallasTemperature watertemp2(&amp;oneWire2);</p><p>// Define Variables</p><p>float hum1; // Stores humidity value in percent<br />float temp1; // Stores temperature value in Celcius<br />float hum2; // Stores humidity value in percent<br />float temp2; // Stores temperature value in Celcius<br />float duration1; // Stores First HC-SR04 pulse duration value<br />float duration2; // Stores Second HC-SR04 pulse duration value<br />float distance1; // Stores calculated distance in cm for First Sensor<br />float distance2; // Stores calculated distance in cm for Second Sensor<br />int iterations = 5;<br />float Celcius1=0;<br />float Celcius2=0;</p><p>// Initialize DHT sensor for normal 16mhz Arduino</p><p>DHT dht1(DHTPIN_1, DHTTYPE); <br />DHT dht2(DHTPIN_2, DHTTYPE);</p><p>void setup() {<br /> Serial.begin (9600);<br /> dht1.begin();<br /> dht2.begin();<br /> watertemp1.begin();<br /> watertemp2.begin();<br /> <br />}</p><p>void loop()<br />{<br /> <br /> // Measure duration for first sensor<br /> <br /> duration1 = sonar1.ping_median(iterations);<br /> <br /> // Add a delay between sensor readings<br /> <br /> delay(10);<br /> <br /> // Measure duration for first sensor<br /> <br /> duration2 = sonar2.ping_median(iterations);<br /> <br /> delay(10); // Delay so DHT-22 sensor can stabalize<br /> <br /> hum1 = dht1.readHumidity(); // Get Humidity value<br /> temp1= dht1.readTemperature(); // Get Temperature value<br /> hum2 = dht2.readHumidity(); // Get Humidity value<br /> temp2= dht2.readTemperature(); // Get Temperature value<br /> watertemp1.requestTemperatures(); <br /> Celcius1= watertemp1.getTempCByIndex(0);<br /> watertemp2.requestTemperatures(); <br /> Celcius2= watertemp2.getTempCByIndex(0);<br /> // Calculate the distances for both sensors</p><p>long t1 = 0, h1 = 0, hp1 = 0, t2 = 0, h2 = 0, hp2 = 0, hpL1 = 0, hpL2 = 0;</p><p>// Waiting for pulse<br /> t1 = duration1 + (0.606 * temp1) + (0.0124 * hum1);<br /> t2 = duration2 + (0.606 * temp2) + (0.0124 * hum2);<br /> <br /> // Calculating distance 1<br /> h1 = t1 / 10;<br /> h1 = h1 - 50; // offset correction<br /> h1 = 30 - h1; // water height, 0 - 50 cm<br /> hp1 = 2 * h1; // distance in %, 0-100 %<br /> hpL1 = hp1*100/70/2; // Letre based on 70 litres at 100%<br /> <br /> // Calculating distance 2<br /> h2 = t2 / 10;<br /> h2 = h2 - 50; // offset correction<br /> h2 = 30 - h2; // water height, 0 - 50 cm<br /> hp2 = 2 * h2; // distance in %, 0-100 %<br /> hpL2 = hp2*100/70/2; // Letre based on 70 litres at 100%<br /> Serial.print("1");<br /> Serial.print("\t"); // for splitting<br /> // Sending to computer<br /> Serial.print(hpL1);<br /> Serial.print("\t"); // for splitting<br /> //Serial.print(" ");<br /> //Serial.print("H:");<br /> Serial.print(hum1);<br /> Serial.print("\t"); // for splitting<br /> //Serial.print("% ");<br /> //Serial.print(" C:");<br /> Serial.print(temp1);<br /> Serial.print("\t"); // for splitting<br /> // DS WATER SENSORS<br /> Serial.print(Celcius1);<br /> Serial.print("\t"); // for splitting<br /> //Serial.print(" ");<br /> Serial.print("2");<br /> Serial.print("\t"); // for splitting<br /> // Sending to computer<br /> Serial.print(hpL2);<br /> Serial.print("\t"); // for splitting<br /> //Serial.print(" ");<br /> //Serial.print("H:");<br /> Serial.print(hum2);<br /> Serial.print("\t"); // for splitting<br /> //Serial.print("% ");<br /> //Serial.print(" C:");<br /> Serial.print(temp2);<br /> Serial.print("\t"); // for splitting<br /> // DS WATER SENSORS<br /> Serial.print(Celcius2);<br /> Serial.println(" ");<br /> <br /> delay(1000);<br />}</p><h2><strong>My question:</strong></h2><p>I would like to be able to add a sensor from sensor setting page and select serial port(number), piece [number] with current setting already in place. I think this would really help add more harward to this project.&nbsp;&nbsp;</p><p>ie..</p><div class="x_content"><div class="col-md-2 col-sm-2 col-xs-12 form-group"><label for="sensor_[nr]_address">Address<span class="required">&nbsp;* = Serial</span></label></div><div class="col-md-2 col-sm-2 col-xs-12 form-group"><label for="sensor_[nr]_type">Type<span class="required">&nbsp;* = Piece</span></label></div><div class="col-md-2 col-sm-2 col-xs-12 form-group">Piece Number = 9 for example (Water Temp 2)</div><div class="col-md-2 col-sm-2 col-xs-12 form-group"><label for="sensor_[nr]_name">Name<span class="required">&nbsp;* = Water Temp 2</span></label></div><div class="col-md-1 col-sm-1 col-xs-12 form-group"><label for="sensor_[nr]_alarm_min">Alarm min<span class="required">&nbsp;* =&nbsp;</span></label></div><div class="col-md-1 col-sm-1 col-xs-12 form-group"><label for="sensor_[nr]_alarm_max">Alarm max<span class="required">&nbsp;* =</span></label></div><div class="col-md-1 col-sm-1 col-xs-12 form-group"><label for="sensor_[nr]_min">Limit min&nbsp;<span class="required">* =</span></label></div><div class="col-md-1 col-sm-1 col-xs-12 form-group"><label for="sensor_[nr]_max">Limit max&nbsp;<span class="required">* =</span></label></div></div>

theyosh, 2018-03-18:
Euh, is the cronjob that is gathering the data once a minute on the same Pi as where my software is running... then there is a very easy way to do it.

The following code could do it:

`with open(“/home/pi/TerrariumPI/static/sensors.json”) as sensordata:
  sensordata.write('{"temperature1": %s, "distance": %s}' % (pieces[0],pieces[1],pieces[2])) 
`
Extend it with all the data you measure. This will end up in a single file at the location:
/home/pi/TerrariumPI/static/sensors.json. Update it once a minute with your cron job.

Then you are able to add an external sensor according to: https://github.com/theyosh/TerrariumPI/wiki/Remote-data

Use in this example the following url: http://[Pi_IP:8090]/static/sensors.json#temperature1 for reading out one temperature sensor.
Repeat this for every sensor like: http://[Pi_IP:8090]/static/sensors.json#distance

I hope it is clear what I try to explain and helps you out

rrgrow, 2018-03-18:
Many Thanks I understand and follow completely in .
The cron is running on the same PI as your software yes.

I tested the py script with the added lines in terminal and get:

SyntaxError: Non-ASCII character '\x93' in file water.py on line 12, but no enco                                  ding declared; see http://python.org/dev/peps/pep-0263/ for details

some missing code or i'm placing the code in the wrong file.
Thanks again

rrgrow, 2018-03-19:
Works with edit below:

with open("/home/pi/TerrariumPI/static/sensors.json","r+") as sensordata: sensordata.write('{"temperature1": %s, "distance": %s}' % (pieces[4],pieces[6]))

theyosh, 2018-03-19:
Cool!

my code was just to give an idea. So you should extend it so it contains all your sensors. And I think that is not that hard to do.

rrgrow, 2018-03-20:
Great work!

Got that all sorted and working like a charm. Thanks for that I thought it would be a nightmare but it was quite easy.

I was wondering if its easy enough to add more tanks,light etc in System environment? one more tank  and that shows the level/Litre by default    as well as / instead of    when the sensor option is applied. I have a timer set for my pump to feed 15 on 15 off (works well but without Water Level/Distance indicator ) I have no refill needs in the way of automation not as yet anyway.
Be nice to have feature in System environment if possible or just a simple copy and paste and change names to suite etc..

closed 2018-03-20 by rrgrow

theyosh, 2018-03-21:
No, at the moment my software is for a single terrarium system. Sorry, but it will be a lot more complicated when you want to control more devices. So for now, it will not be.

rrgrow, 2018-03-22:
Thats fine I can deal with that..
other question: 
would it be easy enough to add 'door' to the type section in add sensors or 'remote' in add door sensor section.

theyosh, 2018-03-22:
Just did... 

 -git pull (on master branch)
- restart software
- clear browser cache

And it should be there... use 0 for closed and 1 for open

rrgrow, 2018-03-24:
Great work I did try myself to do the same thing but with no joy. it works fine just a little feed back for ya:
after you have submitted/added a new door then land back on door settings at anytime the hardware type displays GPIO not Remote as I'm not submitting any changes this doesn't matter but if I had for example 2-3 more doors listed and would like to add another I would have to change all hardware from GPIO back to Remote on all doors listed on page before submitting again. At a guess read status isn't there like in sensor settings.

**“Great minds think alike, though fools seldom differ.”**
[image]
[image]

I'm looking to get pH sensor(s) soon and have some sort of pH up/down dispenser system into the  tank(s) to automate pH levels. I may have to work in simple python and crontab for this.
Check pH level if to high then add 1 ml of pH down turn on circulation pump in tank(s) wait/sleep 10-20 mins then take pH reading again and so on.
Thanks again for your help I'm very happy with my system now. I have no doubt I'll be quizzing you again soon enough lol.

rrgrow, 2018-03-24:
I have made a quick and simple edit to profile.tpl adding a calendar widget from google using the url below and pasting it in just before footer: 
https://calendar.google.com/calendar/embedhelper

My Code:
<iframe src="https://calendar.google.com/calendar/embed?showTitle=0&amp;showPrint=0&amp;showCalendars=0&amp;showTz=0&amp;height=600&amp;wkst=2&amp;hl=en_GB&amp;bgcolor=%23FFFFFF&amp;src=YOURCALENDARNAME%40gmail.com&amp;color=%2329527A&amp;ctz=Europe%2FLondon" style="border-width:0" width="100%" height="600" frameborder="0" scrolling="no"></iframe>

 <br> <br>
YOURCALENDARNAME=(for privacy reasons  I've taken mine out but this should disply yours if selected in gen on page)

I changed the width to 100% to fit into browser window. Works really well and I can now set tasks/events/schedules  etc.. Just have to remember to select the calendar you want to see in the setting page from google otherwise your tasks/events/schedules will not show up. 
School boy error from myself.
also on mobile portrait doesn't resize but landscape does well.  This is something I can live with. 

Just wonted to share this with you as it was just so easy to do and you may like to use it.

rrgrow, 2018-03-29:
[image]

Nice.. You can see what issue I have with the sizing. I did try and change the containers size in css file but had on luck? Thanks for the update 👍

theyosh, 2018-03-29:
Clear browser cache... you have an old css file now loaded. In CSS the frame is 100% width and 400px height

rrgrow, 2018-03-29:
[image]

If you could help me please. I had this same issue since I installed the software about a month ago where the webcam dir gets full of jpg files and only shows a still shot on the front end. I have removed all files out the dir and refreshed to make sure the dir is still clear for a 1 min or so then refreshed / landed back on the webcam page. the dir is full up again to the exact file name ( de2b8d7f52f0d204cc7667a23b2a7902_tile_3_7_7.jpg ). it like the permissions are wrong and it cant stream some how?

nke69, 2018-04-04:
hello, I followed the procedure here above, I get my datas (Arduino) in a json from my python script, T °, distance, etc.

By cons when I insert the path of the json as remote temperature sensor I have this error:

2018-04-04 22: 07: 59,000 - INFO - terrariumWrapper - Starting TerrariumPI server at location: http://192.168.1.61:8090 ...
HTTPConnectionPool (host = '192.168.1.61', port = 8090): Max retries exceeded with url: /static/extern/sensors.json (Caused by NewConnectionError ('<requests.packages.urllib3.connection.HTTPConnection object at 0x6c366a90>: Failed to establish a new connection: [Errno 111] Connection refused ',))

An idea ?

[image]

theyosh, 2018-04-04:
Yes! This happens only during startup. The TerrariumPI webserver is the last part that is started. So the sensors are loaded before the webserver is loaded and cannot read the 'remote' data . And therefore, it will give a warning when you start the software, but in 30 seconds it should update normally and no errors should be given.