theyosh/TerrariumPI #126: No temperature display since the update

Barbara1984, 2018-03-20:
Theyosh, unfortunately it shows me since yesterday no more values in temperature and humidity. Could you help me?

[image]

Aapjes94, 2018-03-20:
Did you remember to clear the cache? What does the log file (system -> log) look like?
Next step would be to disconnect and add the sensors again, both the wiring and in the software.

Barbara1984, 2018-03-20:
Hi, my cache is empty and I have already deleted my sensor and stored again, but he shows me now in the log an error.

[image]

Aapjes94, 2018-03-20:
Are you using regular digital sensors on with GPIO? Have you tried shutting down the pi all together? Shut down TerrariumPi with 2 * ctrl-C within a screen session then run `sudo reboot`.  And if you haven't done so already, double check all the wiring and connections.

Barbara1984, 2018-03-20:
Thanks for your help, but unfortunately it does not work. I have now finished the screen session and restarted, the cable can not have anything because the pi including attachments were not moved. A normal DHT22 sensor is used. Do you have another tip?

[image]

Aapjes94, 2018-03-20:
Hmm then I don't really know. If this happened because of the update, you could try running the update again. Maybe something failed during the update which can be fixed by updating again?

theyosh, 2018-03-20:
It looks a bit strange... But I am missing some engine processes, so I guess something has crashed.

- First, could you see if a reboot would fix this?
- Then look inside the screen what is happening there. There are still some errors not being logged
- Just to be sure, run the installer script again?
- Latest option is to remove the settings.cfg files (make backup) and start a clear system

And could you upload your current config?

theyosh, 2018-03-20:
And are you are the master branch? Because I though you have tried to test some code... so make sure you are at the master branch when updating to a new version

theyosh, 2018-03-21:
Euh, your image only shows INFO and no errors. Nor do I see any sensors in the log.

The log is normal. Wrong image maybe??

Barbara1984, 2018-03-21:
Hi, I have now added my sensor but unfortunately no values ...

[image]

theyosh, 2018-03-21:
Could you please try the above options and give feedback? This is not helping.
Again, I do not see any terrariumEngine messages which indicate there are errors in the console. Please login with ssh and use screen -r to see what errors there are.

Barbara1984, 2018-03-22:
Hi, I have now tried everything, a new settings.cfg, because Pi restarted, because sensor reconnected, the install file again executed, but all this does not help unfortunately. In the SSH, the following is displayed:
[image]

Barbara1984, 2018-03-23:
Hi theyosh, I just looked at the PC and suddenly I have temperature values again although I have not done or changed anything since last night :-)

closed 2018-03-26 by Barbara1984

theyosh, 2018-03-26:
Strange... but cool! Daylight saving time fix??? :)