theyosh/TerrariumPI #127: missing translation in terrariumTranslations.py

nke69, 2018-03-21:
Hi Yosh,

I wanted to add them myself in the file terrariumTranslations.py but after reboot the server, it makes a mistake and does not restart the start.sh ..
I guess it's missing something else but I do not know where :)

WARNING - terrariumTranslations - No translation available for 'environment_field_lights_enable'
WARNING - terrariumTranslations - No translation available for 'environment_field_sprayer_enable'
WARNING - terrariumTranslations - No translation available for 'environment_field_heater_enable'
WARNING - terrariumTranslations - No translation available for 'environment_field_cooler_enable'

theyosh, 2018-03-21:
Euh... these messages are warnings and a reminder (for me) to fix them. But it should not crash your system.

These warnings are because the documentation is not fully up to date. And that will stay for a while, because I still have some features waiting that will change the way the software works, and will probably rewrite a lot of documentation. That is way I am currently neglecting the documentation and translations of those.

So I think for now, use `git checkout terrariumTranslations.py` to revert it back. And leaf it as is.

theyosh, 2018-03-26:
I close this, because it is not a real problem when the original terrariumTranslations.py is used again

closed 2018-03-26 by theyosh