theyosh/TerrariumPI #128: 2018-03-13-raspbian-stretch-lite.zip on rpi3 b

k3tw13z3l, 2018-03-25:
After install -> reboot The service is not available. As I try to start it manually I see the following  

pi@raspberrypi:~/TerrariumPI $ sudo ./start.sh run
2018-03-25 19:41:56,000 - INFO - terrariumWrapper - Starting TerrariumPI server at location: http://192.168.2.7:8090 ...
Traceback (most recent call last):
  File "/home/pi/TerrariumPI/terrariumPI.py", line 11, in <module>
    from terrariumEngine import terrariumEngine
  File "/usr/local/lib/python2.7/dist-packages/gevent/builtins.py", line 93, in __import__
    result = _import(*args, **kwargs)
  File "/home/pi/TerrariumPI/terrariumEngine.py", line 26, in <module>
    from terrariumWebcam import terrariumWebcam
  File "/usr/local/lib/python2.7/dist-packages/gevent/builtins.py", line 93, in __import__
    result = _import(*args, **kwargs)
  File "/home/pi/TerrariumPI/terrariumWebcam.py", line 7, in <module>
    import cv2
  File "/usr/local/lib/python2.7/dist-packages/gevent/builtins.py", line 93, in __import__
    result = _import(*args, **kwargs)
ImportError: libvpx.so.4: cannot open shared object file: No such file or directory
2018-03-25 19:41:59,000 - INFO - terrariumWrapper - Restart counter: 1/5

...etc.. till it stops with a 
INFO - terrariumWrapper - To manny error (6). Shutting down TerrariumPI server!

It seems like the camera stuff is the problem (my python is a bit non-existent...) The 'same' software seems to run on rpi1B 

The libvpx.so.4 can be found in 

$ dpkg -L libvpx4
/.
/usr
/usr/lib
/usr/lib/arm-linux-gnueabihf
/usr/lib/arm-linux-gnueabihf/libvpx.so.4.1.0
/usr/share
/usr/share/doc
/usr/share/doc/libvpx4
/usr/share/doc/libvpx4/AUTHORS.gz
/usr/share/doc/libvpx4/README.gz
/usr/share/doc/libvpx4/changelog.Debian.gz
/usr/share/doc/libvpx4/changelog.gz
/usr/share/doc/libvpx4/copyright
/usr/lib/arm-linux-gnueabihf/libvpx.so.4
/usr/lib/arm-linux-gnueabihf/libvpx.so.4.1

So, as I said my python is not something to be proud about, it either cannot find this file due to some missing path stuff somewhere in the python stack or it is a completely different file that is supposed to life somewhere in the python env.

Any help would be appreciated...

Aapjes94, 2018-03-26:
I've had the restart loop before as well. Installing it again always solved it for me. Either from a fresh install or just running the installer again.

theyosh, 2018-03-26:
You could try to rerun the installer. And do a reboot? Sounds like missing or not loaded libraries

k3tw13z3l, 2018-03-29:
Hm strange, you are right, but indeed a reinstall did not have these issues...

thanx

closed 2018-03-29 by k3tw13z3l