theyosh/TerrariumPI #129: i2c sensors, user schedules, alerts

BashSer, 2018-03-30:
Good day !

Do you plan to add i2c sensors support ? DHT is not verify and quality =( 
How did you get the idea to add more "user" schedules in environment ?
Need something for alarms. Example, file where alarm data will be recorded(leak detector, motion sensor...) and telegram bot can this data forward for user.

This is wonderful project, thank you very much !

Bug:
All sensor, switch and etc. need adding one at a time. If you simultaneously adding two sensors/switch/etc and press "Submit" the changes are not saved.

theyosh, 2018-04-01:
Hi,

There is support for non DHT hardware through OWFS or 1 wire. I personally use the following adapter:
http://www.sheepwalkelectronics.co.uk/product_info.php?cPath=22&products_id=30

and then this: http://www.sheepwalkelectronics.co.uk/product_info.php?cPath=23&products_id=65 as a hub.

And sensors like: http://www.sheepwalkelectronics.co.uk/product_info.php?cPath=23&products_id=55

But they have changed the PCB design and they are a bit bigger now :( . There could be other ways of doing it, but I am using there hardware for 5+ years now. And bought a lot so I do have enough spare to continue this way.

But I have also support for https://github.com/theyosh/TerrariumPI/issues/84 which works directly through I2C.

Scheduling is not something I am going to make soon. Time calculations are always hard to make right. There is a solution to embed an external calendar link (google calendar) to have a some kind of agenda in the system.
Notifications are still on the list. But that will also introduce a lot of programming. Due the what should be alerted or not. Through what channels. https://github.com/theyosh/TerrariumPI/issues/101

I will look into the bug... I do have an idea what is wrong.

BashSer, 2018-04-01:
Sensors: http://www.kandrsmith.org/RJS/Misc/Hygrometers/calib_many.html
Comare sensors and this is small percentage of "good" sensors works on i2c.

Schedule: I mean add extra user schedule environment. Example, i had two water tanks with different liquids and i need to water them from two at different times according to different schedules.

P.S. Sorry for my bad english

BashSer, 2018-04-01:
Alerts:
How about extra level in logging ?
Example, "Alerts" - this will include data about close\open door(in "door" i mean any switch - leak, motion, reed switch, IR flame detector and other), if var(temp) > temp_max_alert more then 10 min or temp > temp+40% and send photo from webcams (to see on flame in terrarium xD), checkbox for sensor when they trigger or  they value = max_value

theyosh, 2018-04-03:
The schedule request is maybe possible. You can add a timer to each power switch. This way you can specify when the power switch should be on and of. Play with the start and stop times together with the on and off duration.

The on and off time specify the time window in which the timer is operating. Lets say on at 10:00, off at 13:00 . That is 5 hours duration.
With a on duration of 10 minutes and off duration of 50 minutes, the power switch will toggle on for 10 minutes every hour. So that is 5 times.

There is no sensor support with this. So if this is not working for you, there is no such thing as scheduling power switches

BashSer, 2018-04-03:
[image]
I mean add button "add user schedule" and copy one of this with all configuration where user can another light(uv, ir, top light, side light for tall plant) and etc.

P.S. Popular sensors in delivery and as i take them i report py scripts for you =D
P.S. I begin little develop py bot for telegram. Later i upload this *if this can work*

BashSer, 2018-04-03:
Nice ! Thank you
In my pre-alpha version skeleton:
[image]
[image]
*sorry, i learn python 2 nights and i hope i can help you with my pre-junior level xD*

theyosh, 2018-04-04:
Update for your bot:

There is a JSON API which you could use to send commands. Replace https://terrarium.theyosh.nl/ with your own host.

https://terrarium.theyosh.nl/api/sensors
https://terrarium.theyosh.nl/api/switches
https://terrarium.theyosh.nl/api/doors
https://terrarium.theyosh.nl/api/webcams

etc. For a full list look at https://github.com/theyosh/TerrariumPI/blob/master/terrariumWebserver.py#L93
and
https://github.com/theyosh/TerrariumPI/blob/master/terrariumWebserver.py#L268

For making changes you need HTTP authentication. This is just a short info. I need to write the full API still so that other software can use it.

BashSer, 2018-04-11:
Hi @theyosh !

After two night i learn py, mysql, api json and write "alpha version" of bot =D
It is not friendly(but inside many comments) and i can't write part of code, because i do not have camera for example =(
Where i can send him ?

Sensors arrived in post office, but i again do not know when I can coding (after three or six days)

Thanks for project and enhancement !

BashSer, 2018-04-13:
May be, but i create account only for this project :D
Ok, later i can test sensors and give you code.
You need temp in C/F and humidity on exit ?

BashSer, 2018-04-19:
Python script for si7021 throught i2c. Include many comments =)
[si7021.zip](https://github.com/theyosh/TerrariumPI/files/1926246/si7021.zip)

Check and write for this sensor:
[image]
[image]

Sorry, for language barrier

BashSer, 2018-04-19:
Python script for htu21d:
[htu21d.zip](https://github.com/theyosh/TerrariumPI/files/1926358/htu21d.zip)

Check on this sensor:
[image]
[image]

P.S. My sensor broken or fake(most likely), therefore script has minor changes. In theory, htu21d = si7021 in measurments (see script above) or has address 0x80.

theyosh, 2018-04-20:
Cool, I will look into this... and see how and what I can use....

BashSer, 2018-04-21:
Sh*t, i coding on work win-notebook with "Code Writer"(i dont know what is this) and... i don't save my work...
Sorry, in the near future, I will not be able to restore this. 
But i can copypaste this:
https://www.raspberrypi-spy.co.uk/2016/07/using-bme280-i2c-temperature-pressure-sensor-in-python/
https://github.com/avislab/sensorstest/blob/master/BME280/BME280.py
https://github.com/adafruit/Adafruit_Python_BME280/blob/master/Adafruit_BME280.py - and Adafruit has script too, but adafruit module has address 0x77(usually 0x76). Sorry.

theyosh, 2018-04-21:
I can figure this out with all the links and code. thanks a lot

theyosh, 2018-04-22:
Hi, turned out that the si7021 and htu21d working the same way as the already supported sht2xx. The code is exactly the same.

The only difference is the timing of the chips. Some do read it faster then others... but the si7021 and htu21d is supported now. But I have to warn, I do not have hardware to test it.

BashSer, 2018-04-22:
Wow, you faster :D
I can test this on workability without longest measurment . Because my solder got stuck in post office and my sensors without pins (on photo above) :D
I will be home ~00:00 (UTC+3)

BashSer, 2018-04-22:
This not my script, but i comment this for easy navigation in datasheet:
[bme280.zip](https://github.com/theyosh/TerrariumPI/files/1935482/bme280.zip)
[image] 
_photo from http://www.avislab.com/blog/bme280_ru/_

It is all sensors where i can ordered based on discussions in internet often used i2c sensors with fine measurment.
My choose is si7021, because they has needed functional and heater for dry dew-point(but i can't start this :D) 
Thank you very much , @theyosh !

theyosh, 2018-04-22:
and bm280 support is added.... I cannot test it... show you should do that if you can.

BashSer, 2018-04-22:
[image]
@theyosh it is you Flash ? 0_o
After 5 hours you implement this xD
I hope after work I will check this all at home, if I fail to check today, then I'll check tomorrow.

BashSer, 2018-04-26:
Good day, @theyosh !

Sorry to keep you waiting, i am not superhero as you. I waiting your job and appreciate your results.
I really tried tested your work, but i am still waiting soldering iron and can't do long measurments and try "crash" scripts(special symbols, escape character and etc...) or compatiable this between sensors and scipts. _There is a suspicion that they are equally suitable for all_

Good news:
1. It is all works !
I tested si7021(blue board) with choose "SI7021" - all works correctly and fast without warnings or another problem on bus 0 and bus 1.
Tested BME 280 with choose "BME280/BMP280" - all works corectly and fast without warnings or another problems on bus 1.
2. Very good idea with autofill field "address" !
At first I did not understand how to enter bus and address on bus and entered just the address on the bus - and autofill made this works :D Need edit hint for this field for user-friendly.
3. New install interface works good and do not explode my system after update as last time ! :D

Faults detected:
[image]
_4 string_
_sorry for quality, too long story with work and access_
As i understand this string in "screen -r" responsible for this sensor and do not give info about action.
Another "problem strings" probably because all another sensor not connected and do not worry about them.

"Problems with me":
Because my soldering iron still somewhere in world post i tested this by closing(short circuit) the contacts in my hands :D _but this works_
[image]

Good developments for your project ! Thank you very much !

BashSer, 2018-04-26:
And this telegram bot, anyway, i can do nothing more with him:
[bot_menu_api.zip](https://github.com/theyosh/TerrariumPI/files/1949666/bot_menu_api.zip)

Just input:
string 22: Usernames who can chat with bots;
string 24: token bot;
string 26 and 27: login and pass for authentication api Terrarium Pi;
and type in terminal "python bot_menu_api.py" or "nohup python bot_menu_api.py &"

I cant works on it because i dont know how configure proxy and this bot cant work.
_Russia try block telegram_ 😞

theyosh, 2018-04-30:
Fixed a small bug. Do a `git pull` and restart. It should then work.

The right address is 40,1 for your sensor.

theyosh, 2018-04-30:
Hmm, this is strange. The only difference I can find is the read timeout. I use way shorter timeouts.

So could you try to edit the file `terrariumI2CSensor.py` at line 124 and increase the value for `__HUMIDITY_WAIT_TIME`

My timing is based on the PDF data sheet. Could be that a bit more time is needed. I use about 13 ms. Where your script is waiting for 300 ms. So you could set it also to 0.3 in the `terrariumI2CSensor.py` file and see it that does do the trick

BashSer, 2018-05-03:
Yes, @theyosh  - you right.
I edit this file after update and restart TerrariumPi and all works without problem.
In line 124 i change 0.013 to 0.13.

theyosh, 2018-05-05:
Cool. I have changed also in code...

theyosh, 2018-05-07:
Then I would suggest to close this issue for now. Alerting will be done in issue #101 . I will add your bot code also in the contrib folder

BashSer, 2018-05-08:
Wow, thank you ! I hope someone can finish him and make international and universal version :D
Good luck you, your project and your gecko phelsuma ! =)

closed 2018-05-08 by BashSer