theyosh/TerrariumPI #130: dht11 on pi zero or pi 1B

k3tw13z3l, 2018-03-31:
LS,

I've been trying to get a dht11 sensor to work on a PI Zer0 WH, unfortunately it is not working and I'm at a loss why...

The strange thing is the SD card works perfectly in a pi3...

on the PIZer0 the example scripts work:
`$ cd Adafruit_Python_DHT/examples`
`$ ./AdafruitDHT.py 11 4 `
`Temp=20.0*  Humidity=55.0%`

If I enter pin 7 in (GPIO4) in the web interface it logs:

`Measured value NoneC from temperature sensor 'Temp' is outside valid range 0.00C - 100.00C in 14.95756 seconds.`

The code looks the same in Adafruit and TerrariumPI (I've even changed the code to hardcode type and pin and retry) but it keeps failing to retrieve a value from the sensor and so the `current=None` 

Anybody any pointers or ideas?

theyosh, 2018-04-01:
Hi,

sorry as you can see here: https://github.com/theyosh/TerrariumPI/issues?utf8=%E2%9C%93&q=is%3Aissue+is%3Aclosed+dht

There is a lot of problems with DHT sensors. It could be that it has to do with my code or not. But the bottom line is that I have it working with a DHT22 for a year now, and that is stable Therefore I conclude that my code is solid and all DHTXX sensors are hardware related.

So, I am not able to help you further. I have invested a lot of time in these issues with the same conclusion: DHTXX sensors are not stable! and should NOT be used! https://github.com/theyosh/TerrariumPI/issues/120

k3tw13z3l, 2018-04-01:
Yeah I saw those tickets. I was hoping someone had figured or found out why it is so unstable (or not even working on the old PI and the Zero) Your code works fine on my pi3 but not on the zero I had planned for the terrarium, and I have no clue what this difference could cause...

theyosh, 2018-04-02:
The last code change I did could help (1 month ago). But also timing of the internal bus and power to the Pi can be of influence.

But I am closing this. As I cannot fix hardware problems with software

closed 2018-04-02 by theyosh

k3tw13z3l, 2018-04-23:
For future readers:

If I run the software as root instead of the pi user it will work.

I've not yet figure out why the rpi1 and the rpi0 do not work. the pi user is member of the gpio group so I must be missing something somewhere else...

Not a problem for me (using root) but something to keep in mind.

silentkilla, 2018-05-03:
I also had to run ./start.sh as root to get my AM2302 working, im a real noob at raspberry pi/linux this so it took a long frustrating time to get this far including giving root a password etc... and im sure if i restart its going to break again.

theyosh, 2018-05-03:
If you change the rights on the TerrariumPI folder to root, the software will start as root next reboot.

`sudo chown root /home/pi/TerrariumPI`

That should make the software run with root user.

silentkilla, 2018-05-04:
Thanks theyosh and k3tw13z3l, yes i am also running a Zero W to save as much power as i can.

silentkilla, 2018-05-06:
I've had to repeat the same 'sudo chown root /home/pi/TerrariumPI' after upgrading to 3.5.0 for what its worth. Also i noticed that on my iphone im getting "A problem repeatedly occurred" error around the same time i put the rights change in place, but its not affecting my friends Samsung. Wiped all the data on the phone for the page etc but no effect.