theyosh/TerrariumPI #131: Too "hardcore" update

BashSer, 2018-04-02:
In log after "update":
```
  Aptitude 0.8.7: log report
Mon, Apr  2 2018 04:41:02 +0300

  IMPORTANT: this log only lists intended actions; actions which fail
  due to dpkg problems may not be completed.

Will install 0 packages, and remove 48 packages.
171 MB of disk space will be freed
========================================
[REMOVE, NOT USED] cgroup-bin:armhf 0.41-8
[REMOVE, NOT USED] cgroup-tools:armhf 0.41-8
[REMOVE, NOT USED] evince:armhf 3.22.1-3+deb9u1
[REMOVE, NOT USED] evince-common:armhf 3.22.1-3+deb9u1
[REMOVE, NOT USED] fonts-droid-fallback:armhf 1:6.0.1r16-1.1
[REMOVE, NOT USED] fonts-noto-mono:armhf 20161116-1
[REMOVE, NOT USED] geoclue-2.0:armhf 2.4.5-1
[REMOVE, NOT USED] gnome-user-guide:armhf 3.22.0-1
[REMOVE, NOT USED] iio-sensor-proxy:armhf 2.0-4
[REMOVE, NOT USED] libavahi-gobject0:armhf 0.6.32-2
[REMOVE, NOT USED] libcgroup1:armhf 0.41-8
[REMOVE, NOT USED] libcupsfilters1:armhf 1.11.6-3+rpi1
[REMOVE, NOT USED] libcupsimage2:armhf 2.2.1-8
[REMOVE, NOT USED] libdjvulibre-text:armhf 3.5.27.1-7
[REMOVE, NOT USED] libdjvulibre21:armhf 3.5.27.1-7
[REMOVE, NOT USED] libevdocument3-4:armhf 3.22.1-3+deb9u1
[REMOVE, NOT USED] libevview3-3:armhf 3.22.1-3+deb9u1
[REMOVE, NOT USED] libgeoclue-2-0:armhf 2.4.5-1
[REMOVE, NOT USED] libgnome-desktop-3-12:armhf 3.22.2-1
[REMOVE, NOT USED] libgs9:armhf 9.20~dfsg-3.2+deb9u1
[REMOVE, NOT USED] libgs9-common:armhf 9.20~dfsg-3.2+deb9u1
[REMOVE, NOT USED] libgxps2:armhf 0.2.4-1
[REMOVE, NOT USED] libijs-0.35:armhf 0.35-12
[REMOVE, NOT USED] libjavascriptcoregtk-3.0-0:armhf 2.4.11-3+rpi1
[REMOVE, NOT USED] libjbig2dec0:armhf 0.13-4.1
[REMOVE, NOT USED] libkpathsea6:armhf 2016.20160513.41080.dfsg-2
[REMOVE, NOT USED] libmbim-glib4:armhf 1.14.0-1
[REMOVE, NOT USED] libmbim-proxy:armhf 1.14.0-1
[REMOVE, NOT USED] libmm-glib0:armhf 1.6.4-1
[REMOVE, NOT USED] libnautilus-extension1a:armhf 3.22.3-1+deb9u1
[REMOVE, NOT USED] libpaper-utils:armhf 1.1.24+nmu5
[REMOVE, NOT USED] libpaper1:armhf 1.1.24+nmu5
[REMOVE, NOT USED] libpoppler-glib8:armhf 0.48.0-2+deb9u2
[REMOVE, NOT USED] libpoppler64:armhf 0.48.0-2+deb9u2
[REMOVE, NOT USED] libqmi-glib5:armhf 1.16.2-1
[REMOVE, NOT USED] libqmi-proxy:armhf 1.16.2-1
[REMOVE, NOT USED] libspectre1:armhf 0.2.8-1
[REMOVE, NOT USED] libwebkitgtk-3.0-0:armhf 2.4.11-3+rpi1
[REMOVE, NOT USED] libwnck-3-0:armhf 3.20.1-3
[REMOVE, NOT USED] libwnck-3-common:armhf 3.20.1-3
[REMOVE, NOT USED] libyelp0:armhf 3.22.0-1
[REMOVE, NOT USED] modemmanager:armhf 1.6.4-1
[REMOVE, NOT USED] poppler-data:armhf 0.4.7-8
[REMOVE, NOT USED] yelp:armhf 3.22.0-1
[REMOVE, NOT USED] yelp-xsl:armhf 3.20.1-2
[REMOVE] epiphany-browser:armhf 1:3.8.2.0-0rpi28
[REMOVE] epiphany-browser-data:armhf 1:3.8.2.0-0rpi28
[REMOVE] gnome-desktop3-data:armhf 3.22.2-1
========================================

Log complete.

===============================================================================

Aptitude 0.8.7: log report
Mon, Apr  2 2018 04:43:47 +0300

  IMPORTANT: this log only lists intended actions; actions which fail
  due to dpkg problems may not be completed.

Will install 12 packages, and remove 0 packages.
1636 kB of disk space will be used
========================================
[UPGRADE] libraspberrypi-bin:armhf 1.20180313-1 -> 1.20180328-1
[UPGRADE] libraspberrypi-dev:armhf 1.20180313-1 -> 1.20180328-1
[UPGRADE] libraspberrypi-doc:armhf 1.20180313-1 -> 1.20180328-1
[UPGRADE] libraspberrypi0:armhf 1.20180313-1 -> 1.20180328-1
[UPGRADE] libssl1.0.2:armhf 1.0.2l-2+deb9u2 -> 1.0.2l-2+deb9u3
[UPGRADE] libssl1.1:armhf 1.1.0f-3+deb9u1 -> 1.1.0f-3+deb9u2
[UPGRADE] openssl:armhf 1.1.0f-3+deb9u1 -> 1.1.0f-3+deb9u2
[UPGRADE] raspberrypi-bootloader:armhf 1.20180313-1 -> 1.20180328-1
[UPGRADE] raspberrypi-kernel:armhf 1.20180313-1 -> 1.20180328-1
[UPGRADE] raspberrypi-sys-mods:armhf 20180315 -> 20180328+1
[UPGRADE] rc-gui:armhf 1.15 -> 1.16
[UPGRADE] rpd-icons:armhf 0.14 -> 0.15
========================================

Log complete.

===============================================================================

Aptitude 0.8.7: log report
Mon, Apr  2 2018 04:47:48 +0300

  IMPORTANT: this log only lists intended actions; actions which fail
  due to dpkg problems may not be completed.

Will install 1 packages, and remove 0 packages.
74.8 kB of disk space will be used
========================================
[INSTALL] python-rpi.gpio:armhf 0.6.3~stretch-1
========================================

Log complete.

===============================================================================
```

After this i lose software and can't re-install, for example:
```
pi@raspberrypi:~ $ sudo apt-get install epiphany-browser
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 epiphany-browser : Depends: libavahi-gobject0 (>= 0.6.22) but it is not installable
                    Depends: libgnome-desktop-3-12 (>= 3.17.92) but it is not installable
                    Depends: libwebkitgtk-3.0-0 (>= 1.10.1) but it is not installable
                    Depends: libwnck-3-0 (>= 2.91.6) but it is not installable
                    Recommends: yelp but it is not installable
                    Recommends: evince but it is not installable
                    Recommends: cgroup-bin but it is not installable
E: Unable to correct problems, you have held broken packages.
```

This is not working on raspberry pi 2 and above:
`[INSTALL] python-rpi.gpio:armhf 0.6.3~stretch-1`

For test PWM i use:
```
git clone https://github.com/metachris/RPIO.git --branch v2 --single-branch 
cd RPIO/
sudo python setup.py install
```
*And this work before update =)*

P.S. I created a backup 20 minutes before update ;)

theyosh, 2018-04-02:
Hi,
just tested it on a Raspberry Pi 2 Model B Rev 1.1. No problems.

apt show python-rpi.gpio
Package: python-rpi.gpio
Source: rpi.gpio
Version: 0.6.3~jessie-1
Maintainer: Ben Croston <ben@croston.org>
Installed-Size: 103 kB
Depends: libc6 (>= 2.7), python (<< 2.8), python (>= 2.7~), python:any (>= 2.7.5-5~)
Homepage: http://sourceforge.net/projects/raspberry-gpio-python/
Priority: optional
Section: python
Download-Size: 23,5 kB
APT-Manual-Installed: yes 
APT-Sources: http://archive.raspberrypi.org/debian/ jessie/main armhf Packages
Description: Python GPIO module for Raspberry Pi
 This is the RPi.GPIO Python module, for supporting GPIO on a
 Raspberry Pi

So package is available. Try again running the installer.

BashSer, 2018-04-02:
Sorry, propably i mixed up it with rpio for PWM through DMA.
But, why "git pull, install.sh" remove so much packets ?)

theyosh, 2018-04-03:
My software expects a clean PI, and will remove all obsolete software that my code does not need.

This is done to speed up installations and updates. I do not need desktop software, so all office software is removed. Also Wolfram Alpha is removed. Those are big packages and take only time and space.

if you read the installation file: https://github.com/theyosh/TerrariumPI/blob/master/install.sh

you can see what is removed. If you do not want that, comment out the remove line

BashSer, 2018-04-03:
Thank you for answer !

closed 2018-04-03 by BashSer