theyosh/TerrariumPI #133: Soil humidity sensor

JohnnyB1989, 2018-04-06:
Hi,

thanks for your great great work. Love it!

I was wondering if you could implement a setting for a sensor that checks the humidity of the soil.

Like the YL-69 (https://www.amazon.de/YL-69-Feuchtigkeitssensor-Hygrometer-Bodenfeuchte-oder/dp/B00QHENH3K)

Thanks and keep on,

Johnny

theyosh, 2018-04-07:
Hi,

this is possible, but it depends how you want to use it.

This is a sensor that when digital connected it will only switch from wet to dry and back. So no information about how wet.

In analog modus, and therefore you need more hardware like http://rasp.io/analogzero/. Then you can read out a value between 0 and 100%. But this is not directly the moisture value. So then measure the resistance, and at a certain value, you have to conclude it is to dry.

Third, the device itself has also a resistor which can be tweaked so that it will light up when to dry. But this will also affect the data in the software. This will change the resistor value.

So, could you explain how you want to use it? Because of my experience with those sensors, is that reading them out analog, does not really give value information. You are only able to see when in the future it will get to dry based on historical values. But you will never get the real actual value.

And keep in mind, they are cheap, and will wear out in 6 - 12 months is my experience

Aapjes94, 2018-04-07:
This is exactly what I use the SHT2x sensor for, which is already supported, see #84.

JohnnyB1989, 2018-04-07:
Hi,

it would be okay, if the sensor tells when the soil is dry at a specific point, which I could adjust by how deep the sensor is in the soil. If the sensor gives a signal a switch is turned on for x seconds, so that the pump and the amount of water can be adjusted by the time.

Aapjes94, 2018-04-11:
Okey, I hope you haven't ordered them yet. The ones I ordered at least weren't any good. One corroded and the other one shorted within a day.

BashSer, 2018-04-11:
They all rust and only way to use them is turn on only for measurment or reduce acidity. Because if they all time turn on - electricity passes through "acid" soil and rust is formed.

P.S. I heard that sensor made of two nails :D

Aapjes94, 2018-04-11:
I was using something like [this](https://www.ebay.com/itm/1-PCS-FS200-SHT20-Soil-Moisture-Sensor-Meter-Humidity-Temperature-Sensor-GL-/162776413155) which I was expecting to actually be able to leave in the soil.

theyosh, 2018-05-02:
Hmmm.... sensor is not working well... does not change in state after almost a week... :(

JohnnyB1989, 2018-05-07:
Thank you so much, I will give it a try!

theyosh, 2018-05-07:
Ok, figured out that calibrating this sensor is a bit trail and error. So I got it working now a bit better. But still not sure how stable it will be.
Extra info about saving your sensor can be read here: https://github.com/theyosh/TerrariumPI/wiki/Hardware#moisture-ytxx-in-dgital-modus

Aapjes94, 2018-05-31:
I know it's only been two weeks, but how is the Chirp holding up?

theyosh, 2018-05-31:
Not that good... It looks like I do not get it stable somehow. Not sure what it is. Could be to long cables. Bad soldering from my part. So I just started testing with a second one. Bought 4 of them :)

Code is working, but I do not get is stable right now somehow

Natrixz, 2018-05-31:
Maybe moisture on/over the soldering? Try coating it :)
Conformal coating seems to work wonders on my humidity sensors

Aapjes94, 2018-05-31:
Natrixz, What sensors are you using?

Natrixz, 2018-05-31:
I use these:
https://taaralabs.eu/1-wire-humidity-temperature-sensor/
With the HIH-5031 and conformal coating. (Basically coat everything except the small hole) and shrink wrap it. Aditionally you could put hot glue inside the shrink wrap but i thought that was over-doing it lol

theyosh, 2018-05-31:
These can also be coated, but then I mis the light sensor: https://www.tindie.com/products/miceuz/i2c-soil-moisture-sensor/

Natrixz, 2018-05-31:
Why? You could just not coat the light sensor (conformal is clear coating btw) or just order the rugged version?:P or just coat the contacts? Lol

dasolan, 2018-06-03:
I found  "[Flora Monitor](https://www.china-gadgets.de/xiaomi-mi-pflanzensensor/)", its BT but has sensors for moisture, lux, EC and temperature. Price ranges about 11 € at Ali. 
There are already existing libarys here at [git](https://github.com/open-homeautomation/miflora). What do you think?

dasolan, 2018-06-03:
Maybe [MiFlora ](https://www.china-gadgets.de/xiaomi-mi-pflanzensensor/) is an alternative?
Its a Bluetooth sensor for 11 Euro at Ali, it detects moisture, lux, temperature and ec (conductivity).
Librarys still existing on git.

theyosh, 2018-06-04:
@dasolan I am afraid it will not supported. The MiFlora library is Python 3.X only. And my software is python 2.7 only still. So that is not compatible....

Also, it requires an extra bluetooth dongle, because the Pi does not have Bluetooth LTE onboard. (Maybe Pi3 does). So this is currently a bit to much right now to support

theyosh, 2018-06-04:
@dasolan Created a new issue #156 . So then I can close this issue

theyosh, 2018-06-10:
I am closing this issue because the original hardware is supported and working

closed 2018-06-10 by theyosh

oldmanbluntz, 2023-09-22:
Sorry to comment on a closed issue, but i didn't want to open a new issue for a quick question. Using the YL-69 through an MCP3008, how do you go about reading multiple YL-69 sensors as separate sensors through the ADC, or do you need one MCP2008 per sensor?