theyosh/TerrariumPI #134: Webinterface on Phone/Tablet

Natrixz, 2018-04-08:
Has anyone except me noticed that the webinterface crashes on phones and tablets?
Not sure what's causing it.

I checked a few things, but atleast it's not the calander that causes it.
(or am I the only person who noticed this?)

Natrixz, 2018-04-09:
Clearing cache doesnt work, it also happens on a brand new, never before used device.
Yep its while loading the front page; it shows very briefly, crashes, reloads and then the browser bitches about crashing several times >,<

k3tw13z3l, 2018-04-10:
Seems to be happening on safari 11.1 on HighSierra OSX 10.13.4 after the OSX update of last week too, at least in my case... 

Clearing cache does not help. Firefox FTW...

:o)

theyosh, 2018-04-10:
hmmm sounds more like an issue with Apple software....

But using a different browser does help?

Natrixz, 2018-04-10:
Haven't tested firefox yet, Chrome doesn't work on my Apple Tablets and Phones

theyosh, 2018-04-10:
I only have an old Apple iPad mini wiht iOS 9.5.X. And there is no problem. And I do not own Android devices. So I can only test on linux desktop, or linux mobile phones

Natrixz, 2018-04-10:
I think it happend on the most recent update, because it used to work just fine...
Shit >.<

k3tw13z3l, 2018-04-10:
Ha! the iPad hasn't  been updated yet, and it still works!

So it seems the update will break safari (where is the changelog..) Damn proprietary stuff!

:o)

The gentelella demo side at colorlib has no problems in my safari on OSX

theyosh, 2018-04-12:
Ha, I have a free Browserstack account to test this! And there I do see the same error. And I think I can test this, and maybe figure out what is causing this.

I guess it has to do with websockets

theyosh, 2018-04-12:
Hmm... this is nice :P. When I have in Mac OS Safari the developer window open, it all works!
When you close the developer window, it will crash on drawing the gauges on the frontpage or the sensor pages.

It looks like handling of the gauges has been the problem in this case. 

Firefox ad Chrome do not have this problem. So I am thinking, that this is an Apple problem. They have already issues with Display drivers and iOS has also display problems. Apple has done some changes there.... could be that some canvas rendering is not working anymore.

I leave it as is. Just download a different browser and it works..

Natrixz, 2018-04-13:
Not on phone and/or tablet.
Neither Chrome nor FireFox fixes the issue on them.
Not sure on OSX since I just own phones and tablets with that OS

k3tw13z3l, 2018-04-13:
On OSX firefox and Chrome work..

sorry... 

:o)

theyosh, 2018-04-13:
Hmm... so for iOS it is not working what browser you take.

I still have the opinion that this is really an Apple issue with there iOS....  sorry for Apple users...

Natrixz, 2018-04-14:
Not to be an ass, but this is so far the only website I encountered on the Tablets/Phones that crashes

theyosh, 2018-04-18:
I found out that it looks like that temperature and humidity graphs are giving problems. Weather graphs don't do..?? 

So I am still looking into this, but takes some more time to figure out what is causing this.....

theyosh, 2018-04-20:
Ok, this is fixed. But still this is caused by Apple and not me nor my software. I had to disable smoothing of the graphs when there are more then one line in the graph.

That is a 100% rendering problem of the device/hardware. Not my software. For now, I have disabled smoothing for all graphs except weather. Because that is a single graph

closed 2018-04-20 by theyosh

Natrixz, 2018-04-20:
Nice work :) 
Altho sucks that Apple (again) creates crap software/devices -,-

k3tw13z3l, 2018-04-20:
Ah thanx! Will try when I get home...

commit 2e30c01e (2018-04-20): Fix stupid Apple rendering bug! Fix #134