theyosh/TerrariumPI #135: Adding pH event in the Environment part ?

nke69, 2018-04-09:
Hi Yosh,

In the environment part would it be possible to add the pH, in the same way as the heating, water level, etc.
This to configure actions such that if the pH is low the CO² reactor starts, on the other hand if the pH is too high then the reactor co2 stops, this via their addition in the part switches.

Cordially.

theyosh, 2018-04-09:
Hi,

yes and no :P As more people ask for a more flexible environment system, I have decided to rewrite that part of the code. This will take some time, but then you can just create unlimited environment parts. So then you can just make a combination of sensors and power switches.

Till then, this pH request has to wait. I guess 2 till 4 weeks max.

nke69, 2018-04-10:
Hello,

That's excellent news ;)
In any case it is a pleasure to be part of this "adventure" for the greater good of our little protected they crawl, walk or swim :)

nke69, 2018-04-12:
Hi Yosh,

I continue in the current subject:
In future environments, will it be possible for example that when a door sensor is open it triggers an action on a relay, for example the activation of a lamp?

I do it now via a python script and it works well, I thought it would be easier to manage everything via the interface :)

theyosh, 2018-04-13:
That kind of trigger could be done. Will be added to the wish list :)

theyosh, 2018-05-02:
Ok, I added the pH to the environment system. Rewrite of code takes more time...

The trigger actions will be done later on... That will also introduce a lot more of configuration settings...

theyosh, 2018-05-05:
So if the environment part is working for you, I would like to close this ticket for now.

nke69, 2018-05-05:
Sorry, I forgot this ticket ..
the environment works perfectly, the CO² is cut as soon as the pH is too high and inversely ..
Thanks again for this work theyosh done;)

closed 2018-05-05 by nke69