theyosh/TerrariumPI #137: OpenCV issue

nke69, 2018-04-10:
Hello,
Last night I made the update 3.4.2 'git pull' and clear browser cache, etc.
But at reboot I have this error message:
I think it comes from the soft part for the webcam but impossible for me to see further;)

OpenCV Error: The operation is neither 'array op array' (where arrays have the same size and the same number of channels),
nor 'array op scalar', nor 'scalar op array') in arithm_op, file /build/opencv-U1UwfN/opencv-2.4.9.1+dfsg1/modules/core/src/arithm.cpp,
line 1287 /build/opencv-U1UwfN/opencv-2.4.9.1+dfsg1/modules/core/src/arithm.cpp:1287: error: (-209) The operation is neither 'array op array'
(where arrays have the same size and the same number of channels), nor 'array op scalar', nor 'scalar op array' in function arithm_op

nke69, 2018-04-10:
I just made the update.
I no longer have the error message in the shell, the webcam takes again the screens with or without the function archive .. and the interface works without problem ..

When I got this error message, the software did not stop, the interface either. On the other hand, the temperature of the CPU has increased by 4-5 degrees. After the update the temperature has returned to normal, it is + - 55 °.

Thank you ;)

theyosh, 2018-04-10:
Ok,

could you test the archive function? Just enable it, and wait some time. There is a small archive icon on the webcam controls. When you click it, it should load archive images it there are any. This is stil a bit beta, but I think it is a cool feature

theyosh, 2018-04-10:
Now I got them also... And I think I know where to look for.

It is related to the archive function. And cheap webcams :) I have a old HD ready usb webcam that works well. But now I am testing with a cheap Chinese cam and that give this problem also.. So I will try to figure this out...

theyosh, 2018-04-10:
Nope... it has to do with image rotations and image sizes... device is not the problem. It is a code issue :(

nke69, 2018-04-10:
I have this model of camera: https: //uk.pi-supply.com/products/raspberry-pi-camera-board-v1-3-5mp-1080p

The archive function is activated and I do not have the error message any earlier, but the saved photos have green areas that remain when you save the photo, here is an example;)

[image]

nke69, 2018-04-10:
it does it only on the archived photos, not on the active photo of the webcam

nke69, 2018-04-10:
same issue after play with option in camera archive function :

OpenCV Error: Sizes of input arguments do not match (The operation is neither 'array op array' (where arrays have the same size and the same number of channels), nor 'array op scalar', nor 'scalar op array') in arithm_op, file /build/opencv-U1UwfN/opencv-2.4.9.1+dfsg1/modules/core/src/arithm.cpp, line 1287
/build/opencv-U1UwfN/opencv-2.4.9.1+dfsg1/modules/core/src/arithm.cpp:1287: error: (-209) The operation is neither 'array op array' (where arrays have the same size and the same number of channels), nor 'array op scalar', nor 'scalar op array' in function arithm_op

theyosh, 2018-04-10:
Yeah, it is still beta. The idea is that every time new image is taken, it is compared with the previous image. And when there are differences, it saves the image for archive. The green borders are there to show where the difference was. In the future I may that optional. But for now, I think it looks cool.

Because just saving a image very few minutes could end up with a lot of still images. Now you got only the images with movement. Which is nicer to look at.

In the mean time, the bug is fixed. So now the archive function should also work when the image resolution changes.

theyosh, 2018-04-10:
I just just just updated the code... another git pull... and it should be gone

nke69, 2018-04-10:
Yes everything is ok,
in any case this feature is really cool, it's great because taking pictures can also do a timelapse, I love it :)

closed 2018-04-10 by theyosh