theyosh/TerrariumPI #139: Day/night difference heater

puyt, 2018-04-12:
Is it possible to make the day/night difference work with the light timer instead of the weather calendar? 

My lights are set to `timer` from 07:30 AM to 11:30 PM so this is my day time for my terrarium but heater will adjust it's range when it's around 8:30 PM

theyosh, 2018-04-13:
Yup, just did.

Do a `git pull` and restart the software. Clear browser cache and go to the environment page and select either weather or lights to determine day and night.

puyt, 2018-04-15:
Something doesn't work yet, the range is correct atm (will have to check again tonight) but the heater isn't turning on.. See my screenshots..

[image]

[image]

theyosh, 2018-04-15:
Heater will NOT run when lights are ON. That is specified and correct.

If the heater should be running 24/7 enable it by clicking yes option at 'enabled when lights are on'