theyosh/TerrariumPI #140: Heater issue in system environment

nke69, 2018-04-14:
Hello,

I have a small concern with the heating in the system environment, indeed I would like my heater is triggered when the values of my sensors are below the values that I determined. that's what happens, but as soon as the sensor arrives in the right range, the heating continues to be on in the environment and I have to cut it manually in the switches.
would I miss something or is it a bad setup on my part?
[image]
[image]

nke69, 2018-04-14:
I found ;)
The heater stops as soon as it exceeds the range for which I attribute the values.
I just play on a smaller range so that it cuts faster;)

closed 2018-04-14 by nke69

theyosh, 2018-04-14:
Yes that is correct. Else the switch will toggle to often and jumps around the lower alarm level. This could harm the electronic equipment