theyosh/TerrariumPI #142: Turn off switch if sensor can't collect data

BashSer, 2018-04-16:
[image]
~22:30-01:30 one sensor disconnected from vcc and environment schedule(heater) continue work and for 3 hours temp up on 10C degrees. If i did not check this before going sleep this can fired...
Can you add check ?
If int(temp/hum data = none) > 20 then Turn off switch else pass
This can save from flood or fire

theyosh, 2018-04-18:
Hi,

I understand your issue. I need to think about this. There are multiple solutions... so let me think about this for a few days

BashSer, 2018-04-19:
Nice, thank you !

theyosh, 2018-05-02:
Ok, then I close this issue.

closed 2018-05-02 by theyosh