theyosh/TerrariumPI #146: TerrariumPi dies when adding HC-SR04

Natrixz, 2018-05-01:
Hey Yosh,

Maybe I messed up something but whenever I add A HC-SR04 TerrariumPi dies of all together.
replaced  resistors with both 1k and 2k as with 10k and 4.7k (currently using this combo) however It appears to just randomly die.
Pi itself is reachable, however TerrariumPi has died of without any form of error.

Any idea's/troubleshooting tips? :)

theyosh, 2018-05-02:
Euh.... not realy.

I usually try `dmesg`
But this sounds like a segfault of somekind. Only my software dies? The Pi itself not?

The only thing I can think of is to try the sensor on a different Pi with custom code. But not sure if you can make custom code. A suggestion is to look for a simple tutorial where they connect the sensor to a Pi. For example: https://electrosome.com/hc-sr04-ultrasonic-sensor-raspberry-pi/

If this will also not work, you could almost say that the sensor is broken. Or you could by another one an just try that one. If that works, you also can conclude that the old one is broken.

If there is no more error messages, it is hard to figure it out, and I would then look at a hardware issue more then software.

Natrixz, 2018-05-03:
I am going to close this, can't figure out in dmesg what's wrong with the thing.
Ordered a few more "different" HC-SR04's.

Apperently they all work, untill I plug in a red (SparksFun) one....that one kills the software and just doesn't work. Both of my two sparksfun one's crash it.
All other brands that I have spare work.

So warning; Don't buy SparksFun HC-SR04 lol

closed 2018-05-03 by Natrixz