theyosh/TerrariumPI #148: Tmpfs for logs

Natrixz, 2018-05-09:
Hey Yosh,

So my first SD-card died this morning.
After this happend i decided to give the SD-card a much longer life by changing the log directory.

In etc/fstab I changed the directory to tmpfs, making it recide in memory.
Downside is that any reboot Will empty the log directory.(and you have a little less memory Free to use)
Upside however is that you wont constantly be writing to the sd-card. (And its fast)

Might be a good option to “ask” (yes/no choice) in the installer?

Unless I missed Some obvious down side?

theyosh, 2018-05-11:
Hm, this is not a bad idea to make. I will change it a bit.

I think a nice solution is that writing of the logfile should be done on a tmpfs but, when it is rotated, move it to a non tmpfs location. So you will keep your rotated logs after a reboot. And will only loose the active logfile during reboot.

I think this should not be a question, but just a normal way of working

Natrixz, 2018-05-12:
Yeh that would be really Nice :)
That way there is Some way to grab old logs when something happens but you dont kill the SD card very quickly :P

theyosh, 2018-05-13:
Just for checking. Do you have also a mount point at:

tmpfs            88M   24K   88M   1% /run/user/1000

Natrixz, 2018-05-14:
Yep, altho I used the $uid because I created a different user instead of pi :)
So mine is 1001.

Reason I did it because it gets deleted every time there is no active sessions left (and re-written as soon as there is one again)

Natrixz, 2018-05-14:
Altho honestly I reckon it could be done without it.
It usually only writes running processess stuff there (that's beeing run with systemd)

theyosh, 2018-05-14:
My solution is that I check if

/run/user/$UID of the user that is running the software.

At my test environment it works well. So I am not sure what you mean with your latest comment.

Natrixz, 2018-05-14:
I meant if it was alot of work lol
so ignore the comment since it's already done/working haha

Natrixz, 2018-05-15:
Appears to work over night for my test pi, now installed (removed my own tmpfs symlinks) on my actual terrariumpi.
Seems to work fine :) 👍

theyosh, 2018-05-16:
Hmm... logrotate does brake it now....? Not sure why... After logrotate the symlink to the tmpfs is gone... have to look into it a bit more

Natrixz, 2018-05-16:
Well, without really looking at/into the code; my guess would be that you set the symlink on the file and then with the logrotate "move" (and rename) the file.
Hence the symlinked file is gone and it creates a new file that's not symlinked

theyosh, 2018-05-16:
agree..  sounds logic and valid

theyosh, 2018-05-16:
And... guess what.... I made my own logfile rollover...totally  forgot about it... last line is.... you will guessed it...

`os.remove(...)`  ;)

Natrixz, 2018-05-16:
Hehe, yeah was just gonna look into the code if I could find something to cause the broken link xD
os.remove will do it lol

Altho that could mean it could be a simple fix =)

Natrixz, 2018-05-22:
Ps, your most recent fix still doesn't fix the log rotation issue ;)
Symlink still go's away

theyosh, 2018-05-23:
yes, I had the same. I fixed in the development version. https://github.com/theyosh/TerrariumPI/blob/development/terrariumLogging.py

If you use the development version of the logging file it should work. Do not switch to development branche yet. It is currently being tested for a new environment system.

Natrixz, 2018-05-24:
I used that file, but it still doesn't work.

theyosh, 2018-05-26:
Damm, strange. For me it is working. Could it be that symlink is created as root and not as the running user? Else I have no clue...

theyosh, 2018-05-28:
Is it possible to do some tests with the development branch? Are you running it in a live environment or a second test environment? Because the development branch should be ready to production, but I cannot test all scenario's. So If you are able to test your own home setup, then I know at least some more.

Natrixz, 2018-05-29:
It's possible yes.
Im running it in both =)

They both seem fine at the moment

theyosh, 2018-05-31:
Ok, new release is out. So I think this issue should be fixed now. Could you try to update and see if it is all working?

Natrixz, 2018-05-31:
Ps so far it all works out fine :)

closed 2018-05-31 by Natrixz