theyosh/TerrariumPI #149: errors after updating

rrgrow, 2018-05-12:
I still get error when updating.. it happens over the last 4-6 updates but I would just reboot and run install again and it completes update but this morning my setup didn't turn lights on and after reboot no TerrariumPI start up. I get:

pi@raspberrypi:/TerrariumPI $ screen -r
There is no screen to be resumed.
pi@raspberrypi:/TerrariumPI $ sudo ./start.sh
2018-05-12 14:28:11,000 - INFO - terrariumWrapper - Starting TerrariumPI server running as user 'pi' at location: http://192.168.1.2:8090 ...
pi@raspberrypi:~/TerrariumPI $ screen -r
atch_all
_notify_patch(events.GeventWillPatchAllEvent(modules_to_patch, kwargs), _warnings)
File "/usr/local/lib/python2.7/dist-packages/gevent/monkey.py", line 167, in _notify_patch
notify_and_call_entry_points(event)
File "/usr/local/lib/python2.7/dist-packages/gevent/events.py", line 112, in notify_and_call_entry_points
subscriber = plugin.load()
File "/usr/lib/python2.7/dist-packages/pkg_resources/init.py", line 2290, in load
self.require(*args, **kwargs)
File "/usr/lib/python2.7/dist-packages/pkg_resources/init.py", line 2307, in require
items = working_set.resolve(reqs, env, installer)
File "/usr/lib/python2.7/dist-packages/pkg_resources/init.py", line 849, in resolve
raise DistributionNotFound(req, requirers)
pkg_resources.DistributionNotFound: The 'cffi>=1.11.5' distribution was not found and is required by the application
2018-05-12 14:28:15,000 - INFO - terrariumWrapper - Restart counter: 1/5
2018-05-12 14:28:15,000 - INFO - terrariumWrapper - Restarting in 10 seconds after running for 4 seconds. Press Ctrl+C now to terminate TerrariumPI.
10 9 8 7 6 5 4 3 2

and while running install.sh:
Co9mand "/usr/bi /python -u -c "import setuptools, tokenize;file='/tmp/pip-build-Uvt81E/cffi/setup.py';f=getattr(tokenize, 'open', open)(file);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, file, 'exec'))" install --record /tmp/pip-X4YltI-record/install-record.txt --single-version-externally-managed --compile" failed with error code 1 in /tmp/pip-build-Uvt81E/cffi/

rrgrow, 2018-05-12:
maybe some system config files changed.

theyosh, 2018-05-12:
needs a new dev package: libffi-dev

quick fix: `sudo apt install libffi-dev` and rerun the installer

will add to the installer

rrgrow, 2018-05-12:
Many thanks it worked 👍 
I had to:
`git reset --hard ` as the file terrariumWebserver.py  even after deleting wouldn't git pull update.
I also had to reinstall `pip install pyserial` for the use of serial (arduino) :
I'm having problems running the crontab line below:

`* * * * *      cd /home/pi/TerrariumPI/static && /usr/bin/python /home/pi/TerrariumPI/static/water.py`
this worked up until now and should still.
water.py has my code to populate the sensors.json and works when I command in terminal but crontab isn't running it maybe the  /usr/bin/python command 
I have checked to see if it in the directory and it is. I have also:
  `sudo /etc/init.d/cron restart`
with no luck. have you any ideas

rrgrow, 2018-05-12:
Many thanks it worked 👍
I had to:
git reset --hard as the file terrariumWebserver.py even after deleting wouldn't git pull update.
I also had to reinstall pip install pyserial for the use of serial (arduino) :

rrgrow, 2018-05-12:
Many thanks it worked 👍
I had to:
`git reset --hard` as the file terrariumWebserver.py even after deleting wouldn't git pull update.
I also had to reinstall `pip install pyserial` for the use of serial (arduino) :

theyosh, 2018-05-14:
Ok, added pyserial to the pip installer list.

theyosh, 2018-05-14:
Can we close this issue?

rrgrow, 2018-05-15:
I still get a error after update and I cant seem to sort it I've everything:

2018-05-14 22:29:50,000 - INFO - terrariumWrapper - Starting TerrariumPI server at location: http://:8090 ...
Traceback (most recent call last):
File "/home/pi/TerrariumPI/terrariumPI.py", line 12, in 
from terrariumWebserver import terrariumWebserver
File "/usr/local/lib/python2.7/dist-packages/gevent/builtins.py", line 96, in import
result = _import(*args, **kwargs)
File "/home/pi/TerrariumPI/terrariumWebserver.py", line 12, in 
from bottle.ext.websocket import GeventWebSocketServer
File "/usr/local/lib/python2.7/dist-packages/gevent/builtins.py", line 96, in import
result = _import(*args, **kwargs)
File "/usr/local/lib/python2.7/dist-packages/bottle.py", line 1797, in load_module
import(realname)
File "/usr/local/lib/python2.7/dist-packages/gevent/builtins.py", line 96, in import
result = _import(*args, **kwargs)
File "/usr/local/lib/python2.7/dist-packages/bottle_websocket/init.py", line 2, in 
from .server import GeventWebSocketServer
File "/usr/local/lib/python2.7/dist-packages/gevent/builtins.py", line 96, in import
result = _import(*args, **kwargs)
File "/usr/local/lib/python2.7/dist-packages/bottle_websocket/server.py", line 4, in 
from geventwebsocket.handler import WebSocketHandler
File "/usr/local/lib/python2.7/dist-packages/gevent/builtins.py", line 96, in import
result = _import(*args, **kwargs)
File "/usr/local/lib/python2.7/dist-packages/geventwebsocket/init.py", line 17, in 
from .resource import WebSocketApplication, Resource
File "/usr/local/lib/python2.7/dist-packages/gevent/builtins.py", line 96, in import
result = _import(*args, **kwargs)
File "/usr/local/lib/python2.7/dist-packages/geventwebsocket/resource.py", line 5, in 
from .exceptions import WebSocketError
File "/usr/local/lib/python2.7/dist-packages/gevent/builtins.py", line 96, in import
result = _import(*args, **kwargs)
ValueError: bad marshal data (unknown type code)
2018-05-14 22:30:33,000 - INFO - terrariumWrapper - Restart counter: 1/5
2018-05-14 22:30:33,000 - INFO - terrariumWrapper - Restarting in 10 seconds after running for 43 seconds. Press Ctrl+C now to terminate TerrariumPI.

theyosh, 2018-05-15:
This looks like a broken sd card. This is not caused by my software.

Have to tried: `dmesg` to see if there are errors?

Or rerun the installer. But I think your SD card is dying

rrgrow, 2018-05-15:
pi@raspberrypi:~ $ dmesg
[ 9874.709848] w1_master_driver w1_bus_master1: Family 0 for 00.bd0000000000.cf                                                                                        is not registered.
[ 9912.595442] w1_master_driver w1_bus_master1: Attaching one wire slave 00.7d00                                                                                       00000000 crc 05
[ 9912.598098] w1_master_driver w1_bus_master1: Family 0 for 00.7d0000000000.05                                                                                        is not registered.
[ 9963.225328] w1_master_driver w1_bus_master1: Attaching one wire slave 00.fd00                                                                                       00000000 crc 89
[ 9963.230094] w1_master_driver w1_bus_master1: Family 0 for 00.fd0000000000.89                                                                                        is not registered.
[10001.345391] w1_master_driver w1_bus_master1: Attaching one wire slave 00.0300                                                                                       00000000 crc e2
[10001.350167] w1_master_driver w1_bus_master1: Family 0 for 00.030000000000.e2                                                                                        is not registered.
[10064.865617] w1_master_driver w1_bus_master1: Attaching one wire slave 00.8300                                                                                       00000000 crc 6e
[10064.870505] w1_master_driver w1_bus_master1: Family 0 for 00.830000000000.6e                                                                                        is not registered.
[10092.426567] w1_master_driver w1_bus_master1: Attaching one wire slave 00.4300                                                                                       00000000 crc a4
[10092.429454] w1_master_driver w1_bus_master1: Family 0 for 00.430000000000.a4                                                                                        is not registered.
[10143.105813] w1_master_driver w1_bus_master1: Attaching one wire slave 00.c300                                                                                       00000000 crc 28
[10143.110611] w1_master_driver w1_bus_master1: Family 0 for 00.c30000000000.28                                                                                        is not registered.
[10219.466044] w1_master_driver w1_bus_master1: Attaching one wire slave 00.2300                                                                                       00000000 crc c1
[10219.470804] w1_master_driver w1_bus_master1: Family 0 for 00.230000000000.c1                                                                                        is not registered.
[10282.985737] w1_master_driver w1_bus_master1: Attaching one wire slave 00.a300                                                                                       00000000 crc 4d
[10282.992088] w1_master_driver w1_bus_master1: Family 0 for 00.a30000000000.4d                                                                                        is not registered.
[10346.505955] w1_master_driver w1_bus_master1: Attaching one wire slave 00.6300                                                                                       00000000 crc 87
[10346.511392] w1_master_driver w1_bus_master1: Family 0 for 00.630000000000.87                                                                                        is not registered.
[10386.904570] w1_master_driver w1_bus_master1: Attaching one wire slave 00.e300                                                                                       00000000 crc 0b
[10386.909381] w1_master_driver w1_bus_master1: Family 0 for 00.e30000000000.0b                                                                                        is not registered.
[10450.384055] w1_master_driver w1_bus_master1: Attaching one wire slave 00.1300                                                                                       00000000 crc 7f
[10450.388827] w1_master_driver w1_bus_master1: Family 0 for 00.130000000000.7f                                                                                        is not registered.
[10499.863750] w1_master_driver w1_bus_master1: Attaching one wire slave 00.9300                                                                                       00000000 crc f3
[10499.868453] w1_master_driver w1_bus_master1: Family 0 for 00.930000000000.f3                                                                                        is not registered.
[10538.903388] w1_master_driver w1_bus_master1: Attaching one wire slave 00.5300                                                                                       00000000 crc 39
[10538.908008] w1_master_driver w1_bus_master1: Family 0 for 00.530000000000.39                                                                                        is not registered.
[10590.943050] w1_master_driver w1_bus_master1: Attaching one wire slave 00.d300                                                                                       00000000 crc b5
[10590.947809] w1_master_driver w1_bus_master1: Family 0 for 00.d30000000000.b5                                                                                        is not registered.
[... cut ...]
also I have had this error when I started a reinstall. is this anything to do with the problem:

```
pi@raspberrypi:~/TerrariumPI $ sudo ./start.sh
>>> /etc/sudoers.d/terrariumpi: syntax error near line 1 <<<
sudo: parse error in /etc/sudoers.d/terrariumpi near line 1
sudo: no valid sudoers sources found, quitting
sudo: unable to initialize policy plugin

```
In the passed I deleted this file for the software to work and had no problems till now.
Thanks for the reply. I'm in the dark literally lol.

theyosh, 2018-05-16:
Next time, pasting 3 or 4 lines with the same error is enough ;)

But the w1_master errors are also outside my software scope. And I can't tell what is happening, but my solution for this is to disconnect all 1wire sensors and add one by one back.

The error: >>> /etc/sudoers.d/terrariumpi: syntax error near line 1 <<< sounds to me a broken SD card. I used this file for 6 months, and never a problem.

I suggest to by a new SD card and reinstall everything.... I think that this card is end of live :(

rrgrow, 2018-05-18:
Agreed to much info I've deleted the post(s) above. Also my appolages if this is long winded too.

# Feed Back:
I have installed  `2018-04-18-raspbian-stretch.img` (I like to have GUI) to another SD card and currently don't get `error: >>> /etc/sudoers.d/terrariumpi: syntax error near line 1 <<<` like I did from the start on the other SD card using your software. Now in relation to the install of `TerrariumPI` I anwser yes to user settings on first page and yes on second page. The errors start on last page of installing python modules after Gevent..

### This Type of Error:
`Co9mand "/usr/bi /python -u -c "import setuptools, tokenize;file='/tmp/pip-build-Uvt81E/cffi/setup.py';f=getattr(tokenize, 'open', open)(file);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, file, 'exec'))" install --record /tmp/pip-X4YltI-record/install-record.txt --single-version-externally-managed --compile" failed with error code 1 in /tmp/pip-build-Uvt81E/cffi/`
 (I would have put up a screen shot but I have it up and running and don't want to go down just yet)

`Red blocked` erorr text displays over the install status but still continues, completes install and I reboot as standard. 
###  On boot I get this Type of Error
pi@raspberrypi:~/TerrariumPI $ screen -r
atch_all
_notify_patch(events.GeventWillPatchAllEvent(modules_to_patch, kwargs), _warnings)
File "/usr/local/lib/python2.7/dist-packages/gevent/monkey.py", line 167, in _notify_patch
notify_and_call_entry_points(event)
File "/usr/local/lib/python2.7/dist-packages/gevent/events.py", line 112, in notify_and_call_entry_points
subscriber = plugin.load()
File "/usr/lib/python2.7/dist-packages/pkg_resources/init.py", line 2290, in load
self.require(*args, **kwargs)
File "/usr/lib/python2.7/dist-packages/pkg_resources/init.py", line 2307, in require
items = working_set.resolve(reqs, env, installer)
File "/usr/lib/python2.7/dist-packages/pkg_resources/init.py", line 849, in resolve
raise DistributionNotFound(req, requirers)
pkg_resources.DistributionNotFound: The 'cffi>=1.11.5' distribution was not found and is required by the application....

### What I did:
Running the install and answering no on second page completes the installs without errors and works but limited (I'll get to that next). This work round worked for me on the other SD card up to 6 days ago (when I rised this issue) then it stopped working completely.  And was 4-6 updates ago it started doing this.

### Working but limited
When using TerrariumPI after setting sensors and switches to pin outs (Static ribbon cable so wrong pin outs all static build) and testing. I would then setup System environment. At this point I would notice ie.. Light would turn on off on off..... in and around 30 sec like I had before. 

**Water tank** will not let me submit settings but all others will. 
  ie.. quick green line across the top and no 'data saved' box.  All setting are the same as before, I'm using the same browser and it auto fills from last visit(s).

So i am not using System environment at the moment and just using timers in sensor setting. 
There is also a problem there too again with the lights on timer ie.. Light would turn on off on off....

This didn't happen before with the lights (Now System environment & Sensor setting) only with the Humidity Extract Fan (Only System environment). All other switches are fine.

### To sum things up 
Maybe my PI is dying or I have another bad SD card. I will buy the new PI b and try it out.
I just wanted to keep you in the loop and updated. 

### Sill love this software!
Cant wait for the new System environment this will open the software's usage to the max.
Great Work as always and Many Thanks.

theyosh, 2018-05-18:
Image 2018-04-18-raspbian-stretch.img is NOT working. Use the lite image. https://github.com/theyosh/TerrariumPI#installation

theyosh, 2018-05-18:
Also, do you use the zip or git clone? Because the zip is probably not having the right installer now. So use git clone to get the latest version. Could help with the installation.

rrgrow, 2018-05-18:
git clone every time. Will try lite . Thanks for the help 👍

rrgrow, 2018-05-19:
Error on System environment save:

2018-05-19 20:38:38,000 - INFO - terrariumWrapper - Starting TerrariumPI server at location: http://192.168.1.21:8090 ...
2018-05-19 20:38:49,000 - INFO - terrariumWebserver - Running webserver at :::8090
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/bottle.py", line 862, in _handle
    return route.call(**args)
  File "/usr/local/lib/python2.7/dist-packages/bottle.py", line 1740, in wrapper
    rv = callback(*a, **ka)
  File "/home/pi/terrariumPI/terrariumWebserver.py", line 43, in webserver_headers
    return fn(*args, **kwargs)
  File "/home/pi/terrariumPI/terrariumWebserver.py", line 81, in wrapper
    return func(*a, **ka)
  File "/home/pi/terrariumPI/terrariumWebserver.py", line 257, in __update_api_call
    result['ok'] = self.__terrariumEngine.set_config(path,postdata,request.files)
  File "/home/pi/terrariumPI/terrariumEngine.py", line 987, in set_config
    update_ok = self.set_environment_config(data)
  File "/home/pi/terrariumPI/terrariumEngine.py", line 787, in set_environment_config
    self.environment.load_environment(data)
  File "/home/pi/terrariumPI/terrariumEnvironment.py", line 69, in load_environment
    if len(config_data['sprayer'].keys()) == 0 or not reloading:
KeyError: 'sprayer'

This is with a new install of LITE and git clone. but only on save and not on start up.

theyosh, 2018-05-20:
This is my fault. Sorry. I commited some non working code...

If you go do: `git checkout 3.5.0` you should go back in code and should work again. I am working hard to change the environment code and a few lines slipped through...

theyosh, 2018-05-21:
I did change the master branch to fix this is guess.. Do you mean with not saving that the form is not processed, or that you will get an error when saving in the console?

The first problem of not processing the form should be fixed in the master

theyosh, 2018-05-31:
Ok, new release is out. So I think this issue should be fixed now. Could you try to update and see if it is all working?

theyosh, 2018-06-10:
Closing this as this is hardware related...

closed 2018-06-10 by theyosh