theyosh/TerrariumPI #150: Higher or Lower Humidity Control

rrgrow, 2018-05-14:
Humidity control (Sprayer) is only one way at the moment ie when humidity gets to low the sprayer will spray based on duration etc.. 
I would like to have a fan come on when the humidity get to high and stop when humidity gets to desired %. I did in the passed change the Sprayer settings in terrariumEnvironment.py:

```
 # Hum checks and actions
      if self.sprayer['enabled']:
        toggle_on = False
        extra_logging_message = ''
        logger.debug('Environment spraying is enabled.')
        logger.debug('Environment spraying is based on: %s' % self.sprayer['mode'])
        if 'sensor' == self.sprayer['mode']:
          # Only spray when the lights are on. Or when explicit enabled during the nights.
          if self.sprayer['night_enabled'] or self.is_light_on():
            # Spray based on the average humidity values of the used sensors
            toggle_on = self.sprayer['humidity']['current'] > self.sprayer['humidity']['alarm_max']
            if toggle_on:
              extra_logging_message = 'Sprayer humdity value %f%% is higher then alarm %f%%.' % (self.sprayer['humidity']['current'],
                                                                                        self.sprayer['humidity']['alarm_max'])
        else:
          # Spray based on time table
          toggle_on = terrariumUtils.is_time(self.sprayer['time_table'])
          if toggle_on is None:
            self.__update_timing('sprayer')

          if toggle_on and len(self.sprayer['sensors']) > 0:
            # Use the extra added sensors for finetuning the trigger action
            toggle_on = self.sprayer['humidity']['current'] > self.sprayer['humidity']['alarm_max']
            if toggle_on:
              extra_logging_message = 'Sprayer humdity value %f%% is lower then alarm %f%%.' % (self.sprayer['humidity']['current'],
                                                                                        self.sprayer['humidity']['alarm_max'])

        if toggle_on:
          if self.is_door_open():
            logger.warning('Environment could not spray for %f seconds based on %s mode because of an open door.%s' % (self.sprayer['spray_duration'],
                                                                                                                       self.sprayer['mode'],
                                                                                                                       extra_logging_message))
          else:
            if not self.is_sprayer_on():
              logger.info('Environment is turning on the sprayer for %f seconds based on %s mode.%s' % (self.sprayer['spray_duration'],
                                                                                                      self.sprayer['mode'],
                                                                                                      extra_logging_message))
            self.sprayer_on()
        else:
          if self.is_sprayer_on():
            logger.info('Environment is turning off the sprayer based on %s mode.' % (self.sprayer['mode'],))

          self.sprayer_off()

      #else:
      #  logger.debug('Make sure that the spayer is off when not enabled at all.')
      #  if self.is_sprayer_on():
      #    logger.info('Environment is turning off the sprayer due to disabling it')
      #  self.sprayer_off()
```

this worked for me up to a point ie when spray duration (fan on) was reached and 0 wait time you would of thought it would if humidity to high switch back on for 60 but this isn't the case what would happen is the fan comes on for a split second then off for a split second for ever till a restart or reboot.

I don't really need sprayer but didn't want to lose it either as I might need it in the future. Just wondering for us growers that need to control humidity downwards if there is a quick Environment fix for this maybe adding a simple Humidity control based on the 'Cooling settings' or a quick edit to the above code (Sprayer) as I'm not every sure if I've edited right. 
I know your working on a new "Add Environment" type setting page but if there is a quick fix for now i'll most appreciate it. 
Many Thanks in advance

theyosh, 2018-05-14:
New setup: You will have a alarm low and alarm high setting with own power switches. So in that way you can have a sprayer and blower controlling together the humidity. Or you use only the sprayer part or the blower part.

[image]

rrgrow, 2018-05-14:
2018-05-14 22:29:50,000 - INFO - terrariumWrapper - Starting TerrariumPI server at location: http://:8090 ...
Traceback (most recent call last):
  File "/home/pi/TerrariumPI/terrariumPI.py", line 12, in <module>
    from terrariumWebserver import terrariumWebserver
  File "/usr/local/lib/python2.7/dist-packages/gevent/builtins.py", line 96, in __import__
    result = _import(*args, **kwargs)
  File "/home/pi/TerrariumPI/terrariumWebserver.py", line 12, in <module>
    from bottle.ext.websocket import GeventWebSocketServer
  File "/usr/local/lib/python2.7/dist-packages/gevent/builtins.py", line 96, in __import__
    result = _import(*args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/bottle.py", line 1797, in load_module
    __import__(realname)
  File "/usr/local/lib/python2.7/dist-packages/gevent/builtins.py", line 96, in __import__
    result = _import(*args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/bottle_websocket/__init__.py", line 2, in <module>
    from .server import GeventWebSocketServer
  File "/usr/local/lib/python2.7/dist-packages/gevent/builtins.py", line 96, in __import__
    result = _import(*args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/bottle_websocket/server.py", line 4, in <module>
    from geventwebsocket.handler import WebSocketHandler
  File "/usr/local/lib/python2.7/dist-packages/gevent/builtins.py", line 96, in __import__
    result = _import(*args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/geventwebsocket/__init__.py", line 17, in <module>
    from .resource import WebSocketApplication, Resource
  File "/usr/local/lib/python2.7/dist-packages/gevent/builtins.py", line 96, in __import__
    result = _import(*args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/geventwebsocket/resource.py", line 5, in <module>
    from .exceptions import WebSocketError
  File "/usr/local/lib/python2.7/dist-packages/gevent/builtins.py", line 96, in __import__
    result = _import(*args, **kwargs)
ValueError: bad marshal data (unknown type code)
2018-05-14 22:30:33,000 - INFO - terrariumWrapper - Restart counter: 1/5
2018-05-14 22:30:33,000 - INFO - terrariumWrapper - Restarting in 10 seconds after running for 43 seconds. Press Ctrl+C now to terminate TerrariumPI.

theyosh, 2018-05-23:
Currently I am finalizing the code and do some testing. Here are some new pictures. There is now also an option to select day lights and night lights.... 
[image]
[image]
[image]

theyosh, 2018-05-27:
Ok, we are almost there. I got a new version running on my live terrarium. You can see it here: https://terrarium.theyosh.nl/

There you can see the new Environment system running and what all the options are. Currently I would like to test it a couple of days more. But I think it is pretty stable at the moment. If you are brave enough, you could switch to the development branch and test it your self. :)
Make a backup of the 'settings.cfg' file if you want to go back to the master branch BEFORE testing development.

theyosh, 2018-05-29:
Cool.... browser cache is a bitch :P

theyosh, 2018-05-31:
Ok, new release is out. So I think this issue should be fixed now. Could you try to update and see if it is all working?

theyosh, 2018-06-10:
Any updates please? As far I can judge, it is working like requested

theyosh, 2018-06-13:
Closing after 15 days of silence... :P

closed 2018-06-13 by theyosh

rrgrow, 2018-06-21:
been working fine for me all this time many thanks 🥇