theyosh/TerrariumPI #151: German update

Barbara1984, 2018-05-19:
Hello, I have revised the German file.

[terrariumpi.zip](https://github.com/theyosh/TerrariumPI/files/2019782/terrariumpi.zip)

theyosh, 2018-05-23:
Cool thanks! I will add in the development version which I hope to release end of this week.

And may I ask you to subscribe at the issue #115 ? And post your new future translation in that issue? So I have all the translation messages in one issue.

theyosh, 2018-05-23:
But, I have already to admit that the code is currently changing a lot, so a part of the translation will be outdated, and there will be new translations missing.

Sorry for changing that a lot and that translations have to be done again.... But I have to do it also for Dutch... :(

Barbara1984, 2018-05-24:
Ok, is not a problem as soon as a new version is out I'm doing a new translation.

closed 2018-05-24 by Barbara1984