theyosh/TerrariumPI #152: Automatic watering

Barbara1984, 2018-05-19:
A question, it is possible with my automatic irrigation, for example, adjust the automatically every 2 days is poured

theyosh, 2018-05-20:
Hmmm, I think not yet, because I use 24 hours as max interval. So I have to look into this...

Barbara1984, 2018-05-20:
Ok, thanks for your effort

theyosh, 2018-05-23:
Does it really have to be at 2 days? What happen if you do it daily, with the half amount of once every 2 days. The amount of water is the same. Not sure if your environment is working with this.

Because when it can be done once every 24 hours, the next version can handle that.

Barbara1984, 2018-05-24:
Hi, yes for me it would be great if I could go to 48 hours, if that does not work then it is not so bad then I really have to use it with 24 hours. But please let me know, thank you.

theyosh, 2018-05-27:
Sneak preview: https://terrarium.theyosh.nl/

Go to the Environment part and look at the moisture part. There you could then select an alarm min option based on timer mode. Select the period in which the pump could run. And below that, the amount of minutes that the pump is actual powered on. So this way, the smallest period is 1 minute per 1 day.

Make sure the period that the pump can run, is only one or two minutes bigger then the power on time.

I hope to release this version soon.

Barbara1984, 2018-05-27:
Hi, I think that's just right. :-)

theyosh, 2018-05-31:
Ok, new release is out. So I think this issue should be fixed now. Could you try to update and see if it is all working?

theyosh, 2018-06-20:
Closing due to complete and tested.

closed 2018-06-20 by theyosh