theyosh/TerrariumPI #153: Cooler mode:Sensor

steveava, 2018-05-29:
This is probably bad setup on my part but I'm trying to use the Cooler function as my ambient temperature can occasionally be higher than the set/alarm temperature and need to cool things down a bit. My plan was to try and use a Peltier modual and just flip the polarity through a relay(switch) on the PI GPIO when required. The heater toggles on and off as you would expect and most of the time that's how it will work but the cooler(switch) never seems to operate when at a high or above temperature setting. Cooler alert is fine but I cant seem to control the GPIO relay(switch).

Manual operation of the relay works ok under Switches/Status but it will switch off after a while. Also just noticed from the log that lighting has an effect on the cooler. In my application the lighting may be on or off but the temperature could still be be too high.

Have I missed something in the setup?

Thanks for your work on TerrariumPI.
Steve.

theyosh, 2018-05-29:
Hi, thanks for the compliment. 

I think it is a code issue. Because I do not use it, and maybe not tested fully. But I am close at releasing a new version, which has just a temperature option, that holds alarms for low values and high values. So when it get to hot you can switch cooler devices.

Preview https://terrarium.theyosh.nl/ . This is currently the development version. If you go to System -> environment, you can see the new setup. And that will work better.

steveava, 2018-05-30:
The low temperature section in the development version looks great. Nice that temperature is all in one place.

Looking forward to 3.6.0 release.

Curious about "Power On". is that a power off delay once the the temperature threshold is met/exceeded.

theyosh, 2018-05-30:
The power on option is used for 'tigger' functionality. If this number is higher then 0, then the power will be on for that amount of minutes and then go off. This is for heating or cooling not needed, but for humidity, moisture, water tank it could be handy. After that, it will wait the settle time, and check again. When still in alarm, it will start over all again.

theyosh, 2018-05-31:
Ok, new release is out. So I think this issue should be fixed now. Could you try to update and see if it is all working?

steveava, 2018-06-01:
Ran through the update without incident. Went to temperature setting but couldn’t add/submit anything. Just keep getting the busy circle at top right of screen. The log is showing this message:

 

Environment temperature should turn on the alarm max but is blocked by: door False, light True

 

Don’t have a door switch and lighting isn’t that important for me but lighting is disabled in Environment and there are no switches in “Doors” so not sure where the error is coming from.

theyosh, 2018-06-01:
Clear browser cache :) Will fix it

steveava, 2018-06-05:
Great, clearing cache seems to solve the problem.

Been heating/cooling and seems to be working ok. Did notice on a couple of occasions that the status indicator in dashboard seem to fluctuate but the hardware(switches) were stable.

Whilst the little flame icon in dashboard stayed off while in range, it came on for heating or cooling. I seem to remember a different symbol on the previous version for cooling but cant remember if it was displayed the same way in dashboard. Hard to know which direction the temperature is moving from a quick dashboard glance.

I was playing around with the idea that I could get an alarm(switch) to sound out if things got to hot or cold(busted heater). The warning(alarm) seems to be the High/Low threshold. I tried to use another sensor with different values but couldn't find a way to operate another switch. I'm probably going about this the wrong way or not looking in the right place but it seems that there could be more values to sensors, alarm/range/display limits. I suppose the other argument is we shouldn't rely on one system to monitor temperature.

theyosh, 2018-06-05:
Not sure if I understand you correctly, but this is how it should work.

- At the temperature part, select all sensors you need to determine the right temperature.
- Then select the power switches at alarm Low for heating equipment
- And select the power switches for the cooling equipment at the high alarm settings.
- Set both light state and door state on ignore value
- Set power on value to zero

The idea is that when it get's to cold, based on the average of the sensors, the low alarm power switches will be put on. When the hot the powerswitches at the high alarm settings would trigger to cool things down.

The alarm low value is based on the alarm min values of the sensors. And the high alarm on the alarm max values of the sensors.

steveava, 2018-06-05:
Yes, that's exactly how I have it set currently and works great. I suppose this was more to do with what I understood "alarm" to mean in the software and initially thought it was an extreme condition rather than the desired temperature range or temperature threshold.  Anyway thinking about an extreme condition like a heater failure, I tried to set a switch to operate when the temperature or humidity was way out. Didn't know if there was a way to do that or if there could be another two values(extreme condition) in the sensors config. I know it's already been mentioned recently about Email notification and that may be a while off but we could do the same with GPIO to trigger another application that can handle Email or Email to SMS or make a sound(alarm). I know that's not very efficient but ok in the short term, that was my thinking.

theyosh, 2018-06-05:
Ah ok that is cool.

Yesh, the alarm icon is indicating that action is needed to keep the values between the min and max. And that will trigger powerswitches. Could be a bit false warning.... but then again, I am a developer, not de UX designer :p

Your SMS or Email alarm is just around the corner ;) At the moment I am developing a general notification system. Supporting email, twitter, pushover, telegram bot....

Also, another security measurement is that when a sensor is not working correctly or giving values above or below the min and max values for 20 minutes, it will be marked as faulty, and when the environment part is depending on that faulty sensor, it will shutdown all power switches. This way overheating or cooling should be prevented.

steveava, 2018-06-05:
The notification stuff sounds great.

The faulty sensor sounds great as well. Already had a situation where I was running an old DHT22 and BME280 and the 22 started playing up, messing up the average temperature.

Not sure what the best sensors are reliability wise but the BME280 seems consistent between units so far.

theyosh, 2018-06-10:
Can this issue being closed? I think it is working now

For testing with email notifications look at #101

steveava, 2018-06-11:
I have it running in a test environment at the moment and temperature control seems to be working ok between heating/cooling.

Haven't had a chance to look at Email notification but will do soon.

theyosh, 2018-06-20:
Closing this due to implemented and working.

closed 2018-06-20 by theyosh