theyosh/TerrariumPI #154: Wrong weather url (http vs https) does not allow TerrariumPi to start

penzoiders, 2018-05-29:
If you had (as I had) weather forecasts based on a plain http://www.yr.no/whatever the TerrariumPi won't restart the webserver upon reboot or CTRL-C in the screen.

basically never gets after
```
2018-05-29 21:41:42,000 - INFO - terrariumWrapper - Starting TerrariumPI server at location: http://xx.xx.xx.xx:8090 ...
```
if you fix in config the `http` with an `https` it works and you then get the TerrariumPi to start 
```
2018-05-29 21:42:03,000 - INFO - terrariumWebserver - Running webserver at :::8090
```

a warning in the screen it would be cool, even better ignore settings if they're bad  and clear settings to defaults.

I've noticed this upgrading from 3.4.1 to 3.5.0

Kargha, 2018-05-29:
I actually noticed something similar with this, where if I'd input specific yr.no URLs, the settings wouldn't save and you'd get no error message (and in the screen on the Pi it would either give a complete error stacktrace, or just a common error). This would happen anytime I'd use the Norwegian type of URL (which looks like this: https://www.yr.no/sted/Sverige/Stockholm/Stockholm/ instead of a the English one, like this https://www.yr.no/place/Sweden/Stockholm/Stockholm/). When using the Norwegian URL, it would just give an error in the console, but not the front end, and wouldn't save the settings at all. When using a URL with special characters, such as åäö, it would give a stacktrace.

So it's rather strict on what it permits, and could use some more front-end errors if it fails to save.

theyosh, 2018-05-29:
@penzoiders  Strange, I run with: http://www.yr.no/place/Netherlands/Groningen/Groningen no https there. So somehow the http -> https redirect is not working for you. Did you do an (re)install by running the installer script? Could be that new packages will fix this.

@Kargha Also strange. I have to look into it. I think I do expect English weather terms. Not Norwegian. So that is the reason. I will see If I can add that as a feedback in the webinterface.

About 'åäö' characters. They should work. I thought I tested it with and changed everything to fully UTF-8 and should work... If you have an not working example I can use that for testing. Saves me a bit time finding such url. We do not have those chars in place names.

Kargha, 2018-05-30:
@theyosh Here are the links I attempted with "å" in the URL:

https://www.yr.no/sted/Sverige/Skåne/Lund/

https://www.yr.no/sted/Sverige/Sk%C3%A5ne/Lund/

penzoiders, 2018-05-30:
@theyosh I did stop the screen session, git pull, sudo ./install.sh (took forever to compile getenv), reboot.

upon reboot it was stuck, commented wheather info on .cfg file, CTRL-C to restart in screen session, all ok with new url... I had no special chars in URL.

theyosh, 2018-05-30:
Ok, changed the code.

1. If an url starts with http:// I will change it to https:// as every source support SSL. So that should not give any problems anymore.

2. Wrong urls are now blocked. I do check if the part /place/ is available, and if not, an error will be given in the web interface. The only issue right now, is that it is a general error. So a look at the terminal is still needed :(. But the system will continue to run. I tried to see if I can translate the non English urls to English urls, but that involves parsing HTML. Just to much work.....

3. the strange characters should not be a problem anymore, because I will now only accept English urls, so those chars are then not used. :)

This will be added in the new 3.6.0 version. @penzoiders Sorry, you will have to do the installation run soon again ;)

Kargha, 2018-05-30:
@theyosh Yea, the thing about translating the non-English URLs wouldn't work, since places will have different names in different languages sometimes. For example: Gothenburg in Sweden is actually Göteborg in Swedish. And the example in the URL above is Skåne in Swedish, but Scania in English. So should just make a notice or something that the English URL has to be used, less painful.

theyosh, 2018-05-31:
Ok, new release is out. So I think this issue should be fixed now. Could you try to update and see if it is all working?

dasolan, 2018-06-03:
Accidently I missed the region in the yr.no URL and saved through nano. 
TerraPi starts but at the settings panel, everything was blank. 
This was before I found the issue, otherwise I had post you the screen msg.

theyosh, 2018-06-04:
@dasolan it will shown in the log as an error: Setting weather source failed! The url '%s' is invalid

theyosh, 2018-06-10:
Closing as this is resolved for me with error notification in browser and an error log message in the logfile when not correctly entered address.

closed 2018-06-10 by theyosh