theyosh/TerrariumPI #155: Error after update

Kollimat, 2018-06-01:
Hello i updated and im getting this error after I try start the server. 

pi@raspberrypi:~/TerrariumPI $ ./start.sh run
2018-06-01 13:38:39,000 - INFO - terrariumWrapper - Starting TerrariumPI server at location: http://192.168.1.150:8090 ...
Traceback (most recent call last):
  File "/home/pi/TerrariumPI/terrariumPI.py", line 17, in <module>
    terrariumEngine = terrariumEngine()
  File "/home/pi/TerrariumPI/terrariumEngine.py", line 70, in __init__
    self.config = terrariumConfig()
  File "/home/pi/TerrariumPI/terrariumConfig.py", line 41, in __init__
    self.__upgrade_config(version)
  File "/home/pi/TerrariumPI/terrariumConfig.py", line 181, in __upgrade_config
    if 'heater_night_source' in data and 'temperature_day_night_source' not in newdata or newdata['temperature_day_night_source'] == '':
KeyError: 'temperature_day_night_source'
2018-06-01 13:38:44,000 - INFO - terrariumWrapper - A new version of TerrariumPI is detected. Please restart.

theyosh, 2018-06-01:
Hmmm, how good are you with python code :)

Try to edit the file /home/pi/TerrariumPI/terrariumConfig.py with vi or nano. Go to line 181 and change the following:

`if 'heater_night_source' in data and 'temperature_day_night_source' not in newdata or newdata['temperature_day_night_source'] == '':`

to 

`if 'heater_night_source' in data and ('temperature_day_night_source' not in newdata or newdata['temperature_day_night_source'] == ''):`

and restart again?
The difference is a ( and )

Kollimat, 2018-06-02:
Thanks, now i have another problem, web server is running on RP but i cannot connect to it and i dont think is runnig corectly because nothing is happening in my terarium.

theyosh, 2018-06-02:
Without error messages I cannot say what is wrong. What is the logfile saying? And is there an error in the screen session?

Could you else upload your settings.cfg file here. Maybe that will give a clue

Kollimat, 2018-06-03:
[terrariumpi.log.2018-06-02.zip](https://github.com/theyosh/TerrariumPI/files/2065793/terrariumpi.log.2018-06-02.zip)
This is a log file from yesterday, still cant load website.

Kollimat, 2018-06-03:
This I found in morning.
2018-06-03 08:21:54,000 - INFO - terrariumWebserver - Running webserver at :::8090
./start.sh: line 115:  5815 Killed                  python ${BASEDIR}/terrariumPI.py
2018-06-03 08:45:00,000 - INFO - terrariumWrapper - Restarting in 10 seconds after running for 1417 seconds. Press Ctrl+C now to terminate TerrariumPI.
10 9 8 7 6 5 4 3 2 1 restart!
2018-06-03 08:45:11,000 - INFO - terrariumWrapper - Starting TerrariumPI server at location: http://192.168.1.150:8090 ...
2018-06-03 08:45:30,000 - INFO - terrariumWebserver - Running webserver at :::8090
./start.sh: line 115:  5897 Killed                  python ${BASEDIR}/terrariumPI.py
2018-06-03 09:08:46,000 - INFO - terrariumWrapper - Restarting in 10 seconds after running for 1415 seconds. Press Ctrl+C now to terminate TerrariumPI.
10 9 8 7 6 5 4 3 2 1 restart!
2018-06-03 09:08:58,000 - INFO - terrariumWrapper - Starting TerrariumPI server at location: http://192.168.1.150:8090 ...
2018-06-03 09:09:19,000 - INFO - terrariumWebserver - Running webserver at :::8090
./start.sh: line 115:  5973 Killed                  python ${BASEDIR}/terrariumPI.py
2018-06-03 09:32:06,000 - INFO - terrariumWrapper - Restarting in 10 seconds after running for 1388 seconds. Press Ctrl+C now to terminate TerrariumPI.
10 9 8 7 6 5 4 3 2 1 restart!
2018-06-03 09:32:20,000 - INFO - terrariumWrapper - Starting TerrariumPI server at location: http://192.168.1.150:8090 ...
2018-06-03 09:32:44,000 - INFO - terrariumWebserver - Running webserver at :::8090
./start.sh: line 115:  6076 Killed                  python ${BASEDIR}/terrariumPI.py
2018-06-03 09:55:49,000 - INFO - terrariumWrapper - Restarting in 10 seconds after running for 1409 seconds. Press Ctrl+C now to terminate TerrariumPI.
10 9 8 7 6 5 4 3 2 1 restart!
2018-06-03 09:56:03,000 - INFO - terrariumWrapper - Starting TerrariumPI server at location: http://192.168.1.150:8090 ...
2018-06-03 09:56:28,000 - INFO - terrariumWebserver - Running webserver at :::8090
./start.sh: line 115:  6155 Killed                  python ${BASEDIR}/terrariumPI.py
2018-06-03 10:19:31,000 - INFO - terrariumWrapper - Restarting in 10 seconds after running for 1408 seconds. Press Ctrl+C now to terminate TerrariumPI.
10 9 8 7 6 5 4 3 2 1 restart!
2018-06-03 10:19:42,000 - INFO - terrariumWrapper - Starting TerrariumPI server at location: http://192.168.1.150:8090 ...
2018-06-03 10:20:09,000 - INFO - terrariumWebserver - Running webserver at :::8090
./start.sh: line 115:  6248 Killed                  python ${BASEDIR}/terrariumPI.py
2018-06-03 10:43:03,000 - INFO - terrariumWrapper - Restarting in 10 seconds after running for 1401 seconds. Press Ctrl+C now to terminate TerrariumPI.
10 9 8 7 6 5 4 3 2 1 restart!
2018-06-03 10:43:14,000 - INFO - terrariumWrapper - Starting TerrariumPI server at location: http://192.168.1.150:8090 ...
2018-06-03 10:43:41,000 - INFO - terrariumWebserver - Running webserver at :::8090

theyosh, 2018-06-03:
Ok, but can you also send me the settings.cfg contents? Of the file? I need that one also.

The log does not show any error, and does indicate that everything is starting up correctly. But the script is killed, which is not good.

Could you also try the following:

1. Login with ssh into your raspberry pi
2. enter the command `dmesg`
3. Copy the output here.

There is a problem which is only happening at your setup, so I have no idea how to replicate this issue. So therefore the settings.cfg file is important here.

Kollimat, 2018-06-03:
[67690.815766] Total swap = 102396kB
[67690.815769] 229376 pages RAM
[67690.815771] 0 pages HighMem/MovableOnly
[67690.815774] 5208 pages reserved
[67690.815776] 2048 pages cma reserved
[67690.815779] [ pid ]   uid  tgid total_vm      rss nr_ptes nr_pmds swapents oo                                                                                                                     m_score_adj name
[67690.815797] [   84]     0    84     6566      516      11       0       65                                                                                                                                  0 systemd-journal
[67690.815803] [  126]     0   126     3592      300       8       0      156                                                                                                                              -1000 systemd-udevd
[67690.815813] [  278]   100   278     4319      234      10       0       98                                                                                                                                  0 systemd-timesyn
[67690.815820] [  320]   108   320     1600      267       7       0      103                                                                                                                                  0 avahi-daemon
[67690.815826] [  321]     0   321     1324      327       6       0       39                                                                                                                                  0 cron
[67690.815832] [  322]     0   322     1844      232       7       0       80                                                                                                                                  0 systemd-logind
[67690.815838] [  325]   105   325     1624      310       9       0      102                                                                                                                               -900 dbus-daemon
[67690.815844] [  348]   108   348     1600        0       7       0       77                                                                                                                                  0 avahi-daemon
[67690.815853] [  353] 65534   353     1323      268       6       0       55                                                                                                                                  0 thd
[67690.815860] [  356]     0   356     5937       94      10       0      184                                                                                                                                  0 rsyslogd
[67690.815866] [  365]     0   365    15357      230      15       0      285                                                                                                                                  0 uv4l
[67690.815872] [  435]     0   435     2540      313       8       0      115                                                                                                                                  0 wpa_supplicant
[67690.815878] [  450]   109   450    28196      573      41       0      875                                                                                                                                  0 motion
[67690.815885] [  462]     0   462      524        0       5       0       34                                                                                                                                  0 hciattach
[67690.815890] [  467]     0   467     1768      154       7       0       76                                                                                                                                  0 bluetoothd
[67690.815896] [  522]     0   522      737      242       5       0       94                                                                                                                                  0 dhcpcd
[67690.815902] [  525]     0   525     4738      814      14       0     1835                                                                                                                                  0 supervisord
[67690.815909] [  557]     0   557     2551      118       8       0      150                                                                                                                              -1000 sshd
[67690.815915] [  565]  1000   565     2415      296       9       0      180                                                                                                                                  0 systemd
[67690.815921] [  568]  1000   568     2859        0       9       0      322                                                                                                                                  0 (sd-pam)
[67690.815927] [  574]  1000   574     1347      320       6       0      118                                                                                                                                  0 screen
[67690.815935] [  578]  1000   578     1161      271       7       0       52                                                                                                                                  0 start.sh
[67690.815940] [  611]     0   611     1049      239       7       0       35                                                                                                                                  0 agetty
[... cut ...]                                                                          B mapped:5872kB dirty:0kB writeback:0kB shmem:580kB writeback_tmp:0kB unstable:0                                                                                                                     kB all_unreclaimable? no
[89263.033382] Normal free:20356kB min:16384kB low:20480kB high:24576kB active_a                                                                                                                     non:4193

Kollimat, 2018-06-03:
Still no change i cant connect to website, maybe reinstall everything be better.

theyosh, 2018-06-03:
Hmm, that is the easiest right now. Will take some time for you, but at the moment I have no other clues...

Kollimat, 2018-06-03:
ok thanks for help, its there a way how to reinstal TerrariumPI without deleting it and installing again ?? i hope i dont need to reinstall whole Raspberry :D

Kollimat, 2018-06-03:
WTF it just start working :D :D i did nothing and im connected to website, just everything is disabled in environment and sometimes didnt show the temperature and humidity, strange

Kollimat, 2018-06-03:
OK false alarm , i dont understand need to reinstall

theyosh, 2018-06-03:
Ok, the easiest is to rename the existing TerrariumPI folder to TerrariumPI.old and restart the installation of my software. This way you do not have to reinstall the Pi.

When the new installation is done, you can copy the history.db and settings.cfg file from the old version to the new version. This will restore your sensor data and settings

Kollimat, 2018-06-04:
Thank for help, I reinstalled everything, copied files, it seems that everything is working but everything is disabled in enviroment and if i change anything nothing happend

theyosh, 2018-06-04:
Clear browser cache should fix it

Kollimat, 2018-06-05:
Thank you for everyhing, it is working again. :)

closed 2018-06-05 by Kollimat