theyosh/TerrariumPI #156: Support for MiFlora

theyosh, 2018-06-04:
Maybe [MiFlora](https://xiaomi-mi.com/sockets-and-sensors/xiaomi-huahuacaocao-flower-care-smart-monitor/) is an alternative?
Its a Bluetooth sensor for 11 Euro at Ali, it detects moisture, lux, temperature and ec (conductivity).
Librarys still existing on git.

jneumaier, 2018-06-04:
Hey! Thanks for your great project, just playing around with it and familiarizing myself with the code. I just ordered two of those to play around with it. Do you wanna have one or did you already order yourself? I would be happy to provide one. I am a developer as well so I would also like to try to help with the implementation. Have not a lot of python experience though.

dasolan, 2018-06-05:
My order is also on the way, I really appreactiate all your efforts!

jneumaier, 2018-06-05:
> Maybe when the code is finished I can use someone to test it

Sure.

theyosh, 2018-06-10:
Delay delivering due out of stock :(

theyosh, 2018-06-15:
Devices on its way. ETA 10 days :(

theyosh, 2018-07-07:
Apparently they do not understand English and have canceled the order.....

theyosh, 2018-07-10:
Re-ordered new ones from difference supplier....

theyosh, 2018-07-25:
Added first support for MiFlora devices. They will be automatically scanned and added to the system. Only missing now is Light due to different measurement then Chirp light devices.

https://github.com/theyosh/TerrariumPI/commit/61f9178f46ffcd12f1d0a244b99bd05baf029640

theyosh, 2018-07-27:
I will close this issue, due to the fact the code is released and this issue is solved.

closed 2018-07-27 by theyosh