theyosh/TerrariumPI #157: Update problem

Barbara1984, 2018-06-06:
Hi Theyosh,
unfortunately works for me the update not see photos, it gets stuck and then
comes the red screen.

[image]
[image]

Barbara1984, 2018-06-06:
[image]

theyosh, 2018-06-06:
Installing gevent module will take 5 minutes.... :( So that needs some patients...

The error about the chirp-rpi I have to look into it... Did you update by using `git pull`?

theyosh, 2018-06-06:
I just run the installer and tested extra about the chirp-pi. But it all works here. Just run the installer.sh again and see what happens

Barbara1984, 2018-06-06:
Yes, it worked great :-)

closed 2018-06-06 by Barbara1984