theyosh/TerrariumPI #158: errors after update to 3.6.0

bartch, 2018-06-09:
I don't know if this error is related to issue #155...
I updated about an hour ago and if I go to system environment page, all switches and sensor settings are gone. So e.g. no light in the Terrarium at the moment. If I try to add them again, the form forces me to add also values for e.g. heating while heating is turned off. If I add some dummy values and go to save. It says that my settings are saved but log and also reload of system environment page tells me something different.

One more question...maybe we are a little bit stupid today, but what's the difference between humidity and moisture sensors? We don't get it ;)

Thanks for your great work!

theyosh, 2018-06-10:
Without error messages I can't tell what is going on. And I have to make some assumptions now. So that is not working.

- You have run the installer script? If it did not finish with the reboot question, please run again
- Have you then rebooted the pi?
- Have you cleaned the browser cache?
- Are there messages in the screen session: https://github.com/theyosh/TerrariumPI/wiki/FAQ#how-to-debug-terrariumpi

At the moment, most fields are required, due to the fact that is easier to program, the depending on the mode.

Difference with humidity and moisture is not much. I think it is more that most people think with humidity of air humidity, and moisture is in the ground. And as far I know, there are calculated different by the sensors.

Also, there are persons running my software for gardening stuff.. And there they use moisture. So that is way I just added it.

dasolan, 2018-06-10:
I don't want you to flood with issues, but the "required" fields are sometimes disturbing. 
Soundcard is annoying, I cannot save any config without a card selected, but onboard is correctly recognized but cannot be chosen or left empty. 
Bypassed with manual config editation.
Only tested with FF:

Do you want to know such little things?

bartch, 2018-06-12:
Sorry for the late reply...
I was going through the environment settings again, today. On saving, the logs in the screen session don't change. In the logs inside the web page I can see the message 

```
2018-06-12 18:23:10,614 - INFO    - terrariumEnvironment - Done reloading terrariumPI Environment 0.002 seconds
2018-06-12 18:23:10,612 - INFO    - terrariumEnvironment - Reloading terrariumPI Environment system
```

After that again:

`2018-06-12 18:23:34,251 - INFO    - terrariumEnvironment - Environment light alarm max is triggered to state False, but has no powerswitches configured`

But, I found the problem...if I use Firefox as browser everything works fine. The problem arises only when using Safari.

theyosh, 2018-06-13:
Sounds again browser caching issue... Safari on Mac here does work like expected. Either clear browser cache or run browser in private/incognito mode

theyosh, 2018-06-13:
@dasolan Your issues are caused either by browsercache (so clear it) or that you are blocking javascript ajax calls. Because I am developing in Firefox and have tested it on Mac and Linux, and the settings page can be saved without selecting any option. In other words, all fields are filled with default values and can be saved right a way. That you are not seeing any soundcard devices has to do that your browser is blocking requests.

theyosh, 2018-06-20:
Closing due to no response, and problem is for 99% browser cache.

closed 2018-06-20 by theyosh

se7en9057, 2020-05-29:
I have cleared cache. still not getting soundcard option in settings. 
current setup. RPI3b+
install using python 3
restarted Pi
using firefox
cleared cache
restarted Pi
still no option for soundcard
(this is after fresh install.)