theyosh/TerrariumPI #159: Proportional temperature control

steveava, 2018-06-11:
There seem to be a number of switch devices in the list but are any capable of proportional control for heating/cooling. I guess that PWM would do that but it isn't recommended with Peltiers. I was looking at a 5 amp variable DC-DC regulator and replacing the trim pot with an I2C digital pot.

Has anything like this or alternative proportional control come up before?

I'm finding GPIO and relays/SSR to be too erratic for my application(4 degree swing) and need a little more stability although I'll have a play with the settle timeout to see how much difference that makes.

Steve.

theyosh, 2018-06-15:
Euh, this is new to me.. So I have to read a bit about it. According to http://www.takaitra.com/posts/503 I think I can support those kind of sensors.

So if you have already a sensor, please give the ID and I can see if I can fix some code :)

steveava, 2018-06-18:
Yea, similar idea but with I2C. I'm using MCP4561.

theyosh, 2018-07-02:
@steveava How good are you with python code? If you are able to test http://www.takaitra.com/posts/503 and it works, I can just add the code to my code. And then this kind of dimmers are then also supported.

For me personally, I do not have the right hardware and therefore not able to test. And adding code without testing is not an option.

steveava, 2018-07-02:
It's not something I do often but have been known to write a line or two. I'll put something together and send it over. Been looking at another I2C device so i'll do some testing with that as well.

theyosh, 2018-09-09:
Did you succeed to get a script running?

steveava, 2018-09-10:
Struggling to find some free time at the moment but still on the cards.

theyosh, 2020-01-28:
Are there any updates on this? As the forum posts are gone...

theyosh, 2020-04-05:
Closing due to no updates in more then 2 months

closed 2020-04-05 by theyosh