theyosh/TerrariumPI #166: Mirror Climate at Geographic Location

httpLov3craft, 2018-07-02:
I understand this may be a bit much, but I think it would be incredible to be able to provide a weather station near the traditional habitat of the flora and/or fauna being cared for, and have TerrariumPI mirror the key elements of that location's climate (temperature, sunrise/sunset, humidity, UV index, etc...) into the Terrarium as best as possible leveraging the available sensors.

Not sure if this would provide beneficial variation in the micro-climate to better support the biome, but it would be an interesting feature to talk to at the least, and could potentially provide easier initial setup / tuning.

theyosh, 2018-07-02:
Ha! This can be done already! It is only a bit hard to do. But here we go....

1. You can pick your own weather location in the system settings. This is used for calculating actions based on sunrise and sunset. So that is already done.

2. You can add external sensors to the system: https://github.com/theyosh/TerrariumPI/wiki/Remote-data . You can use https://openweathermap.org/api to get a free account and pick the right url for current temperature at the selected weather station. So this will then work as an extra sensor for temperature or humidity. And can be used in calculating averages and triggers.

But to get the external 'sensors' to work, you have to fiddle with the JSON urls from the source. But if you have your own weather station that can provide JSON output, you can even use that.

But for me, it does not have any extra value. Because I need to create an environment that is always hotter and more humid then my living room. So for me, if I add the outside values, there will be more spraying then needed.

httpLov3craft, 2018-07-02:
I think you may misunderstand my meaning, or I misunderstand your reply. By way of example, I would like to be able to select a weather station in central Australia in the outback, and have the software use those as reference values in dynamically determining the appropriate set-points for the atmospherics as opposed having to define the ranges manually. This way the micro-climate mirrors the climate of a selected weather station half-way across the world. (Ideally time-shifted by a configurable number of hours in offset) 

Though it is possible that I have not yet god far enough to discover the mechanisms by which to do this. I do not have all the necessary hardware to begin my build yet, so I have only played trivially in the interface and its configuration options.

theyosh, 2018-07-03:
Ah yes, now I got it. For the light part this is done. My lights will be on based on the sunrise and sunset at my location. But that could be changed to any location.

So for temperature you should have some logic that is able to get the average of all sensors, and when it is off by X temperature / percentage activate heater of cooler. Same goes for humidity.
Then you do not need to configure any sensors for alarms and min and max values.

This is a neat idea. And no, this is not possible yet... But maybe ... I can add it as a 'way of operating'...

ese23, 2019-01-26:
Hello, and what about time zones? If i want exact lenght of lightnig in Amazonia, its possible? But for me its unpractical because of "spacetimeshift". Wish to log weather (sunset, temp, hum) somewhere, and after few hours use it in TerrariumPI. Perfect natural mirror enviroment with seasons. But for me Amazonia is near aquator, soo days are near the same as nights all over the year. But temp drop by seasons.

PS what about forecast model thats drawings grafs in web interface of TerrariumPI. This is somekind of historical filtered data. No spikes in temp...

PSPS it will be funy, if its work... and for exmple, they will be huricane somewhere, and your hab will be mimicring heavy rain, temp drop, high wind... just kiding... dont wish this :-)

theyosh, 2021-12-10:
Closing this issue, as with the new [release 4.1.0](https://github.com/theyosh/TerrariumPI/releases/tag/4.1.0) this is now possible.

closed 2021-12-10 by theyosh