theyosh/TerrariumPI #167: Heater control based on temperature sensors or timers?

kaju666, 2018-07-04:
Hi

I can't find this feature.
I have one sensor DHT22 and relay board connected to heater.
I want to turn it on when temperature is lower than 28 degrees od turning off when it is 29.
When i tried to set this in envoirment in the same time it try to turn on and off the heater.
[image]
[image]
[image]

theyosh, 2018-07-04:
Hmm, as shown here it should work as you described. So that is all correct logic. 

But I get the feeling that the environment settings are not saved correctly. What happens when you try to save the environment page? The thing that worries me is the 'invalid date' message in the start and stop fields. They are NOT needed when used in sensor mode, but saving the form could fail and therefore the settings are not saved and activated. In order to 'fix' this, you could set it to timer mode. Then you can edit the start and stop field. Take some simple values like '12:00' and '13:00'.  Then put it back to sensor mode and save the form. If that does solve it, there is a bug that should be fixed.

I hope that this explanation is clear. It depends a bit on what happens when you try to save the environment settings

kaju666, 2018-07-04:
set hours as you told, then switched back to sensor mode.
After clicking submit the relay is fastly turn on and off. And should stay On becouse temp is set to 28 degree and its 27 degreee on sensor.

When i remove Power Switch from High alarm in Temperature environment then after submit the relay is switching to ON state, but ofcourse  don't turn off when temperature get to Alarm Max.

Maybe there should be an option under Power Switch allowing to set if selected power switch has to be off or on.

theyosh, 2018-07-04:
You have put your power switch in GPIO inverse mode. Try normal GPIO. Sounds like it is working exactly the inverse way

kaju666, 2018-07-05:
Yes they are in inverse mode.
I checked and when i set to normal GPIO then it stay's ON after clicking submit.
But now in switches (when it's not inverse mode) its blue when it's on and green when it's off.
Thats why i used inverse mode from begining.

So i think i found a bug :) when using inverse mode envoirment Power Switch should knnow that and also work inverse :)

PS: I also wait till temperature get AlarmMax and switch has turned off and in short time on :( so the temp now is growing more than AlarmMax.

theyosh, 2018-07-05:
You are correct. This is a bug! Very strange... because the only difference is the GPIO command. All other logic should be the same with normal GPIO en other switch types.

So I have to figure this out and see where it goes wrong in my code...

kaju666, 2018-07-06:
But i did what you say, i found that earlier that switching from gpio to gpio inverse didn't work so i deleted the switch and added once again with not inverse mode. But without inverse mode it work like i said blue is on and green is off.

And back to temperature control it is normal that the envoirment is for every 2/3 minutes clicking with switch? It should wait till temp get AlarmMax and then turn off.
And like i said after getting AlarmMax it still clicking all the time on and off shortly in about 1/2 minutes.
It also should wait till temperature lover than AlarmMax and only then power on.
It's imposibble to live with those cliks :P

theyosh, 2018-07-06:
I am sorry, but it is not clear. According to the above graphics, it is running like expected. The temperature is in BETWEEN the min and max value and therefore you cannot say if it should be on or off. If you restarted the TerrariumPI software and the first measure is withing the range, the power will be DOWN. There is no reason to trigger the power when the value is in range.

The heater will be on when the temperature reach BELOW 23 degrees and will go off when the temperature reach OVER 24 degrees.

So the above images are correct in terms of how the software is running.

But could you send me your settings.cfg file? Maybe I can see something there. Also try to do a `git pull` so that the caching issue is resolved.

kaju666, 2018-07-06:
I will reinstall full raspberry from scratch, and try again

Hmm what is not clear :P ? 
I have 4-relay switch activated by LOW state of RPI GPIO(most of them are like that).
When i use GPIO in switch setting the status are incorrect. Blue icon is ON, Green is OFF. And on Homepage the temperature red square is ON and the green square is OFF.
Thats why i use Inverse GPIO to have Blue OFF and Green ON :)
And when using Inverse GPIO the environment is not working good as we determined above

theyosh, 2018-07-06:
A full restart should not be necessary. A lot of work that is not needed.

- What I would do is trying to update with `git pull` so that you have the latest code.
- Remove the settings.cfg file
- Restart the Pi

That should give you a 'clean' version and could you reconfigure it.

And as far as I am able to test, it should just work. I still think there was a mis configuration due to caching issues that causes to use old data.

kaju666, 2018-07-06:
Too late, reinstalled full rapsberry.

After instaling your script, i added am2303 sensor succesfully.
Buuut unfortunatlu cant add any switch, after clicking Submit nothing hapens , green bar is loading fastly but no message about saved settings

theyosh, 2018-07-06:
@kaju666 I think you are currently having the same as on #168 as you already noticed. Should be fixed now.

kaju666, 2018-07-06:
Maybe log screeshot will help you :) Why it is turning off and on? Should stay off becouse temp is higher tan alarm

[image]
[image]
[image]
[image]
[image]
[image]

theyosh, 2018-07-06:
Ah, now I guess I understand what is happening.... I am judging on the images.

I see that you have selected the power switch for both low and high alarm. That is not needed. You want only low alarm. Because the high alarm is used for a cooler.

The idea is that the power switches of the low alarm setting are heaters. The power switches that are used for high alarm are coolers. So you can actively heat and cool it.

But as far as I understand you only have a heater, so you should not select power switches at the high alarm part. Alarm always means power something on. Maybe that was not clear.

For the rest all looks fine, except you should also specify the limit max of your sensor. With 0 all readings are invalid, and after 10 min the sensor will get in error state, in which will power down your heater in the environment part due to 'thinking' of faulty hardware.

I hope that this will solve it...

kaju666, 2018-07-06:
documentation will be very helpful :P, that was so small mistake and so much unnecessary combining with your and my side: P

Already set limit max and min becouse after maybe 1 minute i was having an error.

Thanks for help, i almost give up.

theyosh, 2018-07-06:
It can do a lot of different scenario's with those alarm settings. So it can sometimes be a bit difficult to get it all right. But this is the evolving of the software during the years.

But I am working on documentation. Will only take some time....

theyosh, 2018-07-06:
And if it is working you can close this issue

closed 2018-07-07 by kaju666